Arduino-CI/arduino_ci #26: Add history of `tone` / `noTone` to pinHistory or GODMODE

ifreecarve, 2018-03-28:
## Issue / Feature Request Summary

First reported in #24

The `tone` functions should create some sort of testable history in the GODMODE struct.

dfrencham, 2018-04-04:
What are your thoughts on implementing this?

Something along these lines might suffice:

- ToneHistory class, similar to pin history
- ToneHistory uses a ArduinoCIQueue to record state
- ToneState class or struct with the following members:
  - pin 
  - frequency
  - duration
  - start tick 
  - playSound (bool reflecting play start or end)

```
assertEqual(2, state->toneHistory[PIN].size());
ToneState toneStates[]  = state->toneHistory[PIN].toArray();
assetEqual(true,toneStates[0].play); // started playing
assetEqual(false,toneStates[1].play); //stopped
```

Would that fit with your code base? or is there a better way?

ifreecarve, 2018-04-04:
That's more or less what I'm thinking, although I'm not sure how best to combine that with all of `pulseIn`, `pulseOut`, `PinHistory`, etc.  I put a bunch of work into making `PinHistory` compatible with a readable bytestream to support `SoftwareSerial`.  I can't imagine any reason you'd mix those functions with tone/pulse in a real application, so it should be fine to just create a separate history queue for tone stuff.

The other thing that crossed my mind is that if you start a tone and then `delay()`, then eventually the tone will end.  I'm not sure if it's as simple as revisiting that history / Queue template that I made.  

How many use cases do you think there are?

ifreecarve, 2018-04-04:
Aah, my own ignorance getting the better of me again -- didn't know that these were PWM-pins-only features. Are there other PWM functions I need to support or are those just commanded via `analogWrite()`?