Arduino-CI/arduino_ci #29: Best practices to exclude folders from compilation?

ifreecarve, 2018-04-03:
## Issue / Feature Request Summary
A configuration parameter to exclude folders from compilation. 

Consider a case where all libraries are in `./lib` so they can all be in a single git hub repo (using git submodules). `arduino_ci` walks the tree to find all the header files - irrespective of any compiler directives to use mock headers/classes instead, the libraries get pulled in too (which may break compilation).

dolfandringa, 2019-07-27:
I would really love to have this feature. Issue #69 could have been easily fixed with this feature (jsut add .bundle to the ignore path) but I am combining arduino_ci with https://github.com/ladislas/Bare-Arduino-Project which both fill part of my use-case, but neither fill both. Right now the Bare-Arduino-Project makefiles are adding files to my repos, that cause arduino_ci tests to fail because arduino_ci is including those files in the build process. They are temporary files that I don't need, so manually removing them before running tests fixes the problem. But I wish I could just ignore the respective folders instead.

ianfixes, 2019-07-28:
`arduino_ci` is meant to test standalone libraries, not bundled libraries (i.e., it expects to run from the root of an [Arduino library](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification)). This is an interesting use case though.  I assume that you could simply change the working directory to the library you wanted to test, and then run the script.  Do you have code up that shows the failure?

dolfandringa, 2019-07-29:
@ianfixes I currently made it work from the root directory, when adding the linked PR. A demo version can be found onhttps://github.com/dolfandringa/Bare-Arduino-Project

ianfixes, 2019-07-29:
What does your bare arduino project do if there are multiple libraries in the `lib` directory?

PR #126 (2019-08-15): include ability to exclude certain directories from compilation
## Exclude certain directories from exclusion in compilation

* Add exclude_dirs option to [unittest] section in configuration file
* Add a filter in ArduinoCI::CppLibrary to filter out the files in directories specified there
* Adjust unittests to take the new feature into account

## Issues Fixed

* Fixes #29
PR #134 (2019-09-03): Add support to exclude directories
* Resolves review comments in #126
* Fixes #29

## Highlights from `CHANGELOG.md`

* Unit testing configuration now allows `exclude_dirs` to be set, which prevents stray source files from as part of unit testing allows
closed 2019-09-03 by ianfixes