Arduino-CI/arduino_ci #32: Add Windows & Appveyor CI support

ifreecarve, 2018-04-15:
## Issue / Feature Request Summary

It's time to add Windows as a supported OS.  There should be no technical limitation here beyond the difficulty of developing it without a Windows machine (relying instead on Appveyor directly).

At the moment, that failure due to lack of OS support is expressed in this form:
```
C:\User\someone\GitHub\arduino_ci\SampleProjects\DoSomething>bundle exec arduino_ci_remote.rb
Installing library under test... Traceback (most recent call last):
        5: from C:/Ruby25-x64/lib/ruby/gems/2.5.0/bin/arduino_ci_remote.rb:23:in `<main>'
        4: from C:/Ruby25-x64/lib/ruby/gems/2.5.0/bin/arduino_ci_remote.rb:23:in `load'
        3: from C:/User/someone/Documents/GitHub/arduino_ci/exe/arduino_ci_remote.rb:66:in `<top (required)>'
        2: from C:/User/someone/Documents/GitHub/arduino_ci/exe/arduino_ci_remote.rb:58:in `assure'
        1: from C:/User/someone/Documents/GitHub/arduino_ci/exe/arduino_ci_remote.rb:32:in `perform_action'
C:/User/someone/GitHub/arduino_ci/exe/arduino_ci_remote.rb:66:in `block in <top (required)>': undefined method `install_local_library' for nil:NilClass (NoMethodError)
```

ifreecarve, 2018-04-15:
Installing Arduino on Appveyor: https://thinkognito.wordpress.com/2016/09/22/continuous-integration-fuer-arduino-mit-appveyor/

```yml
version: 1.0.{build}
install:
- ps: >-
    (New-Object Net.WebClient).DownloadFile('http://downloads.arduino.cc/arduino-1.6.11-windows.zip', 'C:\projects\arduino.zip')
    Expand-Archive C:\projects\arduino.zip -dest C:\projects\arduino\

    (New-Object Net.WebClient).DownloadFile('http://downloads.arduino.cc/libraries/NicoHood/HID_Project-2.4.3.zip', 'C:\projects\hidproject.zip')
    Expand-Archive C:\projects\hidproject.zip -dest C:\projects\hidproject\
    Copy-Item 'C:\projects\hidproject\HID_Project-2.4.3' 'C:\projects\arduino\arduino-1.6.11\libraries\HID-Project' -recurse
build_script:
- cmd: >-
    cd C:\projects\arduino\arduino-1.6.11
    arduino_debug.exe --verify --board arduino:avr:leonardo --verbose-build C:\projects\REPOSITORYNAME\PATHTOSKETCH\keyboard_layouts.ino
```

ianfixes, 2018-04-25:
Counterpoint, use cygwin https://github.com/james-d-mitchell/libsemigroups/blob/master/appveyor.yml

```yml
# set clone depth
clone_depth: 5                  # clone entire repository history if not defined

environment:
  matrix:
    - CYG_ARCH: x86
      CYG_ROOT: C:/cygwin
    - CYG_ARCH: x86_64
      CYG_ROOT: C:/cygwin64

install:
  - '%CYG_ROOT%\setup-%CYG_ARCH%.exe -qnNdO -R %CYG_ROOT% -s http://cygwin.mirror.constant.com -l %CYG_ROOT%/var/cache/setup -P autoconf -P automake -P bison -P libgmp-devel -P gcc-core -P gcc-g++ -P mingw-runtime -P mingw-binutils -P mingw-gcc-core -P mingw-gcc-g++ -P mingw-pthreads -P mingw-w32api -P libtool -P make -P python -P gettext-devel -P gettext -P intltool -P libiconv -P pkg-config -P git -P wget -P curl'

build: off

test_script:
  - '%CYG_ROOT%/bin/bash -lc "g++ --version && cd $APPVEYOR_BUILD_FOLDER && ./autogen.sh && ./configure && make check"'
```

ianfixes, 2018-04-25:
Also, https://www.appveyor.com/docs/lang/ruby/

```yml
install:
  - set PATH=C:\Ruby22\bin;%PATH%
  - bundle install

build: off

before_test:
  - ruby -v
  - gem -v
  - bundle -v

test_script:
  - bundle exec rake
```

ianfixes, 2018-04-25:
> If you run the following in a windows shell : 

```
REG QUERY "HKLM\SOFTWARE\Classes\Arduino file\shell\open\command"
```

> You will get the response (assuming it has been installed to C:\Program Files (x86)\Arduino):

```
(Default)    REG_SZ    "C:\Program Files (x86)\Arduino\arduino.exe" "%1"
```
> And if it doesn't exists you should see:

```
ERROR: The system was unable to find the specified registry key or value.
```

per1234, 2018-04-25:
There will only be a registry entry if the Arduino IDE installer was used (or someone manually associated the .ino extension).

ianfixes, 2018-04-25:
That's OK, default behavior is to `force_install` the Arduino binary if the registry key can't be found. That will be the case on any given CI machine.

I would welcome any suggestions you have for fixing https://github.com/ianfixes/arduino_ci/pull/39
 since I don't have a windows machine to test on

tomduff, 2018-05-03:
I ran this on my windows box and found that the following would provide you with the install location:

    # The executable Arduino file in an existing installation, or nil
    # @return [string]
    def self.existing_executable
      arduino_reg = 'SOFTWARE\WOW6432Node\Arduino'
      Win32::Registry::HKEY_LOCAL_MACHINE.open(arduino_reg) do |reg| 
        location = reg.read_s('Install_Dir') 
        puts 'Arduino Install Dir: ' + location
        return location
      end
      nil
    end

It then runs to here:

    The set of compilers (0) isn't empty...                                        ✗
    This may indicate a problem with ArduinoCI!
    Failures: 1

Which  is an issue with the .arduino_ci.yaml file in the DoSomething example. 

Correcting that gets as far as trying to execute g++ which I don't think I have as my machine lunched itself and I am yet to put it back on - I'll see if I can find sometime to get it and add it to my path and let you know what happens.Although I am not 100% sure that what I return in the self.existing_executable function is what you are after.

PR #39 (2018-05-06): Enable Windows CI
## Highlights from `CHANGELOG.md`

* See CHANGELOG.md for more

## Issues Fixed

* Fixes #32
* Fixes #41 
* Fixes #45
closed 2018-05-06 by ianfixes