Arduino-CI/arduino_ci #36: Exception in thread "main" java.lang.NoSuchMethodError: com.apple.eawt.Application.isAboutMenuItemPresent()

mdlima, 2018-04-23:
## System

- OS: OSX
- `ruby -v`:
ruby 2.3.6p384 (2017-12-14 revision 61254) [x86_64-darwin17]
(tried with 2.4 as well)
- `bundle -v`:
Bundler version 1.16.1
- `bundle info arduino_ci`:
  * arduino_ci (0.1.9)
	Summary: Tools for building and unit testing Arduino libraries
	Homepage: http://github.com/ifreecarve/arduino_ci
	Path: /Users/mdlima/Projects/github/arduino_ci
- `g++ -v`:
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 9.1.0 (clang-902.0.39.1)
Target: x86_64-apple-darwin17.5.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
- Arduino IDE version: 1.8.5
- URL of failing Travis CI job: -
- URL of your Arduino project: https://github.com/mdlima/arduino_cob_led_controller

## Issue / Feature Request Summary

Hi Ian, thanks for sharing this work, I've been trying to find a way to unit test my Arduino code and this looks very promising.

I've followed the README steps but am getting an error when executing the gem. I've tried running the example from the TestSomething folder here and got the exact same error. Tried looking for this error but got nowhere.

It would be very helpful if you can point me to where to look for finding a solution. Thanks in advance.

Here's the output:
```
TestSomething ❯❯❯ bundle exec arduino_ci_remote.rb                                      master ✱ ◼
Installing library under test...                                               ✓
Library installed at /Users/mdlima/Documents/Arduino/libraries/TestSomething... ✓
The set of compilers (1) isn't empty...                                        ✓
Checking g++ version...
    Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
    Apple LLVM version 9.1.0 (clang-902.0.39.1)
    Target: x86_64-apple-darwin17.5.0
    Thread model: posix
    InstalledDir: /Library/Developer/CommandLineTools/usr/bin
...Checking g++ version                                                        ✓
Installing board package arduino:sam...                                        ✗
This may indicate a problem with ArduinoCI!
Failures: 1
Last message:  $  java -cp /Applications/Arduino.app/Contents/Java/* -DAPP_DIR=/Applications/Arduino.app/Contents/Java -Dfile.encoding=UTF-8 -Dapple.awt.UIElement=true -Xms128M -Xmx512M processing.app.Base --install-boards arduino:sam
========== Stdout:

========== Stderr:
Exception in thread "main" java.lang.NoSuchMethodError: com.apple.eawt.Application.isAboutMenuItemPresent()Z
	at processing.app.Base.main(Base.java:138)
```

The jar files are in the expected `/Applications/Arduino.app/Contents/Java/` folder.

## Arduino or Unit Test Code, Illustrating the Problem

Error returned when running `bundle exec arduino_ci_remote.rb` from the TestSomething example.

## Arduino Architecture(s) Affected

Uno, all?

ianfixes, 2018-04-23:
Interesting!  It might have something to do with a hack that I put in that avoids showing the splash screen, although I'm running the same Arduino IDE version and I don't get that particular error.  What version of Java are you using?  

My thinking is that I can test whether my hack works, and if it doesn't then I'll fall back on using the "official" Arduino CLI directly

ianfixes, 2018-04-23:
Also, you can see from the output the last command I was attempting: 

```java -cp /Applications/Arduino.app/Contents/Java/* -DAPP_DIR=/Applications/Arduino.app/Contents/Java -Dfile.encoding=UTF-8 -Dapple.awt.UIElement=true -Xms128M -Xmx512M processing.app.Base --install-boards arduino:sam```

Without my hack, the correct way to run that from the command line would be 

```/Applications/Arduino.app/Contents/MacOS/Arduino --install-boards arduino:sam```

I'd be curious whether that command works for you.  If so, I can try to put some code in that would try to run 

```java -cp /Applications/Arduino.app/Contents/Java/* -DAPP_DIR=/Applications/Arduino.app/Contents/Java -Dfile.encoding=UTF-8 -Dapple.awt.UIElement=true -Xms128M -Xmx512M processing.app.Base --bogus-option```

And if I get a java exception instead of `Error: unknown option: --bogus-option` in the output, I can fall back on the "normal" launch method.  Let me know if any of these commands work on your machine, and I can push up a development branch for you to try.

mdlima, 2018-04-23:
I like the idea for the hack, let's see if we can figure it out. I use VSCode and I get the splash screen every time I compile or run the Arduino code, and it gets in the way if I'm using VSCode on full screen as it will switch to the other "Desktop". It also seems a little slow, so maybe the hack can speed things up.

I'm using Java version 9.0.4:
```
$ java -version
java version "9.0.4"
Java(TM) SE Runtime Environment (build 9.0.4+11)
Java HotSpot(TM) 64-Bit Server VM (build 9.0.4+11, mixed mode)
```

The install boards option with Java gets the same exception:
```
$ java -cp "/Applications/Arduino.app/Contents/Java/*" -DAPP_DIR=/Applications/Arduino.app/Contents/Java -Dfile.encoding=UTF-8 -Dapple.awt.UIElement=true -Xms128M -Xmx512M processing.app.Base --install-boards arduino:sam
Exception in thread "main" java.lang.NoSuchMethodError: com.apple.eawt.Application.isAboutMenuItemPresent()Z
	at processing.app.Base.main(Base.java:138)
```

The Arduino command line works fine:
```
$ /Applications/Arduino.app/Contents/MacOS/Arduino --install-boards arduino:sam
Loading configuration...
Initializing packages...
Preparing boards...
Downloading platforms index...
```

mdlima, 2018-04-23:
And using the old way (uncommented from arduino_installation_rb:37 it works:
```
$ bundle exec arduino_ci_remote.rb                                  ✘ 1 master ✱ ◼
Installing library under test...                                               ✓
Library installed at /Users/mdlima/Documents/Arduino/libraries/TestSomething... ✓
The set of compilers (1) isn't empty...                                        ✓
Checking g++ version...
    Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
    Apple LLVM version 9.1.0 (clang-902.0.39.1)
    Target: x86_64-apple-darwin17.5.0
    Thread model: posix
    InstalledDir: /Library/Developer/CommandLineTools/usr/bin
...Checking g++ version                                                        ✓
Installing board package arduino:sam...                                        ✓
```

ianfixes, 2018-04-23:
That's my thinking here.  I'd like to preserve the hack (for both speed and usability).  I'm writing some code now to try the hack, and (if I don't get the output I expect) fall back on the "official" method.

mdlima, 2018-04-23:
Cool, we can get the fallback working then, but I'm also interested in why your hack won't work here. Are you using Java 9 too?

ianfixes, 2018-04-23:
```$ java -version
java version "1.8.0_111"
Java(TM) SE Runtime Environment (build 1.8.0_111-b14)
Java HotSpot(TM) 64-Bit Server VM (build 25.111-b14, mixed mode)```

closed 2018-04-25 by ianfixes

ianfixes, 2018-04-25:
This is fixed in the branch but not in released gem yet

PR #37 (2018-04-25): Bugfixes in support of 0.1.10 release
## Highlights from `CHANGELOG.md`

* Preparing for the addition of windows CI support
* Improve logging when necessary builtin tools aren't found (e.g. wget)
* Attempt downloads more than once
* Change author to my new username
* Improve some documentation
* See CHANGELOG.md for more

## Issues Fixed

* Fixes #33
* Fixes #36