Arduino-CI/arduino_ci #40: SampleProjects/DoSomething and SampleProjects/TestSomething should be in their own repository

scls19fr, 2018-05-03:
Hello,

I think  SampleProjects/DoSomething and SampleProjects/TestSomething should be in their own repository
so it will be easier for a library creator to just look at one of this repository.
Moreover these repositories could have their own build and testing infrastructure with Travis

Kind regards

ianfixes, 2018-05-03:
It's possible to copy these into their own repositories, but I wouldn't like to maintain them that way.  Both of these repositories serve a specific purpose for CI that I do on `arduino_ci` itself.  

* `DoSomething` allows me to [test whether unit test passes and failures are parsed correctly by the test runner](https://github.com/ianfixes/arduino_ci/blob/master/spec/cpp_library_spec.rb)
* `TestSomething` allows me to [test all mock and assertion features of my library](https://github.com/ianfixes/arduino_ci/blob/master/spec/testsomething_unittests_spec.rb)

Regarding your other point:

> these repositories could have their own build and testing infrastructure with Travis

I do that as well.  In `.travis.yml`, you'll see that I'm going into the `TestSomething` project directory and running the build/test script on it: https://github.com/ianfixes/arduino_ci/blob/master/.travis.yml#L17-L19

So in short, by bundling the sample repos with the main CI project, I'm able to use them to check the correctness of the CI scripts themselves.  This has already proven necessary; see #30

Can you talk more about the difficulty you had in setting up a CI-enabled repository for an Arduino library from scratch?  Perhaps I can improve my documentation.

scls19fr, 2018-05-03:
My idea is to remove from SampleProjects/DoSomething all Ruby things such as Gemfile and Gemfile.lock and have something in https://github.com/ianfixes/arduino_ci/blob/master/SampleProjects/DoSomething/.travis.yml

more comparable to 

https://github.com/SMFSW/Queue/blob/70b5436e780243a6eea39f17e772eac702c90685/.travis.yml#L16

(inspired by https://github.com/adafruit/travis-ci-arduino/blob/855c1cb7577f56ea086ea3615828a078fd1242c0/.travis.yml#L4 ) ie with source/curl command to download and scripts from a source repository (https://github.com/ianfixes/arduino_ci/ is this case or a fork of https://github.com/ianfixes/arduino_ci/ )

Although, when changing code in https://github.com/ianfixes/arduino_ci/ you will have to trigger manually CI for https://github.com/ianfixes/DoSomething/ but when developing really, you change more often DoSomething than arduino_ci

I also wonder why https://github.com/ianfixes/arduino_ci/blob/921925f90bf8088b78d847390acdd5390cfbe60c/SampleProjects/TestSomething/Gemfile.lock#L2 have raw value

ianfixes, 2018-05-03:
The existence of `Gemfile.lock` in the repo is a bug, I've opened #41 to track this.

Setting up an `install.sh`-style script sounds like a good idea, I've set up #42 to track this.

closed 2018-05-03 by ianfixes

scls19fr, 2018-05-03:
Thanks @ianfixes  

I wonder why `Gemfile.lock` is present is this repository.
It shouldn't according

https://github.com/ianfixes/arduino_ci/blob/921925f90bf8088b78d847390acdd5390cfbe60c/.gitignore#L3

Maybe `/Gemfile.lock` should be replaced by `Gemfile.lock`?

ianfixes, 2018-05-03:
Yup, I've already made this fix locally