Arduino-CI/arduino_ci #69: Compilation fails due to files given multiple times as arguments to the compiler

hlovdal, 2018-12-30:
## System

- OS: Linux (Fedora 29)
- `ruby -v`: ruby 2.5.3p105 (2018-10-18 revision 65156) [x86_64-linux]
- `bundle -v`: Bundler version 1.17.3
- `bundle info arduino_ci`:
```shell
  * arduino_ci (0.1.14)
        Summary: Tools for building and unit testing Arduino libraries
        Homepage: http://github.com/ianfixes/arduino_ci
        Path: /tmp/dummy/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.14
```
- `g++ -v`:
```shell
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/8/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,objc,obj-c++,ada,go,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl --enable-libmpx --enable-offload-targets=nvptx-none --without-cuda-driver --enable-gnu-indirect-function --enable-cet --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
Thread model: posix
gcc version 8.2.1 20181215 (Red Hat 8.2.1-6) (GCC) 
```
- Arduino IDE version: 1.8.8
- `java -version`:
```shell
openjdk version "1.8.0_191"
OpenJDK Runtime Environment (build 1.8.0_191-b12)
OpenJDK 64-Bit Server VM (build 25.191-b12, mixed mode)
```

## Issue Summary

I try to use arduino_ci as a unit test tool for local development, e.g. only on my machine, not in a github repo.

Something triggers arduino_ci to include some files twice, causing the compiler to fail because of duplicate definitions.

## Arduino or Unit Test Code, Illustrating the Problem

I have created a minimal example to trigger this. Steps to reproduce:

```shell
$ mkdir /tmp/dummy
$ cd /tmp/dummy
$ cat > dummy.ino
void setup() {
}
void loop() {
}
$ mkdir test
$ cat > test/dummy.cpp
#include <ArduinoUnitTests.h>

unittest(dummy_test)
{
  assertEqual(4, 4);
}

unittest_main()
$ cat > Gemfile
source 'https://rubygems.org'
gem 'arduino_ci'
$ cat > .arduino-ci.yaml
compile:
  libraries: ~
  platforms:
    - uno

unittest:
[... 557 lines cut ...]
```shell
g++
-std=c++0x
-o
/tmp/dummy/unittest_dummy.cpp.bin
-DARDUINO=100
-D__AVR_ATmega328P__
-I/tmp/dummy/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.14/cpp/arduino
-I/tmp/dummy/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.14/cpp/unittest
-I/tmp/dummy/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.14/cpp/arduino
-I/tmp/dummy/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.14/cpp/arduino/avr
-I/tmp/dummy/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.14/cpp/arduino/ci
-I/tmp/dummy/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.14/cpp/unittest
/tmp/dummy/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.14/cpp/arduino/Arduino.cpp
/tmp/dummy/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.14/cpp/arduino/Godmode.cpp
/tmp/dummy/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.14/cpp/unittest/ArduinoUnitTests.cpp
/tmp/dummy/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.14/cpp/arduino/Arduino.cpp
/tmp/dummy/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.14/cpp/arduino/Godmode.cpp
/tmp/dummy/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.14/cpp/unittest/ArduinoUnitTests.cpp
/tmp/dummy/test/dummy.cpp
```

## Arduino Architecture(s) Affected

Only tested for Uno, but I assume this is generic.

ianfixes, 2018-12-31:
This seems like a strange bug, since the file list is taken from 3 places that should not overlap.

Can you try this patch?  It should add some harmless command line options to break up the file listing and hopefully help isolate the code that's creating those duplicate entries:

```
diff --git a/lib/arduino_ci/cpp_library.rb b/lib/arduino_ci/cpp_library.rb
index 7df0e44..223f925 100644
--- a/lib/arduino_ci/cpp_library.rb
+++ b/lib/arduino_ci/cpp_library.rb
@@ -241,8 +241,11 @@ module ArduinoCI
     def test_args(aux_libraries, ci_gcc_config)
       # TODO: something with libraries?
       ret = include_args(aux_libraries)
+      ret += ["-DARDUINO_CI_cpp_files_arduino"]
       ret += cpp_files_arduino.map(&:to_s)
+      ret += ["-DARDUINO_CI_cpp_files_unittest"]
       ret += cpp_files_unittest.map(&:to_s)
+      ret += ["-DARDUINO_CI_cpp_files"]
       ret += cpp_files.map(&:to_s)
       unless ci_gcc_config.nil?
         cgc = ci_gcc_config
```

hlovdal, 2019-01-26:
With that patch the last command becomes:

```shell
Last
command:
$
g++
-std=c++0x
-o
/tmp/dummy2/unittest_dummy.cpp.bin
-DARDUINO=100
-D__AVR_ATmega328P__
-I/tmp/dummy2/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.17/cpp/arduino
-I/tmp/dummy2/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.17/cpp/unittest
-I/tmp/dummy2/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.17/cpp/arduino
-I/tmp/dummy2/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.17/cpp/arduino/avr
-I/tmp/dummy2/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.17/cpp/arduino/ci
-I/tmp/dummy2/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.17/cpp/unittest
-DARDUINO_CI_cpp_files_arduino
/tmp/dummy2/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.17/cpp/arduino/Arduino.cpp
/tmp/dummy2/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.17/cpp/arduino/Godmode.cpp
/tmp/dummy2/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.17/cpp/arduino/stdlib.cpp
-DARDUINO_CI_cpp_files_unittest
/tmp/dummy2/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.17/cpp/unittest/ArduinoUnitTests.cpp
-DARDUINO_CI_cpp_files
/tmp/dummy2/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.17/cpp/arduino/Arduino.cpp
/tmp/dummy2/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.17/cpp/arduino/Godmode.cpp
/tmp/dummy2/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.17/cpp/arduino/stdlib.cpp
/tmp/dummy2/bundle_install/ruby/2.5.0/gems/arduino_ci-0.1.17/cpp/unittest/ArduinoUnitTests.cpp
/tmp/dummy2/test/dummy.cpp
```

ianfixes, 2019-01-28:
Aah interesting.  I was in the middle of writing a long comment about how I couldn't figure out the problem but it just jumped out at me.  If I'm reading this correctly, your library is installed at /tmp/dummy2 and the directory of gem dependencies happens to live inside it.  So `arduino_ci` is finding and including itself.

I would have thought the bundle stuff would go into `vendor/bundle` (which I explicitly check for), but in this case it's going into `bundle_install/` somehow.  Did you explicitly do that?  What OS are you on?

I'm going to have to fix `CppLibrary::vendor_bundle?`.  Question for my future self and/or more knowledgable visitors: is there a way to get the master `$BUNDLE_PATH` instead of iterating over all `bundle show --paths` dirs?

ianfixes, 2019-01-28:
@hlovdal if/when you can verify that #98 fixes this, I'll merge that code.

closed 2019-01-28 by ianfixes

hlovdal, 2019-01-30:
Hm, while the change did fix the minimal example reported above which I tested, it breaks when I try to update my main project to 0.1.18. I bisected the change to be this `Fix vendor_bundle?` commit.

Forcing a print of the last command regardless of failure gives an opportunity to make a diff of the successful case using the parent commit of `Fix vendor_bundle?` and the failing one:

```patch
 g++
 -std=c++0x
 -o
 /share/hlovdal/arduino/sketchbook/proj1/unittest_cust_char_tests.cpp.bin
 -DARDUINO=100
 -D__AVR_ATmega328P__
 -I/share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/arduino
 -I/share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/unittest
--I/share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/arduino/avr
--I/share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/arduino/ci
--I/share/hlovdal/arduino/sketchbook/proj1/src
 -I/share/hlovdal/arduino/sketchbook/libraries/SmartLCD
 -DARDUINO_CI_cpp_files_arduino
 /share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/arduino/Arduino.cpp
 /share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/arduino/Godmode.cpp
 /share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/arduino/stdlib.cpp
 -DARDUINO_CI_cpp_files_unittest
 /share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/unittest/ArduinoUnitTests.cpp
 -DARDUINO_CI_cpp_files
-/share/hlovdal/arduino/sketchbook/proj1/src/proj1_keys.cpp
-/share/hlovdal/arduino/sketchbook/proj1/src/proj1_lcd.cpp
-/share/hlovdal/arduino/sketchbook/proj1/src/cust_char.cpp
 -DUNIT_TEST
 -g
 /share/hlovdal/arduino/sketchbook/libraries/SmartLCD/SmartLCDI2C.cpp
 /share/hlovdal/arduino/sketchbook/proj1/test/cust_char_tests.cpp
```

hlovdal, 2019-01-30:
Full compile output (filtered though the `uniq` command)
```shellLocated Arduino binary...                   /home/hlovdal/arduino_ci_ide/arduino
{:compilers=>["g++"], :libraries=>["SmartLCD"], :platforms=>["uno"]}
The set of compilers (1) isn't empty...                                        ✓
Checking g++ version... 
    Using built-in specs.
    COLLECT_GCC=g++
    COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/8/lto-wrapper
    OFFLOAD_TARGET_NAMES=nvptx-none
    OFFLOAD_TARGET_DEFAULT=1
    Target: x86_64-redhat-linux
    Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,objc,obj-c++,ada,go,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl --enable-libmpx --enable-offload-targets=nvptx-none --without-cuda-driver --enable-gnu-indirect-function --enable-cet --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
    Thread model: posix
    gcc version 8.2.1 20181215 (Red Hat 8.2.1-6) (GCC) 
...Checking g++ version                                                        ✓
libasan availability for g++...                                            false
Requested unittest platform 'uno' is defined in 'platforms' YML...             ✓
Unit testing cust_char_tests.cpp with g++... 
bundle_info = {:out=>"/share/hlovdal/arduino/sketchbook/proj1/arduino_ci\n/home/hlovdal/.gem/ruby/gems/bundler-2.0.1/lib/gems/bundler-2.0.1\n/share/hlovdal/arduino/sketchbook/proj1/bundle_install/ruby/2.5.0/gems/os-1.0.0\n/share/hlovdal/arduino/sketchbook/proj1/bundle_install/ruby/2.5.0/gems/rubyzip-1.2.2\n", :err=>"", :success=>true}
bundle_paths = [#<Pathname:/share/hlovdal/arduino/sketchbook/proj1>, #<Pathname:/share/hlovdal/arduino/sketchbook/proj1/bundle_install/ruby/2.5.0/gems>]
gem_path = /share/hlovdal/arduino/sketchbook/proj1
ARDUINO_HEADER_DIR = /share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/arduino
UNITTEST_HEADER_DIR = /share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/unittest
@base_dir = .
all = [#<Pathname:/share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/arduino/Arduino.cpp>, #<Pathname:/share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/arduino/Godmode.cpp>, #<Pathname:/share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/arduino/stdlib.cpp>, #<Pathname:/share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/unittest/ArduinoUnitTests.cpp>, #<Pathname:/share/hlovdal/arduino/sketchbook/proj1/src/proj1_keys.cpp>, #<Pathname:/share/hlovdal/arduino/sketchbook/proj1/src/proj1_lcd.cpp>, #<Pathname:/share/hlovdal/arduino/sketchbook/proj1/src/cust_char.cpp>, #<Pathname:/share/hlovdal/arduino/sketchbook/proj1/test/cust_char_tests.cpp>, #<Pathname:/share/hlovdal/arduino/sketchbook/proj1/test/keypress_fsm.cpp>]
gem_path = /share/hlovdal/arduino/sketchbook/proj1
v = []
gem_path = /share/hlovdal/arduino/sketchbook/proj1

Last command:  $ g++ -std=c++0x -o /share/hlovdal/arduino/sketchbook/proj1/unittest_cust_char_tests.cpp.bin -DARDUINO=100 -D__AVR_ATmega328P__ -I/share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/arduino -I/share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/unittest -I/share/hlovdal/arduino/sketchbook/libraries/SmartLCD -DARDUINO_CI_cpp_files_arduino /share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/arduino/Arduino.cpp /share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/arduino/Godmode.cpp /share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/arduino/stdlib.cpp -DARDUINO_CI_cpp_files_unittest /share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/unittest/ArduinoUnitTests.cpp -DARDUINO_CI_cpp_files -DUNIT_TEST -g /share/hlovdal/arduino/sketchbook/libraries/SmartLCD/SmartLCDI2C.cpp /share/hlovdal/arduino/sketchbook/proj1/test/cust_char_tests.cpp

/usr/bin/ld: /tmp/ccrfQOch.o: in function `test_cust_char_lookup_empty_string::task()':
/share/hlovdal/arduino/sketchbook/proj1/test/cust_char_tests.cpp:22: undefined reference to `proj1::cust_char_lookup(unsigned char const*)'
/usr/bin/ld: /tmp/ccrfQOch.o: in function `test_cust_char_lookup_ASCII_string::task()':
/share/hlovdal/arduino/sketchbook/proj1/test/cust_char_tests.cpp:34: undefined reference to `proj1::cust_char_lookup(unsigned char const*)'
/usr/bin/ld: /tmp/ccrfQOch.o: in function `test_cust_char_lookup_UTF8_string::task()':
/share/hlovdal/arduino/sketchbook/proj1/test/cust_char_tests.cpp:46: undefined reference to `proj1::cust_char_lookup(unsigned char const*)'
/usr/bin/ld: /tmp/ccrfQOch.o: in function `test_cust_char_lookup_unsupported_UTF8_character::task()':
/share/hlovdal/arduino/sketchbook/proj1/test/cust_char_tests.cpp:62: undefined reference to `proj1::cust_char_lookup(unsigned char const*)'
collect2: error: ld returned 1 exit status
...Unit testing cust_char_tests.cpp with g++                                   ✗
Unit testing keypress_fsm.cpp with g++... 
gem_path = /share/hlovdal/arduino/sketchbook/proj1
ARDUINO_HEADER_DIR = /share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/arduino
UNITTEST_HEADER_DIR = /share/hlovdal/arduino/sketchbook/proj1/arduino_ci/cpp/unittest
@base_dir = .
[... cut ...]
Using built-in board package...                                      arduino:avr
Installing board package arduino:avr...                                        ✓
Using pre-existing library...                                           SmartLCD
Setting compiler warning level...                                              ✓
Switching to board for uno (arduino:avr:uno)...                                ✓
Verifying empty.ino...                                                         ✓
Failures: 1
```

hlovdal, 2019-01-30:
With the following debug changes:

```patch
commit 30a5b0b9ec845f43dc5b232a8f703b0686e2ea09
Author: Håkon Løvdal <kode@denkule.no>
Date:   Wed Jan 30 22:21:18 2019 +0100

    debug

diff --git a/lib/arduino_ci/cpp_library.rb b/lib/arduino_ci/cpp_library.rb
index dbff893..5ce59a0 100644
--- a/lib/arduino_ci/cpp_library.rb
+++ b/lib/arduino_ci/cpp_library.rb
@@ -58,6 +58,7 @@ module ArduinoCI
       return false unless base.exist?
 
       vendor_bundle_aliases = [base, base.realpath]
+      puts "vendor_bundle_aliases = #{vendor_bundle_aliases}"
 
       # we could do this but some rubies don't return an enumerator for ascend
       # path.ascend.any? { |part| vendor_bundle_aliases.include?(part) }
@@ -117,6 +118,10 @@ module ArduinoCI
     # @return [Array<Pathname>]
     def cpp_files
       puts "@base_dir = #{@base_dir}"
+      all = cpp_files_in(@base_dir)
+      puts "all = #{all}"
+      v = cpp_files_in(@base_dir).reject { |p| vendor_bundle?(p) }
+      puts "v = #{v}"
       cpp_files_in(@base_dir).reject { |p| vendor_bundle?(p) || in_tests_dir?(p) }
     end
 

commit 7b527931d6219185fef51d4e7843b4a8dfee7f62
Author: Håkon Løvdal <kode@denkule.no>
Date:   Sun Jan 27 15:05:35 2019 +0100

    Issue 96 more debug

diff --git a/lib/arduino_ci/cpp_library.rb b/lib/arduino_ci/cpp_library.rb
index 19d8889..dbff893 100644
--- a/lib/arduino_ci/cpp_library.rb
+++ b/lib/arduino_ci/cpp_library.rb
@@ -116,18 +116,21 @@ module ArduinoCI
     # CPP files that are part of the project library under test
     # @return [Array<Pathname>]
     def cpp_files
+      puts "@base_dir = #{@base_dir}"
       cpp_files_in(@base_dir).reject { |p| vendor_bundle?(p) || in_tests_dir?(p) }
     end
 
     # CPP files that are part of the arduino mock library we're providing
     # @return [Array<Pathname>]
     def cpp_files_arduino
+      puts "ARDUINO_HEADER_DIR = #{ARDUINO_HEADER_DIR}"
       cpp_files_in(ARDUINO_HEADER_DIR)
     end
 
     # CPP files that are part of the unit test library we're providing
     # @return [Array<Pathname>]
     def cpp_files_unittest
+      puts "UNITTEST_HEADER_DIR = #{UNITTEST_HEADER_DIR}"
       cpp_files_in(UNITTEST_HEADER_DIR)
     end
 

commit 8f5fdbe07fac6a8f1d28f574134c38a46c93b60e
Author: Håkon Løvdal <kode@denkule.no>
Date:   Sun Jan 27 15:04:39 2019 +0100

    Issue 69 debug
    
    https://github.com/ianfixes/arduino_ci/issues/69#issuecomment-450652373

diff --git a/lib/arduino_ci/cpp_library.rb b/lib/arduino_ci/cpp_library.rb
index f6f5833..19d8889 100644
--- a/lib/arduino_ci/cpp_library.rb
+++ b/lib/arduino_ci/cpp_library.rb
@@ -243,8 +243,11 @@ module ArduinoCI
     def test_args(aux_libraries, ci_gcc_config)
       # TODO: something with libraries?
       ret = include_args(aux_libraries)
+      ret += ["-DARDUINO_CI_cpp_files_arduino"]
       ret += cpp_files_arduino.map(&:to_s)
+      ret += ["-DARDUINO_CI_cpp_files_unittest"]
       ret += cpp_files_unittest.map(&:to_s)
+      ret += ["-DARDUINO_CI_cpp_files"]
       ret += cpp_files.map(&:to_s)
       unless ci_gcc_config.nil?
         cgc = ci_gcc_config

```

ianfixes, 2019-01-31:
Most recent gem is `0.1.19` although I don't think that update touches any of this.  Are you saying that it worked in 90b3e68403c37b4d0a31a8686d6bf1f3560712db but fails in ad731a1b823d30c5d5f8fe00d5425f41c2d64d56 (`v0.1.18`)?

Can you give me more hints on how to reproduce this behavior locally?  I did the debug in a slightly different way that you might prefer:

```patch
diff --git a/lib/arduino_ci/cpp_library.rb b/lib/arduino_ci/cpp_library.rb
index f6f5833..19d8889 100644
--- a/lib/arduino_ci/cpp_library.rb
+++ b/lib/arduino_ci/cpp_library.rb
@@ -243,8 +243,11 @@ module ArduinoCI
     def test_args(aux_libraries, ci_gcc_config)
       # TODO: something with libraries?
       ret = include_args(aux_libraries)
+      ret << "-DARDUINO_CI_cpp_files_arduino=#{ARDUINO_HEADER_DIR}:#{ARDUINO_HEADER_DIR.realpath}"
       ret += cpp_files_arduino.map(&:to_s)
+      ret << "-DARDUINO_CI_cpp_files_unittest=#{UNITTEST_HEADER_DIR}:#{UNITTEST_HEADER_DIR.realpath}"
       ret += cpp_files_unittest.map(&:to_s)
+      ret << "-DARDUINO_CI_cpp_files=#{@base_dir}:#{@base_dir.realpath}"
       ret += cpp_files.map(&:to_s)
       unless ci_gcc_config.nil?
         cgc = ci_gcc_config
```

hlovdal, 2019-01-31:
No, I am saying it stops working in 90b3e68 and from then on. The newer code is much better at excluding itself, before I had to `git rm -r SampleProjects` and `git rm -r spec/fake_library` when using arduino_ci as a git submodule, but that is no longer required. Although it apparently excludes my source files as well...

Here is a minimal example to reproduce (this time with paste friendly commands):

```shell
mkdir /tmp/compiletest
cd /tmp/compiletest

cat > .arduino-ci.yaml << EOF
compile:
  libraries: ~
  platforms:
    - uno

unittest:
  libraries: ~
  platforms:
    - uno
EOF

cat > Gemfile << EOF
source 'https://rubygems.org'
gem 'arduino_ci', :path => "arduino_ci"
EOF

mkdir src
echo 'bool greater_than_five(int n);' > src/five.hpp

cat > src/five.cpp << EOF
#include "five.hpp"

bool greater_than_five(int n) {
        return n > 5;
}
EOF

mkdir test
cat > test/five_tests.cpp << EOF
#include <ArduinoUnitTests.h>

#include "../src/five.hpp"

unittest(one)
{
        assertFalse(greater_than_five(1));
}

unittest(ten)
{
        assertTrue(greater_than_five(10));
}

unittest_main()
EOF

git clone https://github.com/ianfixes/arduino_ci
cd arduino_ci
git am .../debug.patch # https://github.com/ianfixes/arduino_ci/issues/69#issuecomment-459180028
cd ..
bundle install --path vendor/bundle
bundle exec arduino_ci_remote.rb --skip-compilation
```

which produces 

```shell
The set of compilers (1) isn't empty...                                        ✓
Checking g++ version... 
    Using built-in specs.
    COLLECT_GCC=g++
    COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/8/lto-wrapper
    OFFLOAD_TARGET_NAMES=nvptx-none
    OFFLOAD_TARGET_DEFAULT=1
    Target: x86_64-redhat-linux
    Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,objc,obj-c++,ada,go,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl --enable-libmpx --enable-offload-targets=nvptx-none --without-cuda-driver --enable-gnu-indirect-function --enable-cet --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
    Thread model: posix
    gcc version 8.2.1 20181215 (Red Hat 8.2.1-6) (GCC) 
...Checking g++ version                                                        ✓
libasan availability for g++...                                            false
Requested unittest platform 'uno' is defined in 'platforms' YML...             ✓
Unit testing five_tests.cpp with g++... 

Last command:  $ g++ -std=c++0x -o /tmp/compiletest/unittest_five_tests.cpp.bin -DARDUINO=100
-D__AVR_ATmega328P__ -I/tmp/compiletest/arduino_ci/cpp/arduino -I/tmp/compiletest/arduino_ci/cpp/unittest
-DARDUINO_CI_cpp_files_arduino=/tmp/compiletest/arduino_ci/cpp/arduino:/tmp/compiletest/arduino_ci/cpp/arduino /tmp/compiletest/arduino_ci/cpp/arduino/Arduino.cpp /tmp/compiletest/arduino_ci/cpp/arduino/Godmode.cpp /tmp/compiletest/arduino_ci/cpp/arduino/stdlib.cpp 
-DARDUINO_CI_cpp_files_unittest=/tmp/compiletest/arduino_ci/cpp/unittest:/tmp/compiletest/arduino_ci/cpp/unittest /tmp/compiletest/arduino_ci/cpp/unittest/ArduinoUnitTests.cpp 
-DARDUINO_CI_cpp_files=.:/tmp/compiletest /tmp/compiletest/test/five_tests.cpp

/usr/bin/ld: /tmp/cctXTkoJ.o: in function `test_one::task()':
five_tests.cpp:(.text+0x17): undefined reference to `greater_than_five(int)'
/usr/bin/ld: /tmp/cctXTkoJ.o: in function `test_ten::task()':
five_tests.cpp:(.text+0x85): undefined reference to `greater_than_five(int)'
collect2: error: ld returned 1 exit status
...Unit testing five_tests.cpp with g++                                        ✗
Skipping compilation of examples...                as requested via command line
Failures: 1
```

ianfixes, 2019-02-01:
Oh, I see.  You are downloading `arduino_ci` itself into your library's root directory.  That seems like incorrect usage, and (more to the point) I'm not sure how I'd properly detect it in order to avoid it.

Your `Gemfile` should probably be formed like this:

```shell
cat > Gemfile << EOF
source 'https://rubygems.org'
gem 'arduino_ci', :path => "../arduino_ci"  # note navigation to parent directory
EOF
```

And the installation of arduino_ci should go like this:

```shell
mkdir test
cat > test/five_tests.cpp << EOF
  // blah blah blah
EOF

pushd ".."   # note navigation to parent directory
git clone https://github.com/ianfixes/arduino_ci
cd arduino_ci
git am .../debug.patch # https://github.com/ianfixes/arduino_ci/issues/69#issuecomment-459180028
popd   # now back in arduino library directory

bundle install --path vendor/bundle        # can also use "--deployment" instead of this
bundle exec arduino_ci_remote.rb --skip-compilation
```

Does that make sense?

hlovdal, 2019-02-04:
Thanks. This makes sense. I have now restructured my project to have its content in a subdirectory in parallel with the arduino_ci submodule and it now works fine using latest master version (with some of my additional changes).

ianfixes, 2019-02-06:
Note to self: add to #114