Arduino-CI/arduino_ci #78: undefined method `binary_path' for nil:NilClass (NoMethodError)

scls19fr, 2019-01-11:
## System

- OS: Mac OS 10.13.5
- `ruby -v`: ruby 2.3.0p0 (2015-12-25 revision 53290) [x86_64-darwin16]
- `bundle -v`: Bundler version 1.16.4
- `bundle info arduino_ci`:
```
  * arduino_ci (0.1.16)
	Summary: Tools for building and unit testing Arduino libraries
	Homepage: http://github.com/ianfixes/arduino_ci
	Path: /Users/scls/.rvm/gems/ruby-2.3.0/gems/arduino_ci-0.1.16
```
- `g++ -v`:
```
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 10.0.0 (clang-1000.11.45.5)
Target: x86_64-apple-darwin17.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
- Arduino IDE version: 1.8.6

- `java -version`:
```
java version "1.8.0_192"
Java(TM) SE Runtime Environment (build 1.8.0_192-b12)
Java HotSpot(TM) 64-Bit Server VM (build 25.192-b12, mixed mode)
```

##  Issue
I'm just trying to run unit test locally on my Mac...
(I'm not trying CI for now)

```
$ bundle exec arduino_ci_remote.rb
Arduino package seems to have been downloaded already
Extracting archive with Zip.....................................................................................................
Arduino package seems to have been installed already
Located Arduino binary... /Users/scls/.rvm/gems/ruby-2.3.0/gems/arduino_ci-0.1.16/exe/arduino_ci_remote.rb:311:in `block in <top (required)>': undefined method `binary_path' for nil:NilClass (NoMethodError)
	from /Users/scls/.rvm/gems/ruby-2.3.0/gems/arduino_ci-0.1.16/exe/arduino_ci_remote.rb:40:in `perform_action'
	from /Users/scls/.rvm/gems/ruby-2.3.0/gems/arduino_ci-0.1.16/exe/arduino_ci_remote.rb:75:in `inform'
	from /Users/scls/.rvm/gems/ruby-2.3.0/gems/arduino_ci-0.1.16/exe/arduino_ci_remote.rb:311:in `<top (required)>'
	from /Users/scls/.rvm/gems/ruby-2.3.0/bin/arduino_ci_remote.rb:23:in `load'
	from /Users/scls/.rvm/gems/ruby-2.3.0/bin/arduino_ci_remote.rb:23:in `<main>'
	from /Users/scls/.rvm/gems/ruby-2.3.0/bin/ruby_executable_hooks:15:in `eval'
	from /Users/scls/.rvm/gems/ruby-2.3.0/bin/ruby_executable_hooks:15:in `<main>'
```

Any idea?

ianfixes, 2019-01-11:
It seems to be having some trouble locating the Arduino executable.  This may be a problem on my part, but I'm slightly confused because it works on my machine (where Arduino is installed) and Travis CI (where `arduino_ci` handles the installation from scratch).

Where is `Arduino.app` installed on your system?

scls19fr, 2019-01-11:
It's at /Applications/Arduino.app/
but after running `bundle exec arduino_ci_remote.rb` there's also Arduino.app in my project directory.

Should I really run:

    bundle exec arduino_ci_remote.rb

to start local testing (without Arduino board connected) ?

scls19fr, 2019-01-11:
```
$ bundle install
Fetching git@github.com:ianfixes/arduino_ci.git
git@github.com: Permission denied (publickey).
fatal: Impossible de lire le dépôt distant.

Veuillez vérifier que vous avez les droits d'accès
et que le dépôt existe.

Retrying `git clone 'git@github.com:ianfixes/arduino_ci.git' "/Users/scls/.rvm/gems/ruby-2.3.0/cache/bundler/git/arduino_ci-b26e28e5a8273bc37e77a4446767a08471a72008" --bare --no-hardlinks --quiet` due to error (2/4): Bundler::Source::Git::GitCommandError Git error: command `git clone 'git@github.com:ianfixes/arduino_ci.git' "/Users/scls/.rvm/gems/ruby-2.3.0/cache/bundler/git/arduino_ci-b26e28e5a8273bc37e77a4446767a08471a72008" --bare --no-hardlinks --quiet` in directory /Users/scls/github/scls19fr/gokart_multilaptimer has failed.git@github.com: Permission denied (publickey).
fatal: Impossible de lire le dépôt distant.

Veuillez vérifier que vous avez les droits d'accès
et que le dépôt existe.

Retrying `git clone 'git@github.com:ianfixes/arduino_ci.git' "/Users/scls/.rvm/gems/ruby-2.3.0/cache/bundler/git/arduino_ci-b26e28e5a8273bc37e77a4446767a08471a72008" --bare --no-hardlinks --quiet` due to error (3/4): Bundler::Source::Git::GitCommandError Git error: command `git clone 'git@github.com:ianfixes/arduino_ci.git' "/Users/scls/.rvm/gems/ruby-2.3.0/cache/bundler/git/arduino_ci-b26e28e5a8273bc37e77a4446767a08471a72008" --bare --no-hardlinks --quiet` in directory /Users/scls/github/scls19fr/gokart_multilaptimer has failed.git@github.com: Permission denied (publickey).
fatal: Impossible de lire le dépôt distant.

Veuillez vérifier que vous avez les droits d'accès
et que le dépôt existe.

Retrying `git clone 'git@github.com:ianfixes/arduino_ci.git' "/Users/scls/.rvm/gems/ruby-2.3.0/cache/bundler/git/arduino_ci-b26e28e5a8273bc37e77a4446767a08471a72008" --bare --no-hardlinks --quiet` due to error (4/4): Bundler::Source::Git::GitCommandError Git error: command `git clone 'git@github.com:ianfixes/arduino_ci.git' "/Users/scls/.rvm/gems/ruby-2.3.0/cache/bundler/git/arduino_ci-b26e28e5a8273bc37e77a4446767a08471a72008" --bare --no-hardlinks --quiet` in directory /Users/scls/github/scls19fr/gokart_multilaptimer has failed.git@github.com: Permission denied (publickey).
fatal: Impossible de lire le dépôt distant.

Veuillez vérifier que vous avez les droits d'accès
et que le dépôt existe.

Git error: command `git clone 'git@github.com:ianfixes/arduino_ci.git'
"/Users/scls/.rvm/gems/ruby-2.3.0/cache/bundler/git/arduino_ci-b26e28e5a8273bc37e77a4446767a08471a72008" --bare --no-hardlinks --quiet` in directory
/Users/scls/github/scls19fr/gokart_multilaptimer has failed.
```

scls19fr, 2019-01-11:
Changing for

source 'https://rubygems.org'
gem 'arduino_ci', :git => "https://github.com/ianfixes/arduino_ci.git", :branch => "2019-01-11_small_fixes"

scls19fr, 2019-01-11:
Running
```
$ bundle exec arduino_ci_remote.rb
```

open quickly my Arduino IDE (I just see splash screen a few seconds) and after

    Attempting to download Arduino package with open-uri

is displayed on console

ianfixes, 2019-01-11:
I bet I know the problem... I'm searching for an english-language string in the output, and it's possible that the Arduino IDE might be writing one in French. 

What's the output when you run this from the console?

```
$ /Applications/Arduino.app/Contents/MacOS/Arduino --bogus-option
```

scls19fr, 2019-01-11:
```
Located Arduino binary... /Users/scls/.rvm/gems/ruby-2.3.0/bundler/gems/arduino_ci-2e13f8755546/exe/arduino_ci_remote.rb:311:in `block in <top (required)>': undefined method `binary_path' for nil:NilClass (NoMethodError)
	from /Users/scls/.rvm/gems/ruby-2.3.0/bundler/gems/arduino_ci-2e13f8755546/exe/arduino_ci_remote.rb:40:in `perform_action'
	from /Users/scls/.rvm/gems/ruby-2.3.0/bundler/gems/arduino_ci-2e13f8755546/exe/arduino_ci_remote.rb:75:in `inform'
	from /Users/scls/.rvm/gems/ruby-2.3.0/bundler/gems/arduino_ci-2e13f8755546/exe/arduino_ci_remote.rb:311:in `<top (required)>'
	from /Users/scls/.rvm/gems/ruby-2.3.0/bin/arduino_ci_remote.rb:23:in `load'
	from /Users/scls/.rvm/gems/ruby-2.3.0/bin/arduino_ci_remote.rb:23:in `<main>'
	from /Users/scls/.rvm/gems/ruby-2.3.0/bin/ruby_executable_hooks:15:in `eval'
	from /Users/scls/.rvm/gems/ruby-2.3.0/bin/ruby_executable_hooks:15:in `<main>'
```

is displayed while Arduino splash screen is shown  again and quickly quit

scls19fr, 2019-01-11:
```
$ /Applications/Arduino.app/Contents/MacOS/Arduino --bogus-option
Erreur: option inconnue : --bogus-option
```

I wonder if/when arduino-cli https://github.com/arduino/arduino-cli will be officially released (not alpha) and used in Arduino IDE... it will probably help a lot

ianfixes, 2019-01-11:
Yup, that was it.  This patch (that I just pushed) should fix it:
https://github.com/ianfixes/arduino_ci/pull/79/commits/95857c830209f986df492020e66acad7115c9e2e

Can you reinstall the bundle and try again?

scls19fr, 2019-01-11:
```
$ bundle install
Using os 1.0.0
Using rubyzip 1.2.2
Using arduino_ci 0.1.16 from https://github.com/ianfixes/arduino_ci.git (at 2019-01-11_small_fixes@2e13f87)
Using bundler 1.16.4
Bundle complete! 1 Gemfile dependency, 4 gems now installed.
Use `bundle info [gemname]` to see where a bundled gem is installed.
✔ ~/github/scls19fr/gokart_multilaptimer [master|…4801]
16:16 $ bundle exec arduino_ci_remote.rb
Arduino package seems to have been downloaded already
Arduino package seems to have been extracted already
Arduino package seems to have been installed already
Located Arduino binary... bundler: failed to load command: arduino_ci_remote.rb (/Users/scls/.rvm/gems/ruby-2.3.0/bin/arduino_ci_remote.rb)
NoMethodError: undefined method `binary_path' for nil:NilClass
  /Users/scls/.rvm/gems/ruby-2.3.0/bundler/gems/arduino_ci-2e13f8755546/exe/arduino_ci_remote.rb:311:in `block in <top (required)>'
  /Users/scls/.rvm/gems/ruby-2.3.0/bundler/gems/arduino_ci-2e13f8755546/exe/arduino_ci_remote.rb:40:in `perform_action'
  /Users/scls/.rvm/gems/ruby-2.3.0/bundler/gems/arduino_ci-2e13f8755546/exe/arduino_ci_remote.rb:75:in `inform'
  /Users/scls/.rvm/gems/ruby-2.3.0/bundler/gems/arduino_ci-2e13f8755546/exe/arduino_ci_remote.rb:311:in `<top (required)>'
  /Users/scls/.rvm/gems/ruby-2.3.0/bin/arduino_ci_remote.rb:23:in `load'
  /Users/scls/.rvm/gems/ruby-2.3.0/bin/arduino_ci_remote.rb:23:in `<top (required)>'
```

scls19fr, 2019-01-11:
Sorry @ianfixes but I won't be able to test this more today (because of my schedule).

ianfixes, 2019-01-11:
It looks like your bundle install didn't grab the latest ref: `2019-01-11_small_fixes@2e13f87` should say `2019-01-11_small_fixes@95857c8` (the `95857c8` I posted above).  

You could put the ref right in the Gemfile
```ruby
gem 'arduino_ci', git: "https://github.com/ianfixes/arduino_ci.git", branch: "2019-01-11_small_fixes", ref: "95857c830209f986df492020e66acad7115c9e2e"
```

scls19fr, 2019-01-11:
Ok thanks sorry about my mistake... I will try tomorrow

scls19fr, 2019-01-12:
Arduino.app is now found and previous error seems to be fixed but
it seems that my tests aren't found.

```
09:02 $ tree
.
├── Gemfile
├── Gemfile.lock
├── README.md
└── multilaptimer
    ├── config.h
    ├── debug.h
    ├── laptimer.cpp
    ├── laptimer.h
    ├── mode.h
    ├── multilaptimer.ino
    ├── page.h
    ├── pinout.h
    ├── test
    │   └── test_laptimer.cpp
    ├── utils.cpp
    └── utils.h

2 directories, 14 files
✔ ~/github/scls19fr/gokart_multilaptimer [master|…3]
09:03 $ cat multilaptimer/test/test_laptimer.cpp
#include <ArduinoUnitTests.h>
#include "../laptimer.h"

unittest(your_test_name)
{
  assertEqual(4, doSomething());
}

unittest_main()
✔ ~/github/scls19fr/gokart_multilaptimer [master|…3]
09:03 $ bundle exec arduino_ci_remote.rb
Located Arduino binary...                                /Users/scls/Arduino.app
The set of compilers (1) isn't empty...                                        ✓
Checking g++ version...
    Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
    Apple LLVM version 10.0.0 (clang-1000.11.45.5)
    Target: x86_64-apple-darwin17.7.0
    Thread model: posix
    InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
...Checking g++ version                                                        ✓
libasan availability for g++...                                             true
Requested unittest platform 'uno' is defined in 'platforms' YML...             ✓
Requested unittest platform 'due' is defined in 'platforms' YML...             ✓
Requested unittest platform 'zero' is defined in 'platforms' YML...            ✓
Requested unittest platform 'leonardo' is defined in 'platforms' YML...        ✓
Skipping unit tests; no tests dir at test...
  In case that's an error, this is what was found in the library:
    Gemfile
    Gemfile.lock
    README.md
    multilaptimer/config.h
    multilaptimer/debug.h
    multilaptimer/laptimer.cpp
    multilaptimer/laptimer.h
    multilaptimer/mode.h
    multilaptimer/multilaptimer.ino
    multilaptimer/page.h
    multilaptimer/pinout.h
    multilaptimer/test/test_laptimer.cpp
    multilaptimer/utils.cpp
    multilaptimer/utils.h
...Skipping unit tests; no tests dir at test
Indexing libraries...                                                          ✓
Installing library under test...                                               ✓
Library installed at... /Users/scls/Documents/Arduino/libraries/gokart_multilaptimer
Skipping builds; no examples found in /Users/scls/Documents/Arduino/libraries/gokart_multilaptimer...
    /Users/scls/github/scls19fr/gokart_multilaptimer/Gemfile
    /Users/scls/github/scls19fr/gokart_multilaptimer/Gemfile.lock
    /Users/scls/github/scls19fr/gokart_multilaptimer/README.md
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/config.h
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/debug.h
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/laptimer.cpp
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/laptimer.h
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/mode.h
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/multilaptimer.ino
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/page.h
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/pinout.h
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/test/test_laptimer.cpp
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/utils.cpp
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/utils.h
...Skipping builds; no examples found in /Users/scls/Documents/Arduino/libraries/gokart_multilaptimer
Failures: 0
```

scls19fr, 2019-01-12:
I would like to run unit test for Arduino Mega 2560 but I'm not sure what should be in `.arduino-ci.yml`

I've tried `mega` and also `mega2560`

with `mega`
```
10:39 $ cat .arduino-ci.yml
unittest:
  platforms:
    - mega

compile:
  platforms:
    - mega
✔ ~/github/scls19fr/gokart_multilaptimer [master|✚ 1…3]
10:39 $ bundle exec arduino_ci_remote.rb
Located Arduino binary...                                /Users/scls/Arduino.app
The set of compilers (1) isn't empty...                                        ✓
Checking g++ version...
    Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
    Apple LLVM version 10.0.0 (clang-1000.11.45.5)
    Target: x86_64-apple-darwin17.7.0
    Thread model: posix
    InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
...Checking g++ version                                                        ✓
libasan availability for g++...                                             true
Requested unittest platform 'mega' is defined in 'platforms' YML...            ✗
This may indicate a problem with ArduinoCI, or your configuration
Failures: 1
Last message:  $  java -cp /Users/scls/Arduino.app/Contents/Java/* -DAPP_DIR=/Users/scls/Arduino.app/Contents/Java -Dfile.encoding=UTF-8 -Dapple.awt.UIElement=true -Xms128M -Xmx512M processing.app.Base --get-pref
========== Stdout:
Bibliothèque non valide trouvée dans /Users/scls/Documents/Arduino/libraries/gokart_multilaptimer : /Users/scls/Documents/Arduino/libraries/gokart_multilaptimer
Bibliothèque non valide trouvée dans /Users/scls/Documents/Arduino/libraries/gokart_multilaptimer : /Users/scls/Documents/Arduino/libraries/gokart_multilaptimer
#PREFDUMP#
last.sketch.count=1
update.check=true
platform.auto_file_type_associations=true
gui.scale=auto
editor.window.width.default=500
editor.window.height.default=600
editor.window.width.min=400
editor.window.height.min=290
editor.code_folding=false
editor.font=Monaco,plain,12
editor.antialias=true
editor.external=false
editor.caret.blink=true
editor.invalid=false
editor.linenumbers=false
editor.keys.alternative_cut_copy_paste=false
editor.keys.shift_backspace_is_delete=true
editor.keys.home_and_end_beginning_end_of_doc=false
console=true
console.output.file=stdout.txt
console.error.file=stderr.txt
console.lines=4
console.auto_clear=true
console.length=500
editor.tabs.expand=true
editor.tabs.size=2
editor.indent=true
editor.auto_close_braces=true
editor.divider.size=0
editor.save_on_verify=true
run.options=
run.options.memory=false
run.options.memory.initial=64
run.options.memory.maximum=256
run.display=1
export.application.platform=true
[... 261 lines cut ...]
    /Users/scls/github/scls19fr/gokart_multilaptimer/Gemfile
    /Users/scls/github/scls19fr/gokart_multilaptimer/Gemfile.lock
    /Users/scls/github/scls19fr/gokart_multilaptimer/README.md
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/config.h
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/debug.h
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/laptimer.cpp
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/laptimer.h
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/mode.h
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/multilaptimer.ino
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/page.h
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/pinout.h
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/utils.cpp
    /Users/scls/github/scls19fr/gokart_multilaptimer/multilaptimer/utils.h
    /Users/scls/github/scls19fr/gokart_multilaptimer/test/test_laptimer.cpp
    /Users/scls/github/scls19fr/gokart_multilaptimer/unittest_test_laptimer.cpp.bin.dSYM/Contents/Info.plist
    /Users/scls/github/scls19fr/gokart_multilaptimer/unittest_test_laptimer.cpp.bin.dSYM/Contents/Resources/DWARF/unittest_test_laptimer.cpp.bin
...Skipping builds; no examples found in /Users/scls/Documents/Arduino/libraries/gokart_multilaptimer
Failures: 1
```

This is very clear... it's failing but I don't understand why

    error: use of undeclared identifier 'Serial'

Opening `multilaptimer/multilaptimer.ino` in Arduino IDE, I can compile this code.

ianfixes, 2019-01-12:
In that case, #79 fixes this and I will close this issue.

As for the other questions:

## Including library-under-test from unit test CPP file

```
09:24 $ tree
.
├── Gemfile
├── Gemfile.lock
├── README.md
├── multilaptimer
│   ├── config.h
│   ├── debug.h
│   ├── laptimer.cpp
│   ├── laptimer.h
│   ├── mode.h
│   ├── multilaptimer.ino
│   ├── page.h
│   ├── pinout.h
│   ├── utils.cpp
│   └── utils.h
└── test
    └── test_laptimer.cpp

2 directories, 14 files
✔ ~/github/scls19fr/gokart_multilaptimer [master|…3]
09:24 $ cat test/test_laptimer.cpp
#include <ArduinoUnitTests.h>
#include "../laptimer.h"
//#include "multilaptimer/laptimer.h"  // doesn't work also
```

Based on this tree, the proper way to include `laptimer.h` is `#include "../multilaptimer/laptimer.h"`.

## `mega` and `mega2650`

See the section in REFERENCE.md about defining your own platforms:
https://github.com/ianfixes/arduino_ci/blob/master/REFERENCE.md#overriding-default-build-behavior

The ones included by default are here:
https://github.com/ianfixes/arduino_ci/blob/master/misc/default.yml

You can put a `package:` and `platforms:` section in your own YAML file (with whatever the definitions of those 2 boards are), and that will add to the default set of boards.

## warning "device type not defined"

This could be due to improper configuration of the `mega2650` board in your `.arduino-ci.yaml`.  Usually the use of these boards creates a particular preprocessor flag (like [`__AVR_ATmega328P__`, shown here](https://github.com/ianfixes/arduino_ci/blob/master/misc/default.yml#L24)), and this flag will trigger the definition of all board properties -- up to and including things like serial port objects.  That's the first place I'd look.

closed 2019-01-12 by ianfixes

scls19fr, 2019-01-12:
Thanks a lot @ianfixes 
I feel I'm becoming an arduino_ci addict!

PR #79 (2019-01-12): Small fixes
## Highlights from `CHANGELOG.md`
- Simplified the use of `Array.each` with a return statement; it's now simply `Array.find`
- `autolocate!` for Arduino installations now raises `ArduinoInstallationError` if `force_install` fails
- Errors due to missing YAML are now named `ConfigurationError`
- typo in documentation fixed
* See CHANGELOG.md for more

* Fixes #78