Arduino-CI/arduino_ci #85: This may indicate a problem with ArduinoCI, or your configuration Failures: 4

scls19fr, 2019-01-14:
Hello,

I'm trying Arduino_CI on a different system (at work)...

I can't really understand what's going on...
(installing all this is a pain with Windows...)

Here is my install procedure:

1) Install Ruby (I choosed https://rubyinstaller.org/downloads/ Ruby+Devkit 2.5.3-1 (x64))

2) Install MinGW http://www.mingw.org/ to have gcc / g++

3) Add C:\MinGW\bin to PATH to have g++ available from any directory

4) gem install --http-proxy http://cache.mydomain.fr:3128 bundler

5) cd my_project

6) bundle install

(after setting http_proxy variable... which is a bit different depending if you are doing this on Git Bash or on Windows command terminal cmd)

7) bundle exec arduino_ci_remote.rb

## System

> Feel free to delete this section if you're submitting a feature request

- OS: Windows 7
- `ruby -v`:

    ruby 2.5.3p105 (2018-10-18 revision 65156) [x64-mingw32]

- `bundle -v`:

Bundler version 2.0.1

- `bundle info arduino_ci`:

```
  * arduino_ci (0.1.16)
        Summary: Tools for building and unit testing Arduino libraries
        Homepage: http://github.com/ianfixes/arduino_ci
        Path: C:/Ruby25-x64/lib/ruby/gems/2.5.0/gems/arduino_ci-0.1.16
```

- `g++ -v`:

```
$ g++ -v
Using built-in specs.
COLLECT_GCC=C:\MinGW\bin\g++.exe
COLLECT_LTO_WRAPPER=c:/mingw/bin/../libexec/gcc/mingw32/8.2.0/lto-wrapper.exe
Target: mingw32
Configured with: ../src/gcc-8.2.0/configure --build=x86_64-pc-linux-gnu --host=mingw32 --target=mingw32 --prefix=/mingw --disable-win32-registry --with-arch=i586 --with-tune=generic --enable-languages=c,c++,objc,obj-c++,fortran,ada --with-pkgversion='MinGW.org GCC-8.2.0-1' --with-gmp=/mingw --with-mpfr=/mingw --with-mpc=/mingw --enable-static --enable-shared --enable-threads --with-dwarf2 --disable-sjlj-exceptions --enable-version-specific-runtime-libs --with-libiconv-prefix=/mingw --with-libintl-prefix=/mingw --enable-libstdcxx-debug --with-isl=/mingw --enable-libgomp --disable-libvtv --enable-nls --disable-build-format-warnings
Thread model: win32
gcc version 8.2.0 (MinGW.org GCC-8.2.0-1)
```

- Arduino IDE version: 1.8.8
- `java -version`: can't find java executable... but Arduino IDE is running (fine)

## (Issue Summary / Feature Request)

```
$ bundle exec arduino_ci_remote.rb
Located Arduino binary...       C:\Program Files (x86)\Arduino/arduino_debug.exe
The set of compilers (1) isn't empty...                                        ✓
Checking g++ version...
[... 344 lines cut ...]
T▒l▒chargement du catalogue de biblioth▒ques... T▒l▒charg▒: 342kb sur 741kb.
T▒l▒chargement du catalogue de biblioth▒ques... T▒l▒charg▒: 369kb sur 741kb.
T▒l▒chargement du catalogue de biblioth▒ques... T▒l▒charg▒: 392kb sur 741kb.
T▒l▒chargement du catalogue de biblioth▒ques... T▒l▒charg▒: 418kb sur 741kb.
T▒l▒chargement du catalogue de biblioth▒ques... T▒l▒charg▒: 441kb sur 741kb.
T▒l▒chargement du catalogue de biblioth▒ques... T▒l▒charg▒: 467kb sur 741kb.
T▒l▒chargement du catalogue de biblioth▒ques... T▒l▒charg▒: 490kb sur 741kb.
T▒l▒chargement du catalogue de biblioth▒ques... T▒l▒charg▒: 517kb sur 741kb.
T▒l▒chargement du catalogue de biblioth▒ques... T▒l▒charg▒: 539kb sur 741kb.
T▒l▒chargement du catalogue de biblioth▒ques... T▒l▒charg▒: 566kb sur 741kb.
T▒l▒chargement du catalogue de biblioth▒ques... T▒l▒charg▒: 588kb sur 741kb.
T▒l▒chargement du catalogue de biblioth▒ques... T▒l▒charg▒: 615kb sur 741kb.
T▒l▒chargement du catalogue de biblioth▒ques... T▒l▒charg▒: 637kb sur 741kb.
T▒l▒chargement du catalogue de biblioth▒ques... T▒l▒charg▒: 664kb sur 741kb.
T▒l▒chargement du catalogue de biblioth▒ques... T▒l▒charg▒: 687kb sur 741kb.
T▒l▒chargement du catalogue de biblioth▒ques... T▒l▒charg▒: 713kb sur 741kb.
T▒l▒chargement du catalogue de biblioth▒ques... T▒l▒charg▒: 736kb sur 741kb.
Mise a jour de la liste des biblioth▒que install▒e
Library is already installed: USBHost:1.0.5
========== Stderr:
Chargement de la configuration
Initialisation des paquets...
Pr▒paration des cartes
Using proxy HTTP @ cache.my-domain.fr:3128
```

ianfixes, 2019-01-14:
Unfortunately my only ability to test on windows comes from AppVeyor CI.  Does replicating the steps that are present in the [AppVeyor config](https://github.com/ianfixes/arduino_ci/blob/master/appveyor.yml) help at all?

ianfixes, 2019-01-14:
I think it's likely that this related to the need for 64-bit compilation https://stackoverflow.com/q/22663897/2063546

ianfixes, 2019-01-14:
I don't have a good answer, and I'm happy to leave this open in the hopes that someone else can answer.  Both `g++` and Ruby environments aren't native to Windows, which (I suspect) is 90% of the difficulty in setting this up.

ianfixes, 2019-01-14:
It might be that everything is contingent on having `cygwin64`, but that's a high-level guess.

scls19fr, 2019-01-14:
Removing MinGW... installing Cygwin64 and adding it's bin dir to PATH lead to side effect...
now Git Bash 

```
      2 [main] getent (6368) C:\cygwin64\bin\getent.exe: *** fatal error - cygheap base mismatch detected - 0x1802FF408/0x18031D410.
This problem is probably due to using incompatible versions of the cygwin DLL.
Search for cygwin1.dll using the Windows Start->Find/Search facility
and delete all but the most recent version.  The most recent version *should*
reside in x:\cygwin\bin, where 'x' is the drive on which you have
installed the cygwin distribution.  Rebooting is also suggested if you
are unable to find another cygwin DLL.
g
$ gcc
      2 [main] gcc (4972) C:\cygwin64\bin\gcc.exe: *** fatal error - cygheap base mismatch detected - 0x1802FF408/0x18031D410.
This problem is probably due to using incompatible versions of the cygwin DLL.
Search for cygwin1.dll using the Windows Start->Find/Search facility
and delete all but the most recent version.  The most recent version *should*
reside in x:\cygwin\bin, where 'x' is the drive on which you have
installed the cygwin distribution.  Rebooting is also suggested if you
are unable to find another cygwin DLL.
Segmentation fault
```

I know why I hate developing on Windows.

And with cmd

```
    C:\Users\scelles\Documents\Arduino/libraries/Bridge/src/HttpClient.h
    C:\Users\scelles\Documents\Arduino/libraries/Bridge/src/Mailbox.cpp
    C:\Users\scelles\Documents\Arduino/libraries/Bridge/src/Mailbox.h
    C:\Users\scelles\Documents\Arduino/libraries/Bridge/src/Process.cpp
    C:\Users\scelles\Documents\Arduino/libraries/Bridge/src/Process.h
    C:\Users\scelles\Documents\Arduino/libraries/Bridge/src/YunClient.h
    C:\Users\scelles\Documents\Arduino/libraries/Bridge/src/YunServer.h
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/AUTHORS
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/README.adoc
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/examples/AdvancedChatServer/AdvancedChatServer.ino
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/examples/BarometricPressureWebServer/BarometricPressureWebServer.ino
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/examples/ChatServer/ChatServer.ino
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/examples/DhcpAddressPrinter/DhcpAddressPrinter.ino
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/examples/DhcpChatServer/DhcpChatServer.ino
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/examples/LinkStatus/LinkStatus.ino
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/examples/TelnetClient/TelnetClient.ino
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/examples/UDPSendReceiveString/UDPSendReceiveString.ino
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/examples/UdpNtpClient/UdpNtpClient.ino
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/examples/WebClient/WebClient.ino
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/examples/WebClientRepeating/WebClientRepeating.ino
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/examples/WebServer/WebServer.ino
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/keywords.txt
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/library.properties
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/src/Dhcp.cpp
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/src/Dhcp.h
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/src/Dns.cpp
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/src/Dns.h
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/src/Ethernet.cpp
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/src/Ethernet.h
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/src/EthernetClient.cpp
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/src/EthernetClient.h
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/src/EthernetServer.cpp
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/src/EthernetServer.h
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/src/EthernetUdp.cpp
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/src/EthernetUdp.h
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/src/socket.cpp
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/src/utility/w5100.cpp
    C:\Users\scelles\Documents\Arduino/libraries/Ethernet/src/utility/w5100.h
    C:\Users\scelles\Documents\Arduino/libraries/Firmata/Boards.h
    C:\Users\scelles\Documents\Arduino/libraries/Firmata/Firmata.cpp
    C:\Users\scelles\Documents\Arduino/libraries/Firmata/Firmata.h
    C:\Users\scelles\Documents\Arduino/libraries/Firmata/FirmataConstants.h
[... 253 lines cut ...]
Initialisation des paquets...
PrÚparation des cartes
Using proxy HTTP @ cache.domain.fr:3128

U:\github\scls19fr\myproject>
```

and with Cygwin Terminal

```
$ bundle exec arduino_ci_remote.rb
C:\Ruby25-x64\bin\ruby.exe: No such file or directory -- /cygdrive/c/Ruby25-x64/bin/bundle (LoadError)

scelles@GTE-ENS-SC /cygdrive/u/github/scls19fr/myproject
$ bundle install
C:\Ruby25-x64\bin\ruby.exe: No such file or directory -- /cygdrive/c/Ruby25-x64/bin/bundle (LoadError)

scelles@GTE-ENS-SC /cygdrive/u/github/scls19fr/myproject
$ bundle exec arduino_ci_remote.rb
C:\Ruby25-x64\bin\ruby.exe: No such file or directory -- /cygdrive/c/Ruby25-x64/bin/bundle (LoadError)

scelles@GTE-ENS-SC /cygdrive/u/github/scls19fr/myproject
```

it messed a lot of thing... or a least, a lot of care is required to have a usable test driven development Arduino environment with Windows !

ianfixes, 2019-01-15:
If you can get me the full output of that test you did with `cmd` (please upload it as a [gist](https://gist.github.com/)), I think there might be a bug related to library installation that you are seeing -- and it might be an easy fix.

ianfixes, 2019-01-15:
The `2019-01-15_small_fixes` branch in #90 _might_ resolve this, but that's only a guess.