Arduino-CI/arduino_ci #86: arduino_ci doesn't seems to work properly with mega2560

scls19fr, 2019-01-14:
Hello,

Here is a minimal (non) working example
https://github.com/scls19fr/mwe_arduino_ci

to show problem I was facing (see https://github.com/ianfixes/arduino_ci/issues/78#issuecomment-453776168 ) with Arduino Mega 2560.

Kind regards

PS : arduino_ci log finish with Failures: 1
It may be interesting to show how many tests / examples are being run...
because showing only Failures: 0 can be quite misleading... it can simply be because tests / examples haven't been found

ianfixes, 2019-01-14:
It looks like I added a definition for this board, but it's incomplete -- it's missing the compiler flags that are needed for this compilation target.  e.g. here's what the definition for `leonardo` looks like in the [default yaml file](https://github.com/ianfixes/arduino_ci/blob/master/misc/default.yml#L62-L70):

```yml
  leonardo:
    board: arduino:avr:leonardo
    package: arduino:avr
    gcc:
      features:
      defines:
        - __AVR_ATmega32U4__
      warnings:
      flags:
```

That `__AVR_ATmega32U4__` is the crucial value, and I don't know what it is (or if there are several, what they all should be).  

It appears that not much more is known now than when I first went looking; here are a bunch of search results that just turned up people with the same problem as me:

* "Error compiling for --fqbn arduino:avr:mega (missing cpu flags)" https://github.com/arduino/arduino-cli/issues/17
* "Compiler error: Unknown architecture on mega2560" https://www.avrfreaks.net/forum/compiler-error-unknown-architecture-mega2560
* "Which GCC + avr-libc + arduino for mega2560 actually works?" https://forum.arduino.cc/index.php?topic=66243.0
* "AVR-ATmega2560-GCC/Makefile" https://github.com/matteoserva/ChibiOS-AVR/blob/master/demos/AVR-ATmega2560-GCC/Makefile
* "AVR-GCC with ATMega2560" https://www.avrfreaks.net/forum/avr-gcc-atmega2560-bootloader
* "Do not buy the mega 2560 until gcc is fixed!!" http://forum.arduino.cc/index.php?topic=28269.0

This is a topic I am entirely unqualified to answer; all the existing stuff in the default YAML file came from values I found in other documentation or source files.

If you want to experiment with it (perhaps just using `__AVR_ATmega2560__`, shown in [this gist](https://gist.github.com/ah01/762576) , I'd copy the following section into your `.arduino-ci.yml` file and let me know if it works:

```yml
  mega2560:
    board: arduino:avr:mega:cpu=atmega2560
    package: arduino:avr
    gcc:
      features:
      defines:
        - __AVR_ATmega2560__
      warnings:
      flags:
```

per1234, 2019-01-14:
Here's a list of all the MCU definition macros for AVR:
https://www.microchip.com/webdoc/avrlibcreferencemanual/using_tools_1using_avr_gcc_mach_opt.html
These are widely used in preprocessor conditionals for part-specific code.

scls19fr, 2019-01-15:
Here is my `.arduino-ci.yml` https://github.com/scls19fr/mwe_arduino_ci/blob/master/.arduino-ci.yml
and build status on Travis https://travis-ci.com/scls19fr/mwe_arduino_ci/builds/97342356

ianfixes, 2019-01-15:
Looks like you're missing the toplevel `platforms:` section.  It should look like this:

```yml
platforms:
  mega2560:
    board: arduino:avr:mega:cpu=atmega2560
    package: arduino:avr
    gcc:
      defines:
        - __AVR_ATmega2560__
```

ianfixes, 2019-01-15:
@per1234 that's exactly what I'm looking for!  If you have a list of all known Arduino platforms, I could really use your help on #89.

ianfixes, 2019-01-15:
I see a passing build 🎉 https://travis-ci.com/scls19fr/mwe_arduino_ci/builds/97377931

PR #90 (2019-01-15): Small fixes
## Highlights from `CHANGELOG.md`

* See CHANGELOG.md for more

## Issues Fixed

* Fixes #87 
* Fixes #86 (based on https://github.com/scls19fr/mwe_arduino_ci/runs/50176240)
closed 2019-01-15 by ianfixes