Arduino-CI/arduino_ci #89: Add or update board definitions -- arduino mega at least

ianfixes, 2019-01-14:
Add new board definitions based on [this gist](https://gist.github.com/ah01/762576)

```c++
  #if defined(__AVR_ATmega168__)
    // Arduino Diecimila and older
    LINE("__AVR_ATmega168__", __AVR_ATmega168__);
  #endif
  
  #if defined(__AVR_ATmega328P__)
    // Arduino Duemilanove and Uno
    LINE("__AVR_ATmega328P__", __AVR_ATmega328P__);
  #endif
  
  #if defined(__AVR_ATmega2560__)
    // Arduino Mega 2560
    LINE("__AVR_ATmega2560__", __AVR_ATmega2560__);
  #endif
  
  #if defined(__AVR_ATmega1280__)
    // Arduino Mega
    LINE("__AVR_ATmega1280__", __AVR_ATmega1280__);
  #endif
  
  #if defined(__AVR_ATmega32U4__)
    // Arduino Leonardo
    LINE("__AVR_ATmega32U4__", __AVR_ATmega32U4__);
  #endif
  
  #if defined(__SAM3X8E__)
    // Arduino Due
    LINE("__SAM3X8E__", __SAM3X8E__);
  #endif
  
  // Determine board type
  #if defined(__AVR_ATmega168__) || defined(__AVR_ATmega328P__)
    Serial.println("Regular Arduino");
  #elif defined(__AVR_ATmega2560__) || defined(__AVR_ATmega1280__)
    Serial.println("Arduino Mega"); 
  #elif defined(__AVR_ATmega32U4__)
    Serial.println("Arduino Leonardo");
  #elif defined(__SAM3X8E__)
    Serial.println("Arduino Due");
  #else
    Serial.println("Unknown");
  #endif
```

ianfixes, 2019-01-15:
from https://www.microchip.com/webdoc/avrlibcreferencemanual/using_tools_1using_avr_gcc_mach_opt.html

via @per1234 (https://github.com/ianfixes/arduino_ci/issues/86#issuecomment-454173396) 

Architecture | MCU name | Macro
-- | -- | --
avr1 | at90s1200 | __AVR_AT90S1200__
avr1 | attiny11 | __AVR_ATtiny11__
avr1 | attiny12 | __AVR_ATtiny12__
avr1 | attiny15 | __AVR_ATtiny15__
avr1 | attiny28 | __AVR_ATtiny28__
avr2 | at90s2313 | __AVR_AT90S2313__
avr2 | at90s2323 | __AVR_AT90S2323__
avr2 | at90s2333 | __AVR_AT90S2333__
avr2 | at90s2343 | __AVR_AT90S2343__
avr2 | attiny22 | __AVR_ATtiny22__
avr2 | attiny26 | __AVR_ATtiny26__
avr2 | at90s4414 | __AVR_AT90S4414__
avr2 | at90s4433 | __AVR_AT90S4433__
avr2 | at90s4434 | __AVR_AT90S4434__
avr2 | at90s8515 | __AVR_AT90S8515__
avr2 | at90c8534 | __AVR_AT90C8534__
avr2 | at90s8535 | __AVR_AT90S8535__
avr2/avr25[1] | at86rf401 | __AVR_AT86RF401__
avr2/avr25[1] | ata6289 | __AVR_ATA6289__
avr2/avr25[1] | ata5272 | __AVR_ATA5272__
avr2/avr25[1] | ata6616c | __AVR_ATA6616C__
avr2/avr25[1] | attiny13 | __AVR_ATtiny13__
avr2/avr25[1] | attiny13a | __AVR_ATtiny13A__
avr2/avr25[1] | attiny2313 | __AVR_ATtiny2313__
avr2/avr25[1] | attiny2313a | __AVR_ATtiny2313A__
avr2/avr25[1] | attiny24 | __AVR_ATtiny24__
avr2/avr25[1] | attiny24a | __AVR_ATtiny24A__
avr2/avr25[1] | attiny25 | __AVR_ATtiny25__
avr2/avr25[1] | attiny261 | __AVR_ATtiny261__
avr2/avr25[1] | attiny261a | __AVR_ATtiny261A__
avr2/avr25[1] | attiny4313 | __AVR_ATtiny4313__
avr2/avr25[1] | attiny43u | __AVR_ATtiny43U__
avr2/avr25[1] | attiny44 | __AVR_ATtiny44__
avr2/avr25[1] | attiny44a | __AVR_ATtiny44A__
avr2/avr25[1] | attiny441 | __AVR_ATtiny441__
avr2/avr25[1] | attiny45 | __AVR_ATtiny45__
avr2/avr25[1] | attiny461 | __AVR_ATtiny461__
avr2/avr25[1] | attiny461a | __AVR_ATtiny461A__
avr2/avr25[1] | attiny48 | __AVR_ATtiny48__
avr2/avr25[1] | attiny828 | __AVR_ATtiny828__
avr2/avr25[1] | attiny84 | __AVR_ATtiny84__
avr2/avr25[1] | attiny84a | __AVR_ATtiny84A__
avr2/avr25[1] | attiny841 | __AVR_ATtiny841__
avr2/avr25[1] | attiny85 | __AVR_ATtiny85__
avr2/avr25[1] | attiny861 | __AVR_ATtiny861__
avr2/avr25[1] | attiny861a | __AVR_ATtiny861A__
avr2/avr25[1] | attiny87 | __AVR_ATtiny87__
avr2/avr25[1] | attiny88 | __AVR_ATtiny88__
avr3 | atmega603 | __AVR_ATmega603__
avr3 | at43usb355 | __AVR_AT43USB355__
avr3/avr31[3] | atmega103 | __AVR_ATmega103__
avr3/avr31[3] | at43usb320 | __AVR_AT43USB320__
avr3/avr35[2] | at90usb82 | __AVR_AT90USB82__
avr3/avr35[2] | at90usb162 | __AVR_AT90USB162__
avr3/avr35[2] | ata5505 | __AVR_ATA5505__
avr3/avr35[2] | ata6617c | __AVR_ATA6617C__
avr3/avr35[2] | ata664251 | __AVR_ATA664251__
avr3/avr35[2] | atmega8u2 | __AVR_ATmega8U2__
avr3/avr35[2] | atmega16u2 | __AVR_ATmega16U2__
avr3/avr35[2] | atmega32u2 | __AVR_ATmega32U2__
avr3/avr35[2] | attiny167 | __AVR_ATtiny167__
avr3/avr35[2] | attiny1634 | __AVR_ATtiny1634__
avr3 | at76c711 | __AVR_AT76C711__
avr4 | ata6285 | __AVR_ATA6285__
[... 162 lines cut ...]
avrxmega6 | atxmega128b3 | __AVR_ATxmega128B3__
avrxmega6 | atxmega128c3 | __AVR_ATxmega128C3__
avrxmega6 | atxmega128d3 | __AVR_ATxmega128D3__
avrxmega6 | atxmega128d4 | __AVR_ATxmega128D4__
avrxmega6 | atxmega192a3 | __AVR_ATxmega192A3__
avrxmega6 | atxmega192a3u | __AVR_ATxmega192A3U__
avrxmega6 | atxmega192c3 | __AVR_ATxmega192C3__
avrxmega6 | atxmega192d3 | __AVR_ATxmega192D3__
avrxmega6 | atxmega256a3 | __AVR_ATxmega256A3__
avrxmega6 | atxmega256a3u | __AVR_ATxmega256A3U__
avrxmega6 | atxmega256a3b | __AVR_ATxmega256A3B__
avrxmega6 | atxmega256a3bu | __AVR_ATxmega256A3BU__
avrxmega6 | atxmega256c3 | __AVR_ATxmega256C3__
avrxmega6 | atxmega256d3 | __AVR_ATxmega256D3__
avrxmega6 | atxmega384c3 | __AVR_ATxmega384C3__
avrxmega6 | atxmega384d3 | __AVR_ATxmega384D3__
avrxmega7 | atxmega128a1 | __AVR_ATxmega128A1__
avrxmega7 | atxmega128a1u | __AVR_ATxmega128A1U__
avrxmega7 | atxmega128a4u | __AVR_ATxmega128A4U__
avrtiny10 | attiny4 | __AVR_ATtiny4__
avrtiny10 | attiny5 | __AVR_ATtiny5__
avrtiny10 | attiny9 | __AVR_ATtiny9__
avrtiny10 | attiny10 | __AVR_ATtiny10__
avrtiny10 | attiny20 | __AVR_ATtiny20__
avrtiny10 | attiny40 | __AVR_ATtiny40__

ianfixes, 2019-01-15:
I guess I already had a bunch of this via https://github.com/ianfixes/arduino_ci/blob/master/cpp/arduino/avr/io.h ... the trick is correlating them all to Arduino platforms

scls19fr, 2019-01-18:
I bought and received recently an Arduino MKRZero which is using a SAMD21 Cortex-M0+ 32bit low power ARM MCU. Any chance to have this MCU supported by arduino_ci ?

Here is the list of Arduino boards (from https://www.arduino.cc/en/products.compare )

Board | MCU | Arduino CI support (1) | Platform name (2)
-- | -- | -- | --
Genuino101 | Intel® Curie | ❌
Gemma | ATtiny85 | ✅ | gemma
LilyPad 168 | ATmega168V | ❌
LilyPad 328 | ATmega328P | ❌
LilyPad SimpleSnap | ATmega328P | ❌
LilyPad USB | ATmega32U4 | ❌
Mega 2560 | ATmega2560 | ✅ | mega2560
Micro | ATmega32U4 | ❌
MKR1000 | SAMD21 Cortex-M0+ | ❌
Pro 168 | ATmega168 | ❌
Pro 328 | ATmega328P | ❌
Pro Mini | ATmega328P | ❌
Uno | ATmega328P |  ✅ | uno
Zero | ATSAMD21G18 |  ✅ | zero
Due | ATSAM3X8E |  ✅ | due
Esplora | ATmega32U4 | ❌
Ethernet | ATmega328P | ❌
Leonardo | ATmega32U4 | ✅ | leonardo
Mega ADK | ATmega2560 | ❌
Mini | ATmega328P | ❌
Nano 168 | ATmega168 | ❌
Nano 328 | ATmega328P | ❌
Yùn 32 | ATmega32U4 | ❌
Yùn 9331 | AR9331 Linux | ❌
Arduino Robot | ATmega32u4 | ❌
MKRZero | SAMD21 Cortex-M0+ 32bit low   power ARM MCU | ❌

Others boards

Board | MCU | Arduino CI support (1) | Platform name (2)
-- | -- | -- | --
ESP32 | ESP32 |  ✅ | esp32
ESP8266 | ESP8266 | ✅ | esp8266
Trinket | | ✅ | trinket
m4 | SAMD51 | ✅ | m4
Circuit Playground Classic | ATmega32u4 | ✅ | cplayClassic
Circuit Playground Express | ATSAMD21 ARM Cortex M0 | ✅ | cplayExpress

(1) currently defined in [`misc/default.yml`](https://github.com/ianfixes/arduino_ci/blob/master/misc/default.yml)... to be tested
(2) name of the platform in Arduino CI

ianfixes, 2019-01-28:
Do these platforms have official identifiers (something that can be used  in code, like `promini` or `pro_mini` for the "Pro Mini")?  That's the missing piece here.  I can (of course) invent my own IDs, I just don't want to create accidental conflicts like the 2 "Pro Mini" possibilities above.

scls19fr, 2019-01-28:
Maybe @per1234 can tell us if Arduino platforms have official identifiers.

ianfixes, 2019-01-28:
Note to self, combine with this info from #7 

This looks like a good resource for what `#define`s might be available

https://arduino.stackexchange.com/questions/21137/arduino-how-to-get-the-board-type-in-code

```C++

#if defined(TEENSYDUINO) 

    //  --------------- Teensy -----------------

    #if defined(__AVR_ATmega32U4__)
        #define BOARD "Teensy 2.0"
    #elif defined(__AVR_AT90USB1286__)       
        #define BOARD "Teensy++ 2.0"
    #elif defined(__MK20DX128__)       
        #define BOARD "Teensy 3.0"
    #elif defined(__MK20DX256__)       
        #define BOARD "Teensy 3.2" // and Teensy 3.1 (obsolete)
    #elif defined(__MKL26Z64__)       
        #define BOARD "Teensy LC"
    #elif defined(__MK64FX512__)
        #define BOARD "Teensy 3.5"
    #elif defined(__MK66FX1M0__)
        #define BOARD "Teensy 3.6"
    #else
       #error "Unknown board"
    #endif

#else // --------------- Arduino ------------------

    #if   defined(ARDUINO_AVR_ADK)       
        #define BOARD "Mega Adk"
    #elif defined(ARDUINO_AVR_BT)    // Bluetooth
        #define BOARD "Bt"
    #elif defined(ARDUINO_AVR_DUEMILANOVE)       
        #define BOARD "Duemilanove"
    #elif defined(ARDUINO_AVR_ESPLORA)       
        #define BOARD "Esplora"
    #elif defined(ARDUINO_AVR_ETHERNET)       
        #define BOARD "Ethernet"
    #elif defined(ARDUINO_AVR_FIO)       
        #define BOARD "Fio"
    #elif defined(ARDUINO_AVR_GEMMA)
        #define BOARD "Gemma"
    #elif defined(ARDUINO_AVR_LEONARDO)       
        #define BOARD "Leonardo"
    #elif defined(ARDUINO_AVR_LILYPAD)
        #define BOARD "Lilypad"
    #elif defined(ARDUINO_AVR_LILYPAD_USB)
        #define BOARD "Lilypad Usb"
    #elif defined(ARDUINO_AVR_MEGA)       
        #define BOARD "Mega"
    #elif defined(ARDUINO_AVR_MEGA2560)       
        #define BOARD "Mega 2560"
    #elif defined(ARDUINO_AVR_MICRO)       
        #define BOARD "Micro"
    #elif defined(ARDUINO_AVR_MINI)       
        #define BOARD "Mini"
    #elif defined(ARDUINO_AVR_NANO)       
        #define BOARD "Nano"
    #elif defined(ARDUINO_AVR_NG)       
        #define BOARD "NG"
    #elif defined(ARDUINO_AVR_PRO)       
        #define BOARD "Pro"
    #elif defined(ARDUINO_AVR_ROBOT_CONTROL)       
        #define BOARD "Robot Ctrl"
    #elif defined(ARDUINO_AVR_ROBOT_MOTOR)       
        #define BOARD "Robot Motor"
    #elif defined(ARDUINO_AVR_UNO)       
        #define BOARD "Uno"
    #elif defined(ARDUINO_AVR_YUN)       
        #define BOARD "Yun"

    // These boards must be installed separately:
    #elif defined(ARDUINO_SAM_DUE)       
        #define BOARD "Due"
    #elif defined(ARDUINO_SAMD_ZERO)       
        #define BOARD "Zero"
    #elif defined(ARDUINO_ARC32_TOOLS)       
        #define BOARD "101"
    #else
       #error "Unknown board"
    #endif

#endif
```

per1234, 2019-01-28:
>Do these platforms have official identifiers (something that can be used in code, like promini or pro_mini for the "Pro Mini")?

There is a standard convention that a macro be created `ARDUINO_{build.board}`, where the `{build.board}` property is defined for each board in boards.txt. If `{build.board}` is not defined, the Arduino IDE automatically defines it as `{architecture}_{board ID}`. There is nothing that requires the actual macro to be defined in platform.txt so it's possible that some silly 3rd party hardware package author might leave that off, but I'm not aware of any instances of that. Certainly all of the official Arduino hardware packages are going to define it.

Reference:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#boardstxt

Doom4535, 2019-05-29:
Looking at how arduino handles boards:
* https://github.com/arduino/Arduino/tree/master/hardware
* https://atadiat.com/en/e-arduino-core-source-files-make-new-core-building-steps/

It looks like it gives each architecture has its own 'core' (located in `AppData\Local\Arduino15\packages\arduino\hardware\` on windows) and then presumably uses `#defines` to switch between cores.

The naming for the IO definitions does not seem to follow a convention between architectures, with [sam Arduino.h](https://github.com/matthijskooijman/arduino-platform-sam/blob/master/system/libsam/chip.h) including `chip.h` instead of `avr/io.h`.  I found `chip.h` here: `C:\Users\acovrig\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\system\libsam\chip.h`

For the Godmode virtual core, it would be nice if we had a way to remove the `<avr/io.h>` component from all the `cpp/arduino/*` files except for one.  We could then use the `misc/default.yml` definitions to change where this file pointed when using different arduino cores.  I don't know if we will also need to use this procedure for other files in `cpp/arduino` and `cpp/arduino/avr`.

ianfixes, 2019-05-29:
There are some oddities related to this, IIRC but I absolutely like this idea.

The oddities are that some code files want to include from `arduino/avr`.  See #119 , maybe @hlovdal can comment on possible obstacles?

Doom4535, 2019-06-05:
I've been looking through the project to come up with a plan, and noticed that there is support for installing libraries and boards included in [`/lib/arduino_ci/arduino_cmd.rb`](https://github.com/ianfixes/arduino_ci/blob/cb3fe8df14a744b9a3bc5cdc0168f820f78cde57/lib/arduino_ci/arduino_cmd.rb#L182) and it appears to be called in [`/exe/arduino_ci_remote.rb`](https://github.com/ianfixes/arduino_ci/blob/cb3fe8df14a744b9a3bc5cdc0168f820f78cde57/exe/arduino_ci_remote.rb#L319).

It would appear that the main difference between the `./arduino` build tool and the g++ setup is that the build call also includes a reference to the target board ([issue thread with example](https://github.com/arduino/Arduino/issues/6165#issuecomment-292741335)).  This would make sense, as the error is for missing headers.

Here is some sample output without the micro-controller defined:
`g++ -std=c++0x -o /home/travis/build/Doom4535/arduino_ci/SampleProjects/TestSomething/unittest_defines.cpp.bin -DARDUINO=100 -g -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address -I/home/travis/build/Doom4535/arduino_ci/cpp/arduino -I/home/travis/build/Doom4535/arduino_ci/cpp/unittest -I/home/travis/build/Doom4535/arduino_ci/SampleProjects/TestSomething/test -I/home/travis/build/Doom4535/arduino_ci/SampleProjects/TestSomething /home/travis/build/Doom4535/arduino_ci/cpp/arduino/Arduino.cpp /home/travis/build/Doom4535/arduino_ci/cpp/arduino/Godmode.cpp /home/travis/build/Doom4535/arduino_ci/cpp/arduino/stdlib.cpp /home/travis/build/Doom4535/arduino_ci/cpp/unittest/ArduinoUnitTests.cpp /home/travis/build/Doom4535/arduino_ci/SampleProjects/TestSomething/test-something.cpp /home/travis/build/Doom4535/arduino_ci/SampleProjects/TestSomething/test/defines.cpp`

I'm thinking that this will need to be modified to include something along the lines of:
`~/.arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino` on a linux system.  Where this path is modified to match the core being built against (if it is not the standard avr core, additional cores are added in the users home directory and not the main install location with the 'standard' avr cores).

Where would be a good entry point to try to add an addition include flag into the code, possibly `arduino_cmd`?

More info on arduino board names: https://github.com/arduino/Arduino/issues/8716 and here: https://github.com/arduino/arduino-cli and here: https://github.com/arduino/Arduino/issues/6165

ianfixes, 2019-06-05:
`arduino_cmd.rb` does indeed wrap much of the functionality provided by the Arduino binary.  There are 3 main uses for that:

1. installing boards
2. switching to boards
3. verifying sketches

> the main difference between the `./arduino` build tool and the g++ setup

Just so we're on the same page about the differences:

|     |`./arduino` build tool|`g++` manual compilation|
|---|---------------------|--------------------------|
|**Purpose**| Verify that sketches compile |Standalone unit testing of the library|
|**Output artifact**|Sketch|Unit test binary|
|**Target architecture**|Arduino (chipset specified on command line)|x86|
|**Hardware support**|Board-specific libraries|Mocks or manual references to board headers|
|**Compiler flag generation**| ??? | `cpp_library.rb`|
|**Handles dependent libraries**|Yes|Not yet|

The crux of the issue here is that I'm subverting the normal compilation of a library & sketch -- I want all the board-specific items like registers, ports, etc, but not the actual Arduino architecture.

There's probably a far easier way to do all of this, but I'm not familiar enough with the source code of the Arduino IDE to know where to look or what to do there.

ianfixes, 2019-06-05:
As far as defining your own flags, the quickest way to experiment would be to just put them `.arduino-ci.yml` as per the REFERENCE.md: https://github.com/ianfixes/arduino_ci/blob/master/REFERENCE.md

ianfixes, 2020-12-24:
This can be made easier by https://github.com/arduino/arduino-cli/issues/1090