Arduino-CI/arduino_ci #91: Provide instruction to test libraries / sketches with dependencies

scls19fr, 2019-01-15:
Hello,

I think it should be useful to provide information in doc to test libraries / sketches with dependencies.

I see 2 kind of dependencies:

- registered Arduino libraries
  - latest version
  - a given version number
  - latest version matching some restriction (upper than x.y.z lower than x2.y.z for example)

- unregistered libraries
  - available from a HTTP server
(like https://github.com/itead/ITEADLIB_Arduino_Nextion/archive/v0.7.0.zip )

  - git repository
    - master branch
    - other branch (latest commit)
    - a given commit hash

My opinion is that [arduino-cli](https://github.com/arduino/arduino-cli/) should manage this... (instead of everyone doing its own script)
I opened several issues related to this problem.

arduino-cli
 - Installing unregistered library https://github.com/arduino/arduino-cli/issues/128
 - Version restriction for installing library https://github.com/arduino/arduino-cli/issues/105

A more general issue (that I didn't opened)
Library manager: Library dependencies https://github.com/arduino/Arduino/issues/5795

The concept of environments could help to have several versions of a same library installed without hassle https://github.com/arduino/arduino-cli/issues/108 ... but that's an other story.

What's you opinion about this @ianfixes?
What actionable stuff can be done?

Kind regards

ianfixes, 2019-01-15:
Here's how you'd enable unit tests with a dependent library (via https://github.com/ianfixes/arduino_ci/blob/master/REFERENCE.md):

```yml
unittest:
  libraries:
    - "abc123"
    - "def456"
  platforms:
    - mega2560
```

That will cause the library manager to download `abc123` and `def456` if they aren't already.

For unregistered libraries, I have to leave it to the developer to put the appropriate download commands into their build scripts such that the dependent libs end up in the Arduino library directories.  Whether that means `curl` or `wget` a URL and unzipping it, or `git clone`ing something, that's currently the user's responsibility.

My personal opinion is that Arduino needs a package manager on par with NPM (NodeJS), bundler (Ruby), or Cargo (Rust) that can handle all those download sources, semver, recursive dependencies, etc.  I won't have time to build such a thing and do it correctly.

ianfixes, 2019-01-15:
Is there a specific thing you'd like to see worded differently in the documentation about dependent libraries?

It also occurs to me that I'm very unlikely to be compiling dependent libraries properly.  Getting the list of files in the right order is something I'm not sure how to tackle.

scls19fr, 2019-01-16:
I did https://github.com/scls19fr/mwe_arduino_ci/blob/539938a292459be41981ce2f4dd8391ac857dd27/.travis.yml
Here is my build job https://travis-ci.com/scls19fr/mwe_arduino_ci/builds/97511900
It's failing ... I don't understand why I'm downloading a bad zip file

scls19fr, 2019-01-16:
Looks good now with the following `.travis.yml`
```
sudo: false
language: ruby
before_install:
   - wget https://github.com/itead/ITEADLIB_Arduino_Nextion/archive/master.zip
   - unzip -o master.zip
   - mkdir -p ~/Arduino/libraries/
   - mv ITEADLIB_Arduino_Nextion-master ~/Arduino/libraries/
script:
   - bundle install
   - bundle exec arduino_ci_remote.rb
```

https://travis-ci.com/scls19fr/mwe_arduino_ci/builds/97514947
but on Linux, Arduino libraries are in `~/Arduino/libraries`, on OS X it's `~/Documents/Arduino/libraries`

If it was managed by an other script, user will not have to take care about that.

ianfixes, 2019-01-16:
What I have in #92 should allow you to do this:

```yml
sudo: false
language: ruby
script:
   - bundle install
   - bundle exec ensure_arduino_installation.rb
   - echo "Arduino library directory:"
   - bundle exec arduino_library_location.rb
   - wget https://github.com/itead/ITEADLIB_Arduino_Nextion/archive/master.zip
   - unzip -o master.zip
   - mv ITEADLIB_Arduino_Nextion-master $(bundle exec arduino_library_location.rb)
   - bundle exec arduino_ci_remote.rb
```

These lines aren't strictly necessary: 
```yml
   - bundle exec ensure_arduino_installation.rb
   - echo "Arduino library directory:"
   - bundle exec arduino_library_location.rb
```
They will just make the output a bit easier to understand.

Can you give that branch a try?  The only thing I haven't tested is whether the subshell syntax (wrapping a `command` in `$( command )`) works in Travis.

ianfixes, 2019-01-16:
Made a mistake while pushing things up, the ref you want is 6dcc33e8c336395885bd8ee98bb504e9bf2d9d3c

scls19fr, 2019-01-16:
Tested in https://travis-ci.com/scls19fr/mwe_arduino_ci/builds/97533698

ianfixes, 2019-01-16:
OK, looks like I had some minor formatting bugs in there ([these dots, not being on the same line](https://travis-ci.com/scls19fr/mwe_arduino_ci/builds/97533698#L466)) which I just fixed.  I'll merge that when the latest tests complete.

scls19fr, 2019-01-16:
https://travis-ci.com/scls19fr/mwe_arduino_ci/builds/97540536

closed 2019-01-17 by ianfixes

scls19fr, 2019-01-17:
Thanks @ianfixes

PR #92 (2019-01-17): Enable redirection of installation output
Adds ability to do cross-platform build scripts, ensuring that custom libraries get copied into the correct Arduino library directory.

## Highlights from `CHANGELOG.md`

- `ArduinoInstallation` and `ArduinoDownloader` now allow console output to optionally be set to an `IO` object of choice during `force_install`
- `ArduinoInstallation::force_install` now optionally accepts a version string
- `arduino_library_location.rb` script to print Arduino library location to stdout

## Issues Fixed

* Fixes #91