Arduino-CI/arduino_ci #93: arduino_ci_remote.rb --testfile-select=GLOB shouldn't compile examples

scls19fr, 2019-01-23:
Hello,

    $ bundle exec arduino_ci_remote.rb --testfile-select=test/test_*.cpp

shouldn't compile examples

or we should have a flag to disable compiling examples such as

    $ bundle exec arduino_ci_remote.rb --testfile-select=test/test_*.cpp --examples-reject=*

Kind regards

hlovdal, 2019-01-26:
@scls19fr There already is a `--testfile-reject` option (since version 0.1.17) that seems to do what you are asking for. Is that insufficient? What criteria do you have for something to be an example that should not be compiled?

scls19fr, 2019-01-26:
It doesn't work as expected

```
$ bundle exec arduino_ci_remote.rb --testfile-select=test_led.cpp --testfile-reject=*
Located Arduino binary...                                /Users/scls/Arduino.app
{:compilers=>["g++"], :libraries=>nil, :platforms=>["uno"]}
{:compilers=>["g++"], :libraries=>nil, :platforms=>["uno"], :testfiles=>{:select=>["test_led.cpp"], :reject=>["*"]}}
The set of compilers (1) isn't empty...                                        ✓
Checking g++ version...
    Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
    Apple LLVM version 10.0.0 (clang-1000.11.45.5)
    Target: x86_64-apple-darwin17.7.0
    Thread model: posix
    InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
...Checking g++ version                                                        ✓
libasan availability for g++...                                             true
Requested unittest platform 'uno' is defined in 'platforms' YML...             ✓
Indexing libraries...                                                          ✓
Installing library under test...                                               ✓
Library installed at... /Users/scls/Documents/Arduino/libraries/ArduinoPartsPushButton
Requested library example platform 'uno' is defined in 'platforms' YML...      ✓
Using built-in board package...                                      arduino:avr
Installing board package arduino:avr...                                        ✓
Setting compiler warning level...                                              ✓
Switching to board for uno (arduino:avr:uno)...                                ✓
Verifying 1_on_off.ino...                                                      ✓
Verifying 2_led_blink_forever.ino...                                           ✓
Failures: 0
```

Examples are compiled despite I don't want them to be compiled (because I fix first code and unit tests and only after this, I fix examples).

You can also notice that unit test file `test_led.cpp` is not run despite being explicitly selected.
(it's both selected and rejected according glob... so it's rejected with such an implementation !)

ianfixes, 2019-01-28:
The behavior of `arduino_ci_remote.rb` is to be the end-all-be-all for remote CI.  (I had intended to make a matching script for local testing, but I felt it was wiser to use one script in both places... and the name had stuck.) What this means is that both unit testing and compilation of `examples/` _always_ happen.  

What's missing here is a set of flags like `--skip-unittest` and `--skip-compilation`.  I will add those to the next release.