Arduino-CI/arduino_ci #102: assertEqual does not support nullptr

hlovdal, 2019-01-28:
Trying to test that a function returns a null pointer, e.g.

```C++
	assertEqual(nullptr, result);
```

does not work

```shell
.../ArduinoUnitTests.h:69:56: error: ambiguous overload for ‘operator<<’ (operand types are ‘std::basic_ostream<char>’ and ‘std::nullptr_t’)
               cerr << "      " << lhsRelevance << ": " << lhs << endl;
```

I tried to fix it by extracting that line into a template function and then make a template specialisation for nullptr but did not get the compiler to select it, and have put that on ice for now since there is a simple workaround, just adding
```C++
#define nullptr (void *)NULL
```
at the top of the test file. But it would be nice to eventually have this fixed.

ianfixes, 2019-01-29:
Yup, this seems like part of C++11 that needs to be added to the mocks.  What's the data type of `result` in the code you posted?  I'll add that to a unit test.

ianfixes, 2019-01-29:
I will add `nullptr` and `assertNull(result)`

hlovdal, 2019-01-30:
Those updates still do not fix the issue with `assertEqual`. The problem is that onAssert tries to print the argument of type `nullptr` which it is not able to. There needs to be an additional unit test:

```C++
unittest(nullpointer_equal)
{
  int* myPointer = NULL;
  int **notNullPointer = &myPointer;
  assertEqual(nullptr, myPointer);
  assertNotEqual(nullptr, notNullPointer);

#if 0
  // The following should not compile, which they currently do not but in a
  // cryptic way. If possible and with low effort an explicit denial which
  // produces an explanatory error would be nice.
  assertLessOrEqual(nullptr, myPointer);
  assertMoreOrEqual(nullptr, notNullPointer);
  assertLess(nullptr, myPointer);
  assertMore(nullptr, notNullPointer);
  assertFalse(nullptr);
  assertTrue(nullptr);
  assertAnyLhsRhsFunction(..., nullptr);
#endif
}
```

ianfixes, 2019-01-30:
Thanks, those unit tests are exactly what was needed.

closed 2019-01-30 by ianfixes

ianfixes, 2019-01-31:
@hlovdal You gave me an excellent idea here.

I've set up a new branch called `tdd` (https://github.com/ianfixes/arduino_ci/tree/tdd).  If you can write unit tests that show a problem, make a PR using the `tdd` branch as a base.  That will give me something to work from when I go about fixing the issue.

ianfixes, 2019-02-07:
This fix is live in the latest version, sorry for the arbitrary delay there

PR #103 (2019-01-29): Unit test additions (both Ruby and C++)
## Highlights from `CHANGELOG.md`
- `keepachangelog_manager` gem to begin streamlining the release process
- `unittest_setup()` and `unittest_teardown()` macros, my thanks to @hlovdal for contributing this code
- Added rspec sensitivity to the environment variable `$ARDUINO_CI_SKIP_SPLASH_SCREEN_RSPEC_TESTS` (for `arduino_ci` gem hackers)
- Added rspec sensitivity to the environment variable `$ARDUINO_CI_SKIP_RUBY_RSPEC_TESTS` (for `arduino_ci` gem hackers)
- Added rspec sensitivity to the environment variable `$ARDUINO_CI_SKIP_CPP_RSPEC_TESTS` (for `ard [... cut]
PR #106 (2019-01-30): Fixed nullptr and made some general test speedups
## Highlights from `CHANGELOG.md`
### Added
- Added rspec sensitivity to the environment variable `$ARDUINO_CI_SELECT_CPP_TESTS=<glob>` (for `arduino_ci` gem hackers)
- `assertNotNull()` and `assureNotNull()` C++ comparisons

### Changed
- `CiConfig::allowable_unittest_files` now uses `Pathname` to full effect
- `nullptr` now defined in its own class

### Fixed
- Assertions on `nullptr`
- The defintion of `nullptr`

* See CHANGELOG.md for more

## Issues Fixed

* Fixes #102
PR #112 (2019-02-05): Fix nullptr support again, remove custom implementation
## Highlights from `CHANGELOG.md`

* Homegrown `nullptr` definition removed. 
* Proper template and ostream operations added for `std::nullptr`
* See CHANGELOG.md for more

## Issues Fixed

* Fixes #102  
* Corrects #106
* Supercedes #110