Arduino-CI/arduino_ci #109: Platform dependent examples

PRosenb, 2019-02-02:
Thanks for creating this project.

In my library, I have examples for all supported platforms as well as some specific ones only for Arduino Uno or only for ESP32. That's needed to show specific features only available on these platforms.
It would be great to add an option to either exclude examples from compilation or the ability to specify their target platform.
Many thanks for considering something like that.

ianfixes, 2019-02-03:
That feature [already exists](https://github.com/ianfixes/arduino_ci/blob/master/REFERENCE.md#control-how-examples-are-compiled), but I'll take this as feedback that the documentation needs to be improved.  

TL;DR, you need to add a new file to each of your platform specific example directories called `.arduino-ci.yml`.  It should contain the following:

```yml
# Specify that this example should only be run on the `esp32` platform
compile:
  platforms:
    - esp32 # or "uno", or whatever platform
```

Is there anything I could add or word differently in [the referenec documentation](https://github.com/ianfixes/arduino_ci/blob/master/REFERENCE.md) to make this more evident?

ianfixes, 2019-02-03:
If you give me a link to an `arduino_ci`-enabled branch (with these tests failing), I'd be glad to take a look.

PRosenb, 2019-02-04:
Many thanks @ianfixes for your detailed explanation, I appreciate it.
I in fact saw that documentation but didn't catch that it can be specified for each example separately even though it was written earlier. What about changing it like that?
[Pull Request](https://github.com/ianfixes/arduino_ci/pull/111)

PR #111 (2019-02-04): Expand Reference.md section on configuration overrides for examples
- add hint to file `.arduino-ci.yml` related to examples
- define expected and actual on the assert functions 

* Fixes #109
closed 2019-02-04 by ianfixes