Arduino-CI/arduino_ci #127: This is the simplest arduino-ci.yml I could get working in a gitlab-ci.yml script

user-name-is-taken, 2019-07-29:
[Link to the arduino-ci.yml](https://gitlab.com/robotics_fun/DC_motors/blob/master/.arduino-ci.yaml). I'm posting this because the example you gave here didn't work for my project and I almost posted issues for [this](https://gitlab.com/robotics_fun/DC_motors/-/jobs/260929416) job's confusing errors.

ianfixes, 2019-07-29:
Hi Michael, I'm sorry to hear that you've encountered these problems.  I'm a bit confused by the two links you've posted, because they appear to be for 2 separate issues.  Is that correct?  

It looks like you're trying to draw attention to this part of the config:

```yaml
  testfiles:
  # THIS IS REQUIRED
    # files matching this glob (executed inside the `test/` directory) will be whitelisted for testing
    select:
      - "*-*.*"
```

Did you find that the `testfiles` section or just the `select` subsection was required?  A link to a failing build would be very helpful for me, but I will take a look.  

Please don't hesitate to open a separate issue for the confusing error message; it's my goal to eliminate such things from this project.

ianfixes, 2019-07-29:
My first thought was that I was failing to guard against `nil` when I read the config, but it looks like I have that pretty well covered.  https://github.com/ianfixes/arduino_ci/blob/master/lib/arduino_ci/ci_config.rb#L292-L303

So I will need to see the build output to get a sense of what's failing.

user-name-is-taken, 2019-07-30:
[Before fix](https://gitlab.com/robotics_fun/DC_motors/pipelines/73406939) and [after fix](https://gitlab.com/robotics_fun/DC_motors/pipelines/73407951)

ianfixes, 2019-07-30:
According to [this diff](https://gitlab.com/robotics_fun/DC_motors/compare/ec3fa1e5d72c00f33e08756cdce4e72c3f441668...b17f647524458d978df1a7481c8848af6c7a2aa8), I think the real reason the test started passing is this line:

```diff
-    - bundle exec arduino_ci_remote.rb
+    - bundle exec ensure_arduino_installation.rb
```

`ensure_arduino_installation.rb` is simply the first of the operations that `arduino_ci_remote.rb` performs; it's included so that you can troubleshoot the arduino installation prior to running your tests on it.  This leads me to believe that the real (and sole) problem is #128 so I will dive into that.  

Can you address my https://github.com/ianfixes/arduino_ci/issues/128#issuecomment-515965402 ?

ianfixes, 2019-07-31:
It looks like there isn't a configuration issue here, so I'm closing this.  But this report helped uncover 2 very valid bugs (#128 and #129), so thank you!

closed 2019-07-31 by ianfixes