Arduino-CI/arduino_ci #128: get_pref': undefined method `[]' for nil:NilClass

ianfixes, 2019-07-29:
## System

- OS: _(Travis/OSX/Linux/Windows)_
- `ruby -v`: `ruby 2.5.1p57 (2018-03-29 revision 63029) [x86_64-linux-gnu]`
- `bundle -v`: `Bundler version 1.16.1`
- `bundle info arduino_ci`: `Installing arduino_ci 0.2.0`
- Arduino IDE version:
- `java -version`:
- URL of failing CI job: https://gitlab.com/robotics_fun/DC_motors/-/jobs/260929416
- URL of your Arduino project: https://gitlab.com/robotics_fun/DC_motors

## (Issue Summary / Feature Request)
```
$ bundle exec arduino_ci_remote.rb
Attempting to download Arduino package with open-uri...[note: elipses elided]...
Extracting archive with tar
Located Arduino binary...                           /root/arduino_ci_ide/arduino
/var/lib/gems/2.5.0/gems/arduino_ci-0.2.0/lib/arduino_ci/arduino_cmd.rb:106:in `get_pref': undefined method `[]' for nil:NilClass (NoMethodError)
	from /var/lib/gems/2.5.0/gems/arduino_ci-0.2.0/lib/arduino_ci/arduino_cmd.rb:78:in `lib_dir'
	from /var/lib/gems/2.5.0/gems/arduino_ci-0.2.0/exe/arduino_ci_remote.rb:176:in `perform_unit_tests'
	from /var/lib/gems/2.5.0/gems/arduino_ci-0.2.0/exe/arduino_ci_remote.rb:383:in `<top (required)>'
	from /usr/local/bin/arduino_ci_remote.rb:23:in `load'
	from /usr/local/bin/arduino_ci_remote.rb:23:in `<main>'
```

ianfixes, 2019-07-29:
My only guess as to why this is happening is that running `arduino --get-pref` is returning something that fails to parse as a preferences list.  

We will have to debug that in-situ; in the case of [this particular configuration](), the following `test` section should accomplish it:

```yml
test:
  script:
    - bundle exec ensure_arduino_installation.rb
    # this appears to install the arduino binary at /root/arduino_ci_ide/arduino
    #   according to https://gitlab.com/robotics_fun/DC_motors/-/jobs/260929416
    - /root/arduino_ci_ide/arduino --get-pref
    - bundle exec arduino_ci_remote.rb
```

user-name-is-taken, 2019-07-30:
- So I tried 2 things. The second one seems to have worked although it still says the job is failing.
  1. [Doing the tests you suggested](https://gitlab.com/robotics_fun/DC_motors/pipelines/73828054)
  2. [Removing arduino-cli](https://gitlab.com/robotics_fun/DC_motors/-/jobs/262638326)
      - As stated above,  this seems to have worked, but is still failing. Perhaps it's failing because  I don't have an arduino environment anymore?

ianfixes, 2019-07-31:
> [Doing the tests you suggested](https://gitlab.com/robotics_fun/DC_motors/-/jobs/262637286)

This is a failure mode I haven't seen before:

```
$ /root/arduino_ci_ide/arduino --get-pref
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
cc.arduino.contributions.SignatureVerificationFailedException: package_index.json file signature verification failed
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:91)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
	at processing.app.Base.<init>(Base.java:262)
	at processing.app.Base.main(Base.java:145)
```

This `SignatureVerificationFailedException` is a failure with the arduino binary itself, so I can't really speak to the cause of it.  Perhaps it's related to the files in the image your CI system is using?  I noticed that it gets past this step in later builds.  

In case it helps your troubleshooting, FYI `ensure_arduino_installation.rb` should use the built-in version if it finds one in `$PATH`.  According to the build output, that's not happening here; it's using `/root/arduino_ci_ide/arduino` which is the path that arduino_ci will use when it decides it must download its own copy of the IDE.

In any case, what you did in this job has solved the problem: https://gitlab.com/robotics_fun/DC_motors/-/jobs/262638326

So the root of the issue described here is that I'm not detecting a failure of the Arduino binary, and so I will need to improve that.

Unfortunately, I think you've also uncovered #129 which I will fix now.

ianfixes, 2019-07-31:
To fix this issue, it would be helpful to know if the arduino binary itself gave a return code for failure (because if not, I will have to extract that from the captured STDOUT).  

Can you return to the revision with the failing arduino binary and try this?

```yml
test:
  script:
    - bundle exec ensure_arduino_installation.rb
    # this appears to install the arduino binary at /root/arduino_ci_ide/arduino
    #   according to https://gitlab.com/robotics_fun/DC_motors/-/jobs/260929416
    - /root/arduino_ci_ide/arduino --get-pref ; echo $?
    - bundle exec arduino_ci_remote.rb
```
Note the addition of `; echo $?` to log the return code of the command

ianfixes, 2019-07-31:
Actually, nevermind.  I think the bug on my end is that I detect a failure of that command but continue silently.  That's fixed in #130 

Can you update your `Gemfile` to use my latest branch?

```ruby
gem 'arduino_ci', :git => 'https://github.com/ianfixes/arduino_ci.git', :branch => '2019-07-31_bugfixes'
```

Then we can see if it solves your problems, after which I can cut a release.

user-name-is-taken, 2019-08-01:
https://gitlab.com/robotics_fun/DC_motors/-/jobs/263962588

ianfixes, 2019-08-01:
It looks like [you've added that line to your `.gitlab-ci.yml` file](https://gitlab.com/robotics_fun/DC_motors/commit/245e05de6b59913c9e25fa74454579fe5dd0de9c)... 

That line belongs in the [`Gemfile`](https://gitlab.com/robotics_fun/DC_motors/blob/245e05de6b59913c9e25fa74454579fe5dd0de9c/Gemfile):

```diff
  source 'https://rubygems.org'
- gem 'arduino_ci'
+ gem 'arduino_ci', :git => 'https://github.com/ianfixes/arduino_ci.git', :branch => '2019-07-31_bugfixes'
```

user-name-is-taken, 2019-08-01:
https://gitlab.com/robotics_fun/DC_motors/-/jobs/263976164

ianfixes, 2019-08-01:
Thanks!  Apparently I have some other issue with including the libraries, although I'm not quite sure what at first glance.  It's including `AccelStepper` properly, so I'm not sure why it's failing to do the  motor shield library.  My suspicion is something related to spaces in the directory name, failing to be quoted somewhere and thus appearing as a "nonexistent" directory.  I'll work on that.

Thanks for you help, and your patience.

user-name-is-taken, 2019-08-01:
Thank you so much for all your hard work!

ianfixes, 2019-08-06:
I see the problem, and it appears you did too: there is a discrepancy between the library name as it appears in the manager -- `Adafruit Motor Shield V2 Library` -- and its directory name -- `Adafruit_Motor_Shield_V2_Library`.  I'm trying to see if there's a convenient way for me to extract this, but otherwise I will need to come up with a more robust way to configure this.

ianfixes, 2019-08-06:
After I fix #132 I think this will all work

ianfixes, 2019-08-06:
I've updated the `2019-07-31_bugfixes` branch, if you want to try that again. Note that you'll need to refer to the library in your `.arduino-ci.yml` file by its proper name -- `Adafruit Motor Shield V2 Library`, with spaces -- for things to work.  I will automatically convert those spaces to underscores in the compilation step.  

Spoiler, I pulled down your code locally and the CI works for me... but you have [other compilation problems](https://stackoverflow.com/q/18940175/2063546) in your code.

ianfixes, 2019-08-07:
@user-name-is-taken #130 is ready to go, I'm hoping to have you test it [like you did earlier](https://github.com/ianfixes/arduino_ci/issues/128#issuecomment-517290864) before I cut a new release.  No particular rush, but can you let me know either way?

closed 2019-08-12 by ianfixes

user-name-is-taken, 2019-08-14:
https://gitlab.com/robotics_fun/DC_motors/-/jobs/271984030

ianfixes, 2019-08-14:
This should be all you need in your `.arduino-ci.yaml`.  Note that the name of the library is the name that appears in the Arduino library manager, _not_ the name of the directory that appears in your Arduino libraries (now that #132 is fixed)
```yml
compile:
  libraries:
    - Adafruit Motor Shield V2 Library
    - Regexp
    - AccelStepper
  platforms:
    - uno

unittest:
  libraries:
    - Adafruit Motor Shield V2 Library
    - Regexp
    - AccelStepper
  platforms:
    - uno #arduino:avr:uno
```

Optionally, now that I released  v`0.2.1` of the gem you might want to update your `Gemfile` to point to that
```diff
- gem 'arduino_ci', :git => 'https://github.com/ianfixes/arduino_ci.git', :branch => '2019-07-31_bugfixes'
+ gem 'arduino_ci', '~> 0.2.1'
```

user-name-is-taken, 2019-08-14:
https://gitlab.com/robotics_fun/DC_motors/-/jobs/272742857

ianfixes, 2019-08-15:
🎉 Excellent!

I just want to reiterate: by copying and pasting large parts of your `.arduino-ci.yaml` file, you've introduced some unnecessary cruft into your project.

* The `packages` and `platforms` sections in your project's config file will simply override what is in the default configuration.  In this case, they are identical to the default config, so this is just a wall of text that does literally nothing.
* `unittest / compilers` is really designed for you to force the use of a specific compiler, or for a developer of `arduino_ci` itself to look for compiler discrepancies that can break the testing process (of which there are a few).  By having multiple entries here, you're just doubling the testing time.
* By having a `testfiles / select` entry, you're forcing `arduino_ci` to only consider unit test files whose names fit the glob `*-*.*`.  So if you renamed `good-null.cpp` to `goodnull.cpp`, you would find that the tests no longer ran.  

I know I documented at least some of this in https://github.com/ianfixes/arduino_ci/blob/master/REFERENCE.md

Thanks for your help in wringing these bugs out of arduino_ci!

PR #130 (2019-08-12): Bugfixes 
## Highlights from `CHANGELOG.md`

### Changed
- Arduino command failures (to read preferences) now causes a fatal error, with help for troubleshooting the underlying command

 ### Fixed
- Arduino library dependencies are now installed prior to unit testing, instead of prior to compilation testing.  Whoops.

## Issues Fixed

* Fixes #128 
* Fixes #129 
* Fixes #132