Arduino-CI/arduino_ci #131: Unittests fail to compile as soon as `<vector>` is included in any test file.

dolfandringa, 2019-08-05:
## System

- OS: _Linux_
- `ruby -v`: ruby 2.6.3p62 (2019-04-16 revision 67580) [x86_64-linux]
- `bundle -v`: Bundler version 1.17.2
- `bundle info arduino_ci`:
   * arduino_ci (0.2.0)
        Summary: Tools for building and unit testing Arduino libraries
        Homepage: http://github.com/ianfixes/arduino_ci
        Path: /home/me/arduino_ci_test/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.2.0
- `g++ -v`: 
Besides below, also affects GCC8 on Ubuntu
```
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,objc,obj-c++,ada,go,d,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl --enable-offload-targets=nvptx-none --without-cuda-driver --enable-gnu-indirect-function --enable-cet --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
Thread model: posix
gcc version 9.1.1 20190503 (Red Hat 9.1.1-1) (GCC)
```
- Arduino IDE version: 1.8.6
- `java -version`: java version "1.8.0_201"
- URL of failing Travis CI job:
- URL of your Arduino project:

## (Issue Summary / Feature Request)
When I `#include <vector>` in a unittest, tests fail to compile with
```
Last command:  $ g++ -std=c++0x -o /home/me/arduino_ci_test/unittest_test_me.cpp.bin -DARDUINO=100 -g -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address -D__AVR_ATmega328P__ -I/home/me/arduino_ci_test/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.2.0/cpp/arduino -I/home/me/arduino_ci_test/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.2.0/cpp/unittest -I/home/me/arduino_ci_test /home/me/arduino_ci_test/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.2.0/cpp/arduino/Arduino.cpp /home/me/arduino_ci_test/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.2.0/cpp/arduino/Godmode.cpp /home/me/arduino_ci_test/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.2.0/cpp/arduino/stdlib.cpp /home/me/arduino_ci_test/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.2.0/cpp/unittest/ArduinoUnitTests.cpp /home/me/arduino_ci_test/mytest.cpp /home/me/arduino_ci_test/test/test_me.cpp

In file included from /home/me/arduino_ci_test/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.2.0/cpp/arduino/WString.h:8,
                 from /home/me/arduino_ci_test/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.2.0/cpp/unittest/Compare.h:3,
                 from /home/me/arduino_ci_test/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.2.0/cpp/unittest/Assertion.h:7,
                 from /home/me/arduino_ci_test/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.2.0/cpp/unittest/ArduinoUnitTests.h:4,
                 from /home/me/arduino_ci_test/test/test_me.cpp:1:
/usr/include/c++/9/bits/stl_bvector.h: In member function ‘std::vector<bool, _Alloc>::size_type std::vector<bool, _Alloc>::_M_check_len(std::vector<bool, _Alloc>::size_type, const char*) const’:
/home/me/arduino_ci_test/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.2.0/cpp/arduino/AvrMath.h:20:18: error: expected unqualified-id before ‘(’ token
   20 | #define max(a,b) ((a)>(b)?(a):(b))
      |                  ^
...Unit testing test_me.cpp with g++                                           ✗
```
I only need to include `vector` in my unttest for it to fail. I don't need to put it in the actual code.  Although I don't use vector in the actual code, for unittesting it is very useful. It looks like the `min` and `max` functions in `cpp/arduino/AvRMath.h` are the culprit. Why are those being re-defined? I managed to temporarily work around the issue by renaming those functions to `get_min` and `get_max` and renaming the references to them in `WString.h` and `Stream.h`, but I expect that is a bad idea if the code depends on them working as defined in `AvRMath.h`

## Arduino or Unit Test Code, Illustrating the Problem
.arduino-ci.yml
```
unittest:
  platforms:
    - uno
```
mytest.cpp
```
#include "mytest.h"
int myTest(void) {
  millis();  // this line is only here to test that we're able to refer to the builtins
  return 4;
};
```
test/test_me.cpp
```
#include <ArduinoUnitTests.h>
#include <Arduino.h>
#include <vector>

unittest(me) {
    assertEqual(5, 5);
}

unittest_main();
```

## Arduino Architecture(s) Affected
at least uno. Haven't tested others.

ianfixes, 2019-08-05:
Although I'm not quite sure where to start debugging this, I have a very direct answer to one of your questions:

> It looks like the min and max functions in cpp/arduino/AvRMath.h are the culprit. Why are those being re-defined?

These functions are provided somewhere within the Arduino function library, but not by c++ itself:
* https://www.arduino.cc/reference/en/language/functions/math/max/
* https://www.arduino.cc/reference/en/language/functions/math/min/

In order to compile Arduino libraries that reference these functions, I need to define them for myself.  My current draft of the c++ mocks does a lot of dancing around the standard library to avoid various conflicts (`String.h` being the most annoying one, IIRC).   So I'm open to the idea that there may be an easier or better way.  

Will the fix that you're describing here work for Arduino library code (i.e. code under test) that calls Arduino's builtin functions (https://www.arduino.cc/reference/en/) ?

dolfandringa, 2019-08-14:
I am starting to understand the issue better. I think the issue is that the macros in `<AvRMath.h>` for defining the `max` and `min` functions, interfere with `std::max and `std::min` from the c++ stdlib: http://www.cplusplus.com/reference/algorithm/max/
The `std::max` and `std::min` functions are used by `std::vector`. So as long as you don't `#include <vector>` there is no problem. But as soon as you do, I think the preprocessor messes up `std::max` and `std::min`.

ianfixes, 2019-08-14:
That's correct.  As I understand it, the stdlib isn't available to arduino at all.  However, there are certainly alternatives that you should be able to use like https://www.arduinolibraries.info/libraries/vector

dolfandringa, 2019-08-15:
Yeah, we don't use stdlib at all for the embedded code. We only want to use it for testing, in our case we wanted to standardise a Mock class that keeps track of which functions were called with which arguments. Similar to what libraries like [FakeIt](https://github.com/eranpeer/FakeIt) do. That's where the vectors came in. But we gave up on it and went back to simpler constructs that don't depend on stdlib/vectors.

closed 2019-08-15 by dolfandringa

ianfixes, 2019-08-15:
Believe me, nobody felt the conflict between the Arduino code and standard libraries more than I did while I was setting up all the compilation mocks... it's a shame that Arduino exploded in popularity before this stuff got sorted out, although I suspect the size of the compiled binary might have been a major factor in the decision.

I'm also frustrated that I can't do a better job of detecting these sorts of errors and displaying a friendlier error message regarding the incompatibility of the libraries.  If you have any suggestions there, I'll gladly reopen this as a way to improve that aspect.