Arduino-CI/arduino_ci #132: arduino_ci can't handle cases where library manager package names are different than installed directory names

ianfixes, 2019-08-06:
## Issue Summary

A discrepancy between the library name as it appears in the manager (e.g. `Adafruit Motor Shield V2 Library`) and its installed location (e.g. `Adafruit_Motor_Shield_V2_Library`) cause arduino_ci to be unable to find the installed library for unit test compilation purposes.

It looks like even in the package list (https://downloads.arduino.cc/libraries/library_index.json) the output directory isn't something that can be trivially extracted.  I'd have to make an assumption based on parsing the `website`, `repository`, `url`, or `archiveFileName` fields... and this bug is already the result of a bad assumption.

```json
    },
    {
      "name": "Adafruit Motor Shield V2 Library",
      "version": "1.0.6",
      "author": "Adafruit",
      "maintainer": "Adafruit \u003cinfo@adafruit.com\u003e",
      "sentence": "Library for the Adafruit Motor Shield V2 for Arduino. It supports DC motors \u0026 stepper motors with microstepping as well as stacking-support.",
      "paragraph": "Library for the Adafruit Motor Shield V2 for Arduino. It supports DC motors \u0026 stepper motors with microstepping as well as stacking-support.",
      "website": "https://github.com/adafruit/Adafruit_Motor_Shield_V2_Library",
      "category": "Device Control",
      "architectures": [
        "*"
      ],
      "types": [
        "Recommended"
      ],
      "repository": "https://github.com/adafruit/Adafruit_Motor_Shield_V2_Library.git",
      "url": "http://downloads.arduino.cc/libraries/github.com/adafruit/Adafruit_Motor_Shield_V2_Library-1.0.6.zip",
      "archiveFileName": "Adafruit_Motor_Shield_V2_Library-1.0.6.zip",
      "size": 19995,
      "checksum": "SHA-256:b9eb18081cb68a5cc4bf9f35afab732e6111c45b6c1e07566fd92625f0d3b4fe"
    },
```

per1234, 2019-08-06:
Library just replaces spaces in the `name` value with _ when creating the library installation folder. Any other problematic characters are disallowed so you only need to deal with spaces:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format
>Library names must contain only basic letters (A-Z or a-z) and numbers (0-9), spaces ( ), underscores (_), dots (.) and dashes (-). It cannot start or end with a space, and also it cannot start with a number.

ianfixes, 2019-08-06:
This is excellent and very timely information.  It would save me from having to do this

```ruby
    # We can't assume that library names and output directories align.
    # ...but we did once, and this ensures backwards compatiblity with that for the moment
    # convert array ["a", "b"] to {"a"=>"a", "b"=>"b"}
    # convert hash {"c"=>"C", "d"=>nil, "e"=>""} to {"c"=>"C", "d"=>"d", "e"=>"e"}
    #
    # @param unknown_object Hash, Array, or hopefully Nil
    # @return [Hash<String, String>]
    def _coerce_libraries_to_hash(unknown_object)
      case unknown_object.class
      when Array then unknown_object.zip(unknown_object).to_h
      when Hash then unknown_object.map { |k, v| [k, (v.nil? || v.empty?) ? k : v]}.to_h
      else {}
      end
    end
```

Do you have a code listing or some other reference that says "library names will always be the same on disk as they are in the library manager, except with spaces replaced by underscores"? I don't want to fail this assumption a second time.

per1234, 2019-08-06:
Here it is:
https://github.com/arduino/Arduino/blob/1.8.9/arduino-core/src/processing/app/BaseNoGui.java#L836-L874

Much of the Java code in the Arduino IDE is in the progress of being replaced by a separate tool (arduino-cli) so that the Java code will be solely for handling the GUI. So in the near future this will be the code used by the Arduino IDE (and already is for arduino-cli users):
https://github.com/arduino/arduino-cli/blob/0.4.0/arduino/utils/filenames.go#L20-L38

The sanitizing code does allow for replacing characters other than space, but  there is an automated system in place that blocks the addition to the Library Manager index of any library with a `name` value that doesn't meet the requirements I quoted in my previous reply, so in effect Library Manager always installs libraries to a folder that matches their `name` value with any spaces replaced with _.

The Library Manager indexer job locks libraries in to the `name` value of the most recent tag at the time the library was submitted for inclusion in the index. If the library author later changes the `name` value in their library.properties and makes a release, the Library Manager still retains the original `name` value.

In the case of the Arduino IDE's **Sketch > Include Library > Add .ZIP Library** installation method, the `name` value is not used at all. The library installation folder is simply the unmodified folder name found in the .zip file and the library author is responsible for giving that folder a valid name.

ianfixes, 2019-08-07:
Excellent, thank you so much!  I am going to merge #130 as soon as @user-name-is-taken can confirm my fixes

PR #130 (2019-08-12): Bugfixes 
## Highlights from `CHANGELOG.md`

### Changed
- Arduino command failures (to read preferences) now causes a fatal error, with help for troubleshooting the underlying command

 ### Fixed
- Arduino library dependencies are now installed prior to unit testing, instead of prior to compilation testing.  Whoops.

## Issues Fixed

* Fixes #128 
* Fixes #129 
* Fixes #132
closed 2019-08-12 by ianfixes