Arduino-CI/arduino_ci #135: Problems with Streams

lasselukkari, 2020-01-24:
First of all, thanks for putting together this amazing project! I has been really useful for me.

Here is my situation. I have a class that inherits the Stream by wrapping another instance of a Stream. I do this so that my code works with any Stream input like Wifi, Ethernet and Serial without any changes. It also to make it compatible with anything else that expects a Stream like JSON parsers and so on. The Stream base class should the same on every Arduino. This is the key feature to write cross platform libraries for all Arduinos.

Because this project implements a non standard Stream where some of the expected functions are missing or mocked with the "GODMODE" testing anything that uses the Stream is impossible. For example, any class that is derived from the Stream should be able to access the protected Stream function `timedRead` or set the `_timeout` property.  

I believe the Stream class should be left as it is with the real platforms. To be honest, in my opinion all the mocks for classes that are derived from Stream or Client are rather pointless. Instead people should write their code to work with the base classes and for testing a mocked class that implement the virtual functions should be used instead. This is the standard practice to write testable code but now it is impossible. I feel that this decision actually encourages bad programming practises. Maybe providing mock for the HardwareSerial is useful because that is normally used directly but atleast the Stream should stay as it is.

I had to fork this project to be able to move forward. I'm fine with that. I just feel that this may cause problems for many others too and may be the wrong approach.

What are your thoughts on this? Have a nice weekend!

ianfixes, 2020-01-24:
Thanks for writing!

The main thing I'm hearing in what you wrote is:

> Bug: protected `Stream` functions like `timedRead` and/or the `_timeout` property are not defined 

I'm changing the issue title to this, and I'd welcome any info you have on how you'd expect these functions to work during testing (as I don't have much personal experience with the deeper parts of `Stream`).  

As far as the need for Stream (as well as _all_ of the mocked classes I provide in this project) is that Arduino's libraries were written for an AVR backend and simply will not compile _at all_ on any other architecture.  

As far as testing, you're right -- in a perfect world people would use dependency injection and isolate themselves from low-level components.  But the Arduino ecosystem was built without testing in mind, so between the hard-coded assumption that you will have a `setup()` and `loop()` entry point and the fact that you can only compile on AVR architecture really limited my options.  

Even the idea of mocking the low-level boards support (i.e. your target architecture is a software emulator) is problematic, because then you have to write one mock per board type. 
 
I still hope against hope that the Arduino folks will push for more than compilation tests.  The Adafruit team seems to be driving that effort and "unit tests" don't seem to be in their vocabulary.  I've mentioned unit testing a few times on the developer mailing list and get absolute silence in response.

lasselukkari, 2020-01-24:
The Stream works just fine if you just copy the existing original Stream.h and Stream.cpp from the Arduino core. This what I did in my own fork. There should be no need to modify the Stream at all and it just causes problems. If you really need to mock for example the HardwareSerial do it just there and leave the Stream as it is.

ianfixes, 2020-07-11:
I received [some helpful information](https://github.com/ianfixes/arduino_ci/issues/140#issuecomment-656966055) that might be relevant to this issue.  At the very least, I can see that my reverse-engineering of the IDE's `g++` invocation wasn't totally accurate.  (Compare the listing in that issue comment with [the flags I use](https://github.com/ianfixes/arduino_ci/blob/master/lib/arduino_ci/cpp_library.rb#L327), `-std=` and `-O1` in particular.)

I think the proper thing for me to do here is take a step back and try to rebuild the mocks based on the _ correct_ set of compiler flags. @lasselukkari is that something you'd be interested in being involved with?

matthijskooijman, 2020-08-26:
I believe that the proper approach here would be to keep `Stream` completely unmodified compared to the Arduino version, since it's just an abstract superclass that can compile on a PC just fine. Then any mocking should be done in subclasses. Something like a `MockedStream` class that extends `Stream` and is again extended by `HardwareSerial` would make sense. It seems there is already `TapedStream` class that occupies this place, but instead of being implemented on top of the standard Arduino `Stream` class (by just implementing the `read` and `write` methods), it seems to be heavily coupled to changes in the `Stream` class.

Also see #145 for some thoughts on further decoupling the mocking behaviour form the object being mocked.

ianfixes, 2020-08-26:
I've opened https://github.com/ianfixes/arduino_ci/issues/146 ... I should have opened it sooner.

lasselukkari, 2021-01-17:
The Arduino project has moved the common pieces to a shared repo: https://github.com/arduino/ArduinoCore-API. IMO these files should be taken there and left as unmodified as possible.

I see that you are working hard currently to add new features. Thanks for that. But I would like to see more simplification and clean up than features. I would like to contribute to the project but unfortunately I do not agree with the general direction at the moment.

ianfixes, 2021-01-18:
It's funny that you bring this up, because I've been looking at revisiting the entire set of C++ mocks.  The feeling of "overcomplicating it" was exactly what I thought of when looking at an ["official" `Arduino.h`](https://github.com/arduino/ArduinoCore-avr/blob/366b1b8/cores/arduino/Arduino.h). 

In the beginning I was just creating compilation mocks as fast as I could and brute-forcing my way through compiler errors.  Things like the hardware need to be mocked, but everything else can/should just use the existing libraries as much as possible!  I think the only snag might be `pgmspace.h` and how it interacts with String and whatnot.  I can't remember exactly how that shook out.

Big picture is that I need to create viable mocks for _each platform_.  For example, [what `avr-libc` provides](https://www.nongnu.org/avr-libc/user-manual/modules.html), which is the general topic of #241 and #251 

>  I do not agree with the general direction at the moment

What do you perceive that direction to be?

lasselukkari, 2021-01-18:
> What do you perceive that direction to be?
Too many unnecessary built in features. 

Personally I would be happy to implement the mocks on my own to the project I'm testing when needed instead of getting everything built in or at least that I could control when they are included. Yes, millis and micros should advance by default for example but I do no need the "godmode".

I would personally only need the files from Arduino CoreApi, a test test runner and a way to compile the examples for selected target platforms. If everything else would be optional I would be really happy.

lasselukkari, 2021-01-18:
> What do you perceive that direction to be?

Too many unnecessary built in features. 

Personally I would be happy to implement the mocks on my own to the project I'm testing when needed instead of getting everything built in or at least that I could control when they are included. Yes, millis and micros should advance by default and it's needed that people can control the return value of the wall clock but I do no need the "godmode".

I would personally only need the files from Arduino CoreApi, a test test runner and a way to compile the examples for selected target platforms. If everything else would be optional I would be really happy. I do understand that in the projects I'm currently testing with this it's easy to abstract the hardware away. This may not always be the case for libraries that interact with the hardware more directly. I just hope that for example I would not have to deal with the mocked built in mocked Stream. Instead in my own project I have this [MockStream.h](https://github.com/lasselukkari/aWOT/blob/master/test/mocks/MockStream.h) It is not a complete solution for all use cases but something like that would be better than directly mocking the Stream, Client or any of the other files that come from the  CoreApi.

I happen to have time to contribute to open source at the moment and I would be pappy to help with this project too. The problems is in the current form I can not use it and had to do my own fork.

ianfixes, 2021-01-18:
I'm sorry that the journey of this project has frustrated you to the point of wanting to abandon it.  I think you and I agree on the need to reduce the complexity of the mocks, and I am certainly planning to revisit a lot of my previous design decisions now that ArduinoCore-API is available (in a way that [it wasn't at the time I started writing this](https://github.com/arduino/ArduinoCore-API/tree/3dbd7d7d338ff2d93c8b9dd8c833967fdb69b7c6)).  

I would absolutely prefer to remove a good chunk of C++ code from this project if I could fall back on a more official version of it.  The only things I want to have mocks for concern the clock and IO devices (pins, SPI, serial, etc) -- to be able to set up deterministic sequences of inputs for repeatable testing.  And the only strong opinion that I have is this: for an existing library to support unit testing, it should not need to be completely rewritten (which seems only fair, since most of the popular resources on how to write libraries do not suggest using unit-testable design patterns).  Arduino is aimed at a group of users who may be programming for the first time, after all.

I can think of 2 possible ways to reconcile my advocacy for "godmode" with your preference to use non-mocked libraries by default and inject your own custom mocks where desired.

The first option would be to allow (in the unit tests) a different `#include` option -- rather than [`#include <Arduino.h>`](https://github.com/Arduino-CI/arduino_ci/blob/master/SampleProjects/TestSomething/test/pinhistory.cpp#L2), we could make it possible to include a file with a different name that offers non-mocked setup.  Or a config file option?  Anything that will satisfy the compiler seems reasonable to me.

The second option, which I'd prefer, (and where I'd like to hear perspectives of Arduino contributors like @matthijskooijman and @per1234) is for ArduinoCore-API to take on features that support [unit testing](https://github.com/arduino/ArduinoCore-API/issues?q=is%3Aissue+is%3Aopen+unit) or [function mocks](https://github.com/arduino/ArduinoCore-API/issues?q=is%3Aissue+is%3Aopen+mock).   In this setup, there would be some API function (or functions) exposed by Arduino that would essentially allow dependency injection for all the builtin functions.  Is that kind of suggestion a non-starter?

per1234, 2021-01-18:
@ianfixes have you taken a look at the ArduinoCore-API unit tests?
https://github.com/arduino/ArduinoCore-API/tree/master/test/src

ArduinoCore-API is all the hardware agnostic code of the core library so it shouldn't really require much mocking. Where it is necessary, there are probably already some of the mocks written because the coverage is up to 96% now.

ianfixes, 2021-01-19:
I'll admit that I'm not 100% clear on how something like their [`HardwareSerial.h`](https://github.com/arduino/ArduinoCore-API/blob/master/api/HardwareSerial.h) could be used for mocking purposes without also dragging in the hard-coded Stream reference.

ianfixes, 2021-01-29:
@per1234 can you comment on my assessment of being unable to mock HardwareSerial without mocking Stream?

per1234, 2021-01-29:
The truth is that, even though I have some experience with unit tests and mocking in other contexts, I really don't understand much of what is happening in this project. I was hoping to pick up some understanding by osmosis, but I've been watching this repo for years now without getting very far.

ianfixes, 2021-01-29:
I can honestly say the same about the Arduino C++ infrastructure for boards and platforms 😉 

What's the right forum where I could engage with the Arduino folks for a deeper dive on this sort of thing?

per1234, 2021-02-02:
I think probably you are already using the available communication channels so I don't have anything else to suggest.

Arduino forum: Arduino users, but mostly not monitored by anyone from Arduino other than myself.

Arduino developers mailing list: A subset of the 3rd party developers. Seems to be fairly well monitored by Arduino, even when not actively participating in the conversations.

There is an official Discord server: https://discord.gg/jQJFwW7. I have only taken a quick look at it so I can't say whether it would be of any use to you.