Arduino-CI/arduino_ci #136: Millis() and micros() can’t be used for detecting timeouts

lasselukkari, 2020-01-24:
I believe one of the most common use cases for the time related functions `millis()` and `micros()` is to detect timeouts. In a nutshell you take the current time and loop until you succeed on something or bail out when you hit the timeout. An example of such a loop can be seen here:
https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/Stream.cpp#L31-L40

Many people use the same pattern in their libraries:
https://github.com/arduino-libraries/ArduinoHttpClient/blob/master/src/HttpClient.cpp#L414-L420

The problem is that the return value of millis() does not advance between calls if it’s not manually set. When calling a function that expects the clock to go forward it will never return because it is impossible to set a new value for the millis while the other function is being executed.

An easy fix would to increment the return values of millis() and micros() by one on every call. If needed this behaviour could be enabled or disabled.

ianfixes, 2020-01-24:
I would probably end up supporting a set of `GODMODE` callbacks for any of these system functions being hit, that allows the end user to adjust the time (or for that matter, other inputs) by some amount.

ianfixes, 2020-06-20:
Did this stop being an issue?  I think it's a valuable addition although due to the pandemic I've been less able to make progress on this project.

lasselukkari, 2020-06-20:
Yes, it is still an issue.