Arduino-CI/arduino_ci #139: Example sketches with tests?

dhempy, 2020-04-13:
## (Issue Summary / Feature Request)

Can anyone point to some open repos that use arduino_ci? The documentation here seems quite complete at the detail level...but a little challenging to put all together for a newbie Arduino developer. 

In my case, I'm thoroughly comfortable with unit testing from years of Ruby testing with rspec. I'm just struggling with a bit of the glue to bring it all together on this platform. Once I see the patterns in place and get my first test running, I'm sure all the detail docs will answer my questions.

I'm happy to put contribute to to documentation if you all can point to a few good quality examples.

Thanks in advance.

ianfixes, 2020-04-13:
> The documentation here seems quite complete at the detail level...but a little challenging to put all together for a newbie Arduino developer.

That's pretty accurate.  Did the documentation at least take you as far as the [`DoSomething` sample project](https://github.com/ianfixes/arduino_ci/tree/master/SampleProjects/DoSomething)?  That is meant to be a bare-bones example of how to put the test fixtures in place.  

As far as open source projects, I opened pull requests on a few of them since I had used those libraries as test cases for developing my library:

* https://github.com/PaulStoffregen/OneWire/pull/67
* https://github.com/adafruit/Adafruit_FONA/pull/91
* https://github.com/adafruit/Adafruit-WS2801-Library/pull/21
* https://github.com/SMFSW/Queue/pull/9
* https://github.com/sdesalas/Arduino-Queue.h/pull/1
* https://github.com/bguest/AcceleratedLED/pull/1

Is this more or less what you're looking for?  I'd gladly accept a `GettingStarted.md` from an actual newbie's perspective!

dhempy, 2020-04-13:
Thanks for those links, Ian. I think I found the README for DoSomething...but perhaps glossed over the fact that it was just the readme, and there was a whole project underneath it to explore. *blush*

I did not find the TestSomething project. After a few minutes of browsing, I'm still not clear on the difference between the two, but I'll spend some time exploring both. 

And thanks for the links to real-world projects using your tool. I'll look at a few of them, as well. 

I'm totally game to write up something if it is fruitful.

dhempy, 2020-04-13:
ps. This is the project we're building: https://github.com/dhempy/air-assist
It is an open-source ventilator for COVID-19 overflow patient. You can learn more about it here: https://www.vent.guide/

ianfixes, 2020-04-14:
TestSomething is more of a dummy project that I use to test the `arduino_ci` library itself ... I have a few hardware mocks and this was the easiest way to unit test them.  I don't think it would offer much in the way of good examples for what you're working on.

As far as your code goes, `arduino_ci` can only really work on an arduino library, not an arduino sketch.  So if you factor your code into classes such that the standard arduino functions like `setup()` and `loop()` are just thin wrappers around library functions.

ianfixes, 2020-04-14:
> but perhaps glossed over the fact that I was only looking at the readme, missing the fact that there was a whole project underneath it to explore

That's a good indication that the README could use some improving!  I'll take any suggestions you have there for how to make the example-project-ness of it stand out.

jgfoster, 2020-09-09:
@dhempy, Having faced the same learning curve, I developed [Blink](https://github.com/jgfoster/Blink) as a template. You can also watch a [video](https://youtu.be/vaS4AU_PXQo) of the process. 

@ianfixes, At first I was intending to add Blink to your SampleProjects, but now that I understand that we can _use_ arduino_ci without downloading this project it makes more sense to me to have a small demo/template people can use to start. Does that make sense? 

Advice, suggestions, and corrections are welcome.

ianfixes, 2020-09-09:
> we _can_ use arduino_ci without downloading this project

I'm calling out this point in particular because it sounds like I don't do a good job in the docs of explaining precisely this.

> makes more sense to me to have a small demo/template [i.e. "Blink"] people can use to start.

I agree completely.  The problem for me was that as this project evolved, it was difficult to keep the sample project in proper sync with the feature set (I'd change a function name in the codebase but not the sample project).  So by testing the sample projects as part of `arduino_ci`'s _own_ CI, it helped me keep those updated.  At the expense of standalone examples.

But conceptually, you're right -- it leads to confusion.  I think a standalone repo is probably the best way to accomplish the starter template idea you're talking about.

ianfixes, 2021-01-11:
@dhempy I assume that https://github.com/Arduino-CI/arduino_ci/issues/139#issuecomment-689235778 will satisfy your needs here, if not then please reopen with more information about what is missing

closed 2021-01-11 by ianfixes

dhempy, 2021-01-21:
Thank you, @ianfixes  and @jgfoster !