Arduino-CI/arduino_ci #140: missing definitions for zero

FedeBev, 2020-06-13:
Hi there,
thanks for this work!

## System

- OS: linux (ubuntu 18.04)
- `ruby -v`: ruby 2.5.1p57 (2018-03-29 revision 63029) [x86_64-linux-gnu]
- `bundle -v`: Bundler version 2.1.4
- `bundle info arduino_ci`: arduino_ci (0.3.0)
- `g++ -v`: gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04) 
- Arduino IDE version: 1.8.12
- `java -version`: openjdk version "1.8.0_252"

## Issue Summary
I'm testing a code that on arduino (physical board) mkr 1010 compiles with no issues, however with *arduino-ci* is failing with the following errors

```
Last command:  $ g++ -std=c++0x -o <my_path>/cronparser/unittest_test.cpp.bin -DARDUINO=100 -g -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address -D__SAMD21G18A__ -DARDUINO_SAMD_ZERO -I/var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino -I/var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/unittest -I<my_path>/cronparser /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/Arduino.cpp /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/Godmode.cpp /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/stdlib.cpp /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/unittest/ArduinoUnitTests.cpp <my_path>r/cronparser/cronparser.cpp /<my_path>/cronparser/test/test.cpp

In file included from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/avr/pgmspace.h:17:0,
                 from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/ArduinoDefines.h:3,
                 from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/Godmode.h:2,
                 from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/Arduino.cpp:1:
/var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/avr/io.h:623:6: warning: #warning "device type not defined" [-Wcpp]
 #    warning "device type not defined"
      ^~~~~~~
In file included from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/avr/pgmspace.h:17:0,
                 from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/ArduinoDefines.h:3,
                 from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/Godmode.h:2,
                 from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/Godmode.cpp:1:
/var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/avr/io.h:623:6: warning: #warning "device type not defined" [-Wcpp]
 #    warning "device type not defined"
      ^~~~~~~
In file included from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/Godmode.cpp:3:0:
/var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/SPI.h: In member function ‘uint16_t SPIClass::transfer16(uint16_t)’:
/var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/SPI.h:97:11: error: ‘SPCR’ was not declared in this scope
     if (!(SPCR & (1 << DORD))) {
           ^~~~
/var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/SPI.h:97:11: note: suggested alternative: ‘SPL’
     if (!(SPCR & (1 << DORD))) {
           ^~~~
           SPL
/var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/SPI.h:97:24: error: ‘DORD’ was not declared in this scope
     if (!(SPCR & (1 << DORD))) {
                        ^~~~
In file included from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/avr/pgmspace.h:17:0,
                 from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/unittest/Compare.h:2,
                 from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/unittest/Assertion.h:7,
                 from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/unittest/ArduinoUnitTests.h:4,
                 from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/unittest/ArduinoUnitTests.cpp:1:
/var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/avr/io.h:623:6: warning: #warning "device type not defined" [-Wcpp]
 #    warning "device type not defined"
      ^~~~~~~
In file included from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/avr/pgmspace.h:17:0,
                 from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/unittest/Compare.h:2,
                 from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/unittest/Assertion.h:7,
                 from /var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/unittest/ArduinoUnitTests.h:4,
                 from /mnt/Dati/Progetti/smart-water/cronparser/test/test.cpp:1:
/var/lib/gems/2.5.0/gems/arduino_ci-0.3.0/cpp/arduino/avr/io.h:623:6: warning: #warning "device type not defined" [-Wcpp]
 #    warning "device type not defined"
      ^~~~~~~
```

Works on the following platforms:
- uno
- due
- mega

## Arduino or Unit Test Code, Illustrating the Problem
The issue seem to be related to the import of `WString.h`. My code uses nothing strictly related to the SAMD processor, indeed here is the hole import list

```c++
#include <time.h>
#include <string.h>
#include <WString.h>
```

## Arduino Architecture(s) Affected

zero, as far as i know.

Thanks!

ianfixes, 2020-06-14:
I assume this is related to an unsupported architecture (or more accurately, one that I haven't yet defined in the default config).  What's in your `.arduino_ci.yml`?

FedeBev, 2020-06-15:
Thanks for the quick feedback!

According to the [default definitions for release 0.3.0](https://github.com/ianfixes/arduino_ci/blob/v0.3.0/misc/default.yml) the zero board should be defined, but maybe I'm missing something.

The following config works like a charm
```yaml
compile:
  libraries: ~
  platforms:
    - uno
    - due
    - mega2560
    - leonardo
    # - zero

unittest:
  libraries: ~
  platforms:
    - uno
    - due
    - mega2560
    - leonardo
    # - zero
```

but adding _zero_ board all the error are shown

```yaml
compile:
  libraries: ~
  platforms:
    - uno
    - due
    - mega2560
    - leonardo
    - zero

unittest:
  libraries: ~
  platforms:
    - uno
    - due
    - mega2560
    - leonardo
    - zero
```

ianfixes, 2020-06-19:
I think I see the problem, but I'm not sure how to fix it.  Somewhere in [`cpp/arduino/avr/`](https://github.com/ianfixes/arduino_ci/tree/master/cpp/arduino/avr) there needs to be a  file that has all the `#define`s for that board, and something in [`io.h`](https://github.com/ianfixes/arduino_ci/blob/master/cpp/arduino/avr/io.h) that refers to `__SAMD21G18A__`.

I'm having trouble getting any good google results for that, which is frustrating to say the least.  I assume that SAMD boards are incompatible with any of the list of `__AVR_`* that is in there now.

FedeBev, 2020-06-22:
Hi @ianfixes, thanks for your interest.

Can this [stackoverflow question](https://stackoverflow.com/questions/2224334/gcc-dump-preprocessor-defines) help with the issue?
As far as I know there isn't a marco definitions list for each board (that's frustrating too), thus I'm looking for a way to reverse engineering the problem

ianfixes, 2020-06-22:
The problem is that I would have to find a way to dump the preprocessor defines from within the Arduino IDE, to test what happens when you select the zero.  I guess I could brute-force it by just making a dummy sketch that prints out all the constants I find in `io.h`.  That's not ideal because it won't give me any sense of what the "top-level" constant to be defined is.  But it will be a start.

FedeBev, 2020-06-22:
I'll look for a way too.
Since the arduino tools are all open source I'm trying to find something in the arduino-cli source code.
If you need to test something on some MKR board just let me know.

FedeBev, 2020-07-08:
Dealing with another issue (not related to this project) I ran into the following line printed in the IDE console

```
/home/.../snap/arduino/41/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -Wall -Wextra -Wno-expansion-to-defined -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=48000000L -DARDUINO=10813 -DARDUINO_SAMD_MKRWIFI1010 -DARDUINO_ARCH_SAMD -DUSE_ARDUINO_MKR_PIN_LAYOUT -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8054 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino MKR WiFi 1010\"" -DUSE_BQ24195L_PMIC -I/home/.../snap/arduino/41/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/home/.../snap/arduino/41/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/ -I/home/.../snap/arduino/41/.arduino15/packages/arduino/hardware/samd/1.8.6/cores/arduino -I/home/.../snap/arduino/41/.arduino15/packages/arduino/hardware/samd/1.8.6/variants/mkrwifi1010 -I/home/.../snap/arduino/current/Arduino/libraries/TaskScheduler/src -I/home/.../snap/arduino/current/Arduino/libraries/WiFiNINA/src -I/home/.../snap/arduino/current/Arduino/libraries/RTCZero/src -I/home/.../snap/arduino/current/Arduino/libraries/StreamUtils/src -I/home/.../snap/arduino/41/.arduino15/packages/arduino/hardware/samd/1.8.6/libraries/SPI /home/.../snap/arduino/current/Arduino/libraries/RTCZero/src/RTCZero.cpp -o /tmp/arduino_build_827430/libraries/RTCZero/RTCZero.cpp.o
```

the intresting part of that is
```
 -DARDUINO=10813 -DARDUINO_SAMD_MKRWIFI1010 -DARDUINO_ARCH_SAMD -DUSE_ARDUINO_MKR_PIN_LAYOUT -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8054 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino MKR WiFi 1010\"" -DUSE_BQ24195L_PMIC
```

@ianfixes can you see something useful in the definitions above for this issue?

ianfixes, 2020-07-11:
Sure, reformatting that:

```
 -DARDUINO=10813 
 -DARDUINO_SAMD_MKRWIFI1010 
 -DARDUINO_ARCH_SAMD 
 -DUSE_ARDUINO_MKR_PIN_LAYOUT 
 -D__SAMD21G18A__ 
 -DUSB_VID=0x2341 
 -DUSB_PID=0x8054 
 -DUSBCON 
"-DUSB_MANUFACTURER=\"Arduino LLC\"" 
"-DUSB_PRODUCT=\"Arduino MKR WiFi 1010\""
 -DUSE_BQ24195L_PMIC
```

My [current platform definition for the Arduino zero](https://github.com/ianfixes/arduino_ci/blob/master/misc/default.yml#L37-L46) looks like this:

```yml
  zero:
    board: arduino:samd:arduino_zero_native
    package: arduino:samd
    gcc:
      features:
      defines:
       - __SAMD21G18A__
       - ARDUINO_SAMD_ZERO
      warnings:
      flags:
```

So it looks like we can test out these settings, by adding this to your own `.arduino_ci.yml`:

```yml
platforms:
  zero:
    board: arduino:samd:arduino_zero_native
    package: arduino:samd
    gcc:
      features:
      defines:
       - ARDUINO_ARCH_SAMD
       - ARDUINO_SAMD_MKRWIFI1010
       - ARDUINO=10813
       - __SAMD21G18A__
       - ARDUINO_SAMD_ZERO
       - USE_ARDUINO_MKR_PIN_LAYOUT
       - USB_VID=0x2341
       - USB_PID=0x8054
       - USBCON
       - USE_BQ24195L_PMIC
      warnings:
      flags:
```

(This will override the defaults for the `zero` platform.)  When I do this, I still get the same error. This makes me think that I'm missing a header that's relevant to SAMD.... possibly something in `-I/home/.../snap/arduino/41/.arduino15/packages/arduino/hardware/samd/1.8.6/cores/arduino`!

But this is very interesting.

ianfixes, 2020-07-11:
For now, I'll assume that _all_ of this is the interesting part and reformat it for future reference

```
/home/.../snap/arduino/41/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ 
-mcpu=cortex-m0plus 
-mthumb 
-c 
-g 
-Os 
-Wall 
-Wextra
-Wno-expansion-to-defined 
-std=gnu++11 
-ffunction-sections 
-fdata-sections 
-fno-threadsafe-statics 
-nostdlib 
--param max-inline-insns-single=500 
-fno-rtti 
-fno-exceptions 
-MMD 
-DF_CPU=48000000L 
-DARDUINO=10813 
-DARDUINO_SAMD_MKRWIFI1010 
-DARDUINO_ARCH_SAMD 
-DUSE_ARDUINO_MKR_PIN_LAYOUT 
-D__SAMD21G18A__ 
-DUSB_VID=0x2341 
-DUSB_PID=0x8054 
-DUSBCON 
"-DUSB_MANUFACTURER=\"Arduino LLC\"" 
"-DUSB_PRODUCT=\"Arduino MKR WiFi 1010\"" 
-DUSE_BQ24195L_PMIC 
-I/home/.../snap/arduino/41/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ 
-I/home/.../snap/arduino/41/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/ 
-I/home/.../snap/arduino/41/.arduino15/packages/arduino/hardware/samd/1.8.6/cores/arduino 
-I/home/.../snap/arduino/41/.arduino15/packages/arduino/hardware/samd/1.8.6/variants/mkrwifi1010 
-I/home/.../snap/arduino/current/Arduino/libraries/TaskScheduler/src 
-I/home/.../snap/arduino/current/Arduino/libraries/WiFiNINA/src 
-I/home/.../snap/arduino/current/Arduino/libraries/RTCZero/src 
-I/home/.../snap/arduino/current/Arduino/libraries/StreamUtils/src 
-I/home/.../snap/arduino/41/.arduino15/packages/arduino/hardware/samd/1.8.6/libraries/SPI 
/home/.../snap/arduino/current/Arduino/libraries/RTCZero/src/RTCZero.cpp 
-o /tmp/arduino_build_827430/libraries/RTCZero/RTCZero.cpp.o
```

PR #183 (2020-11-13): Suggested fixes
### Trying it out
You can test this beta of the gem directly from github by putting this in your `Gemfile`:

```ruby
gem 'arduino_ci', git: 'https://github.com/ianfixes/arduino_ci.git', branch: '2020-10-16_suggestions'
```

## Highlights from `CHANGELOG.md`
### Added
- `arduino_ci_remote.rb` CLI switch `--skip-examples-compilation`
- `CppLibrary.header_files` to find header files
- `LibraryProperties` to read metadata from Arduino libraries
- `CppLibrary.library_properties_path`, `CppLibrary.library_properties?`, `CppLibrary.library_properties` to expose library properties of a [... cut]
commit 7d7d8103 (2020-11-13): Make SPI.h work on non-AVR unittests

The SPI library contains some code to read the SPCR register to
determine the byte order of 16-bit transfers, presumably because there
is no official Arduino API to set it, so sketches had to rely on setting
this register directly. By reading it, the SPI unittest implementation
can detect how to emulate multibyte transfers.

However, this register is only defined by avr/io.h when the unittest
emulates an AVR platform, so this code would fail to compile on other
platforms.

This adds a preprocessor guard around this code, defaulting to the
lsb-first, which  [... cut]
closed 2020-11-13 by ianfixes