Arduino-CI/arduino_ci #143: Godmode should probably use std::string (or STL container maybe) for SPI/Wire/UART data

matthijskooijman, 2020-08-26:
The Arduino `String` class is only intended for string data currently. In particular, it cannot handle embedded 0x0 bytes, which you might very well want to send or receive. See https://github.com/arduino/ArduinoCore-API/issues/111 and https://github.com/arduino/ArduinoCore-API/pull/97. If those issues are fixed, the String class could still be used, though.

ianfixes, 2020-08-26:
Can you add some context here?  Is this an area where arduino_ci deviates from the standard Arduino library currently?  Or is it at risk of deviating in the future? 

I had some trouble using `std::` anything when compiling, as we are essentially mocking the AVR behavior on non-AVR... which was frustrating.  If fixing #146 enables `std::string` to be used, I'm all for it.

matthijskooijman, 2020-08-27:
> Is this an area where arduino_ci deviates from the standard Arduino library currently?

No, the Arduino String class is just broken when it comes to embedded nul bytes. Internally, the class uses some standard libc functions (e.g. things like `strcpy`) that will stop processing at the first embedded nul, so things might break when you use the `String` class for arbitrary binary data, rather than actual strings.

Also, since the Godmode stuff is only for unittests (i.e. it never has to actually run on AVR/MCU), it can use non-Arduino classes just normally (though using the Arduino String class could make interaction with other library code easier, of course).

> I had some trouble using std:: anything when compiling, as we are essentially mocking the AVR behavior on non-AVR... 

Hm, I haven't actually tried `std::string` yet, I'll have a look at #146 then.