Arduino-CI/arduino_ci #155: Downloading old version of IDE

jgfoster, 2020-09-08:
## System: macOS

See #149 for details.

## Downloading 1.8.6 instead of most recent (currently 1.8.13)

This is a follow-up from the "bad CPU type in executable" problem reported in #149. If a copy of the IDE is not found, we download one. Unfortunately, the (hard-coded?) version obtained (1.8.6) does not work on macOS 10.16 because it is not a 64-bit application. 

## Workaround

Delete the Arduino.app in the Documents directory and install the current version in the /Applications directory.

jgfoster, 2020-09-27:
Another work-around is to edit `./vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/lib/arduino_ci/arduino_installation.rb` to update `DESIRED_ARDUINO_IDE_VERSION`.

Would it be reasonable to allow this to be configurable in `.arduino-ci.yml`?

ianfixes, 2020-09-28:
I'm trying to put a fine point on what the fix should be here.  `arduino_ci` is only one part of the CI configuration -- even though it's often invoked as a one-liner in the sample scripts, it's not meant to be a replacement for the shell scripting that one would have to perform in a `.travis.yml` (or similar) file.  So the automatic installation of libraries or the IDE itself is meant as a failsafe or a convenience and not a replacement for explicit environment setup.

`DESIRED_ARDUINO_IDE_VERSION` is really a guarantee that the version we download matches the CLI behavior that `arduino_ci` codes against -- I'd be nervous about allowing that to be set to an arbitrary value as I have no guarantee that the CLI will be compatible.

Perhaps the better solution here is to detect 64-bit-only OSes and handle that.  And possibly to bump the supported version from `1.8.6` to a more modern version.  Does `1.8.13` download & run correctly in 32-bit OSes?

jgfoster, 2020-09-28:
According to the [downloads](https://www.arduino.cc/en/main/software) page, Arduino IDE version 1.8.13 runs on Window 7 and later, macOX 10.10 and later, and Linux 32-bit and 64-bit. While I have Windows 10 64-bit, when I run the Arduino IDE installer it puts the executable in `C:\Program Files (x86)` which is the 32-bit directory so I assume that it is running a 32-bit app even on the 64-bit OS. Thus, I think updating to 1.8.13 would be safe and sufficient and this issue could be closed.

per1234, 2020-09-28:
I haven't tried 32 bit, but I closely follow the Arduino forum and the Arduino IDE's issue tracker, and this is the sort of thing you definitely hear about when compatibility is lost. So I think there is no problem with new IDE versions and 32 bit OS.

The most recent losses I'm aware of were Windows XP/Vista and the older macOS versions, which was caused by Go dropping support, and subsequently breaking compatibility with those OS versions:
- https://github.com/arduino/Arduino/issues/10018
- https://github.com/arduino/Arduino/issues/10047

ianfixes, 2020-11-16:
Well... we're off to a rocky start, immediately encountering https://github.com/arduino/Arduino/issues/9351

I was planning to go all-in on #206 after bumping the version, so this is a setback. But this seems like the appropriate next step.

closed 2020-11-27 by ianfixes