Arduino-CI/arduino_ci #157: Suggest consistent use of .yml or .yaml

jgfoster, 2020-09-08:
SampleProjects/DoSomething/.arduino-ci.yaml vs. SampleProjects/TestSomething/.arduino-ci.yml (and many others).

See https://stackoverflow.com/questions/21059124/is-it-yaml-or-yml.

I don't feel at all strongly about this, it is just something that was confusing for a few moments.

ianfixes, 2020-09-08:
I explicitly allow both because of how frustrating it was to troubleshoot filenames that had the wrong extension

https://github.com/ianfixes/arduino_ci/blob/master/lib/arduino_ci/ci_config.rb#L41-L46

If we can solve for the troubleshooting scenario ("my arduino_ci isn't doing what I configured it to do and I can't tell why!"), then it would be easier to justify picking a single spelling.  

TL;DR the confusion is real 😅

jgfoster, 2020-09-27:
I now understand that you use both in your own code so that you verify that your own code works with both. That seems reasonable (though it initially seemed sloppy, now I see that it was intentional!). I hope it is okay if I close this since I asked the question initially.

closed 2020-09-27 by jgfoster