Arduino-CI/arduino_ci #158: "device type not defined" when testing without .arduino-ci.yml

jgfoster, 2020-09-08:
## System

See #149 for system details.

## "device type not defined" errors

Based on ./REFERENCE.md I believe that `.arduino-ci.yml` is optional, but when I left it out I got an error. I added it back and gave it the values found in `misc/default.yml` and the error reproduced. By making changes to the file I gradually reduced it to see that the error reproduces with `zero` as the only entries in the `platforms` section. I don't need zero (I only use Mega2560) but I was trying to do something that was generalized.

There are many errors, but following is the first:

```
Requested unittest platform 'zero' is defined in 'platforms' YML...            ✓
Unit testing test.cpp with g++... 

Last command:  $ g++ -std=c++0x -o /Users/jfoster/Documents/Arduino/libraries/Blink/unittest_test.cpp.bin -DARDUINO=100 -g -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address -D__SAMD21G18A__ -DARDUINO_SAMD_ZERO -I/Users/jfoster/Documents/Arduino/libraries/Blink/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/arduino -I/Users/jfoster/Documents/Arduino/libraries/Blink/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/unittest -I/Users/jfoster/Documents/Arduino/libraries/Blink /Users/jfoster/Documents/Arduino/libraries/Blink/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/arduino/Arduino.cpp /Users/jfoster/Documents/Arduino/libraries/Blink/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/arduino/Godmode.cpp /Users/jfoster/Documents/Arduino/libraries/Blink/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/arduino/stdlib.cpp /Users/jfoster/Documents/Arduino/libraries/Blink/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/unittest/ArduinoUnitTests.cpp /Users/jfoster/Documents/Arduino/libraries/Blink/Blink.cpp /Users/jfoster/Documents/Arduino/libraries/Blink/test/test.cpp

In file included from /Users/jfoster/Documents/Arduino/libraries/Blink/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/arduino/Arduino.cpp:1:
In file included from /Users/jfoster/Documents/Arduino/libraries/Blink/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/arduino/Godmode.h:2:
In file included from /Users/jfoster/Documents/Arduino/libraries/Blink/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/arduino/ArduinoDefines.h:3:
In file included from /Users/jfoster/Documents/Arduino/libraries/Blink/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/arduino/avr/pgmspace.h:17:
/Users/jfoster/Documents/Arduino/libraries/Blink/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/arduino/avr/io.h:623:6: warning: "device type not defined" [-W#warnings]
#    warning "device type not defined"
```

## Arduino or Unit Test Code, Illustrating the Problem

## Arduino Architecture(s) Affected
- zero

ianfixes, 2020-09-08:
Can you post the contents of the config file that produced errors?

jgfoster, 2020-09-08:
`.arduino-ci.yaml` 

```
compile:
  libraries: ~
  platforms:
    - zero

unittest:
  libraries: ~
  platforms:
    - zero
```

ianfixes, 2020-09-09:
I can think of 2 things that could be going wrong here.

1. There is a bug in the library and I'm not actually pulling in [`default.yml`](https://github.com/ianfixes/arduino_ci/blob/master/misc/default.yml) as the "base" configuration. But [I'm pretty sure I test for this](https://github.com/ianfixes/arduino_ci/blob/master/spec/ci_config_spec.rb#L4-L22).
2.  `.arduino-ci.yml` files can exist in multiple places in the directory structure, such that [deeper ones override shallower ones](https://github.com/ianfixes/arduino_ci/blob/master/REFERENCE.md#indirectly-overriding-build-behavior-medium-term-use-and-advanced-options).  Perhaps you copied [this example configuration containing the entry `zero: ~` into your project somewhere](https://github.com/ianfixes/arduino_ci/blob/master/REFERENCE.md#defining-new-arduino-platforms)?

jgfoster, 2020-09-09:
Should `zero` work? When I use `mega2560` as the platform, the test passes; but when I use `zero` as the platform I get the error. So, yes, it only happens if I explicitly (with my own config file) or implicitly (by using the default) include `zero` as a platform.

ianfixes, 2020-09-09:
I would expect that either _all_ the platforms in `default.yml` work in a CI context, or _none_ of them.  But I'm looking again... and 
> `Requested unittest platform 'zero' is defined in 'platforms' YML...            ✓`

So this was a red herring, my fault.  Also:

> Should `zero` work?

Now that you mention it... this is indeed a bug.  But a duplicate bug of #140 😞

ianfixes, 2020-09-09:
In the short term I'm wondering if I should just disable the entry in default.yml, but that really won't help all the gem versions that are already in the wild.

closed 2020-09-09 by ianfixes