Arduino-CI/arduino_ci #166: Support for EEPROM

jgfoster, 2020-10-02:
As @per1234 explained [here](https://github.com/Arduino-CI/arduino_ci/issues/150#issuecomment-692444392), [SPI](https://www.arduino.cc/en/Reference/SPI) and [Wire](https://www.arduino.cc/en/Reference/wire) are [platform bundled libraries](https://arduino.github.io/arduino-cli/latest/platform-specification/#platform-bundled-libraries) and I see that mocks are included for them in `arduino_ci`. Is [EEPROM](https://www.arduino.cc/en/Reference/EEPROM) similar? Should it be part of the base or an add-on like [LiquidCrystal](https://github.com/Arduino-CI/LiquidCrystal)?

per1234, 2020-10-03:
The primary difference between the EEPROM vs SPI and Wire libraries is that not all boards platforms have a bundled EEPROM library. The reason being that not all microcontrollers have EEPROM. Some boards platforms of microcontrollers without EEPROM provide an EEPROM library that uses flash memory to simulate EEPROM ([example](https://arduino-esp8266.readthedocs.io/en/latest/libraries.html#eeprom)).

Platform | Has EEPROM library
-|-
arduino:avr | Y
arduino:megaavr | Y
arduino:sam | N
arduino:samd | N
arduino:mbed | N
arduino-beta:mbed | N
arduino:arc32 | Y

ianfixes, 2020-10-05:
Is this "bundled library" characteristic related to #140 ?

per1234, 2020-10-05:
No. It only means that the library is only available when compiling for one of the boards of the platform the library is bundled with.

For example, if you have the Arduino AVR Boards installed, then the EEPROM library is also installed and you can use it when compiling for the Uno. However, if you then try to compile a sketch that uses the EEPROM library for the Zero, compilation will fail:
```
fatal error: EEPROM.h: No such file or directory

 #include <EEPROM.h>
          ^~~~~~~~~~
```
The reason for this is that the library is AVR-specific. Making it accessible to other platforms would only be harmful, since it won't compile for other architectures.

jgfoster, 2020-10-05:
How can I tell if `EEPROM` is supported by the board? In `cpp/arduino/arv/` there are over 40 files that include `#define EEPROM_SIZE` but when I build for `mega2560` that macro is not defined. It seems that it ought to be defined. Shall I just add it to `iom2560.h`?

per1234, 2020-10-05:
AVR has an `E2END` macro:
https://www.nongnu.org/avr-libc/user-manual/group__avr__io.html

As for a way to determine EEPROM support for any arbitrary board of any architecture, I don't know.

Another thing to consider might be that there is a significant difference in the way the ESP8266 and ESP32 libraries work. A sketch that uses the EEPROM library perfectly on AVR won't work at all on the Espressif boards if it doesn't call EEPROM.begin() and EEPROM.commit()/EEPROM.end() in order to write the data to flash. I don't know whether that's outside the scope of the project.

ianfixes, 2020-10-06:
It might be worth instrumenting the `.begin()` / `.end()` in `GODMODE`, if nothing else, so that library writers can validate their assumptions about these functions being called.  I'm not familiar enough with how that's used in practice to say how that should be implemented though.

per1234, 2020-10-06:
Documentation of the additional API of the ESP9266/ESP32 boards platforms that extends the official Arduino EEPROM library API is here:
https://arduino-esp8266.readthedocs.io/en/2.7.4_a/libraries.html#eeprom

PR #198 (2020-11-10): Rewrite EEPROM.h to avoid GNU-licensed code
Fixes #166 (along with #184).
PR #183 (2020-11-13): Suggested fixes
### Trying it out
You can test this beta of the gem directly from github by putting this in your `Gemfile`:

```ruby
gem 'arduino_ci', git: 'https://github.com/ianfixes/arduino_ci.git', branch: '2020-10-16_suggestions'
```

## Highlights from `CHANGELOG.md`
### Added
- `arduino_ci_remote.rb` CLI switch `--skip-examples-compilation`
- `CppLibrary.header_files` to find header files
- `LibraryProperties` to read metadata from Arduino libraries
- `CppLibrary.library_properties_path`, `CppLibrary.library_properties?`, `CppLibrary.library_properties` to expose library properties of a [... cut]
closed 2020-11-13 by ianfixes