Arduino-CI/arduino_ci #168: Multiple definition of 'main' when including a library with tests

rosogon, 2020-10-03:
## System

- OS: Linux
- `ruby -v`: ruby 2.3.1p112
- `bundle -v`: Bundler version 2.1.4
- `bundle info arduino_ci`: arduino_ci (0.3.0)
- `g++ -v`: gcc version 5.4.0 20160609
- URL of your Arduino project: https://github.com/rosogon/arduino_ci_two_libraries

## (Issue Summary / Feature Request)

If libB depends on libA, both with arduino_ci tests, trying to run the unit tests on libB will raise the error "multiple definition of main". This happens because [cpp_files_libraries](https://github.com/Arduino-CI/arduino_ci/blob/master/lib/arduino_ci/cpp_library.rb#L188) returns the whole tree of the referenced library (libA in the example), not removing the files in test/.

Con: Skipping those files can break libraries containing files in test/ (which are needed to use the library). Weird, but could happen.

## Arduino or Unit Test Code, Illustrating the Problem

The following repo illustratres the example: https://github.com/rosogon/arduino_ci_two_libraries

rosogon, 2020-10-03:
I have modified it like this: https://github.com/rosogon/arduino_ci/commit/640201afb14bfcd80a2ecd0a17e1679b87e9b92e

If you agree with the approach, I can prepare tests and PR.

ianfixes, 2020-10-04:
Hi rosogon, thanks for contributing this!  And I'm sorry to hear about this limitation of the library, that's really annoying.  

At the moment I'm torn -- I have a 2+ year old feature request for a `test/` directory to be officially adopted by the Arduino library specification.  I know this is pedantic of me, but I'm hesitant to hard-code a directory to exclude from the compiler (even one that conforms to a convention that I wholeheartedly support) because I can't guarantee that _no arduino library anywhere_ might be using that directory for actual code.

Really there are 2 choices here: 1 is whether there is going to be some official movement toward `test/` being "official" in the near future (I just created #169 to track that).  If so, I will absolutely adopt the change you suggest.  The second-best option would be more cumbersome... adding settings to `.arduinoci.yml` that allow you to hand-code selection/rejection criteria for each of the dependencies:

* https://github.com/Arduino-CI/arduino_ci/blob/master/REFERENCE.md#control-how-unit-tests-are-compiled-and-run
* https://github.com/Arduino-CI/arduino_ci/blob/master/lib/arduino_ci/ci_config.rb#L301

I saw that @per1234 happened to make some moves on this in the past few days, I'm not sure if there is extra information there.

rosogon, 2020-10-04:
Hi @ianfixes , 

Thanks for your detailed answer. I understand your concerns. 

While we wait until that `test/` is official (let's hope it is), in https://github.com/rosogon/arduino_ci/commit/5b0b5bc115c52301ba39720d37e14ac27dca5c3f you have another approach using the `exclude_dirs` field to reject directories in the included libraries. Another field could be used (exclude_library_dirs?) to make it safer. Let me know your opinion.

ianfixes, 2020-10-05:
Yes exactly -- this is the model I would use for the fallback solution.  It might make sense to just institute that functionality now, but the question is of how best to do it.  What I'm leaning toward right now is a more robust handling of the "libraries" section, from a simple array to a more ambitious hash.  

In other words, not only do I think it's important to add what you're suggesting, I think that there are a _lot_ of things that could be improved by configuring the individual libraries.  And it would be better to tie that to the rest of the library config instead of the compiler specifically.

If you were looking for a quick fix, I'm sorry to be taking you down this rabbit hole.

matthijskooijman, 2020-10-06:
>  I know this is pedantic of me, but I'm hesitant to hard-code a directory to exclude from the compiler (even one that conforms to a convention that I wholeheartedly support) because I can't guarantee that no arduino library anywhere might be using that directory for actual code.

I actually think that you might be compiling more code than the Arduino IDE currently. Arduino libraries either have their source files contained in the `src` folder (which is compiled recursively) for modern libraries, or inside the root and `utility` subfolder (which are compiled non-recursively) for legacy libraries. So I think this is something ArduinoCI could (and should) just match, and which would solve the problem with a `test` subfolder AFAIU?

See also [the Arduino library spec](https://arduino.github.io/arduino-cli/latest/library-specification) and [this issue I filed about unclarity about legacy vs new-style libraries](https://github.com/arduino/arduino-cli/issues/1013).

rosogon, 2020-10-06:
> If you were looking for a quick fix, I'm sorry to be taking you down this rabbit hole.

No worries. I have the quick fix running locally :)

I agree that the best approach is what you propose, specially if the idea is to store more information. It was actually my first thought, but discarded because of not being backwards compatible. 

Anyway, as @matthijskooijman's comments, there are library files being included to compile that shouldn't.

ianfixes, 2020-10-07:
Thanks for filing that @matthijskooijman, I'm now subscribed. I'll implement the logic you suggest here in advance of getting an answer there.

ianfixes, 2020-10-07:
Looks like I have the logic partially implemented already, just need to make it conditional: https://github.com/Arduino-CI/arduino_ci/blob/744365d/lib/arduino_ci/cpp_library.rb#L238-L253

per1234, 2020-10-07:
You can see Arduino CLI's logic here:
https://github.com/arduino/arduino-cli/blob/master/arduino/libraries/loader.go
This also applies to Arduino IDE

So there are two library layouts:
- recursive (called the "1.5 format" in the specification)
- flat (called the "1.0 format" in the specification)

This is how layout is determined:
```
+--------------------+    +--------+
|                    | no |        |
| library.properties +--->+ legacy |
|                    |    |        |
+----------+---------+    +---+----+
           |                  |
           |yes               |
           v                  |
   +-------+------+           |
   |              |    no     |
   | "src" folder +---------->+
   |              |           |
   +-------+------+           |
           |                  |
           |yes               |
           v                  v
     +-----+-----+        +---+--+
     |           |        |      |
     | recursive |        | flat |
     |           |        |      |
     +-----------+        +------+
```
layout | root compiled | `src` compiled | `utility` compiled
-|-|-|-
recursive | no |recursively | no
flat | yes | no | yes

matthijskooijman, 2020-10-07:
Thanks for the detailed answer, @per1234!

rosogon, 2020-10-07:
> Looks like I have the logic partially implemented already, just need to make it conditional: https://github.com/Arduino-CI/arduino_ci/blob/744365d/lib/arduino_ci/cpp_library.rb#L238-L253

Take into account  that Find.find at [cpp_files_in](https://github.com/Arduino-CI/arduino_ci/blob/744365d/lib/arduino_ci/cpp_library.rb#L161), recursively enters all directories. As one of the checked dirs is "", whole tree is actually returned. I guess cpp_files_in must receive a `recursive` parameter, so you can be recursive on `/src` but not on `/` (see Find.prune).

ianfixes, 2020-10-16:
I'm taking a look at this now

ianfixes, 2020-10-19:
@rosogon can you try my branch from #183 and see if this fixes the issue?  You will need to update the dependency in your `Gemfile` as follows:

```ruby
gem 'arduino_ci', git: 'https://github.com/ianfixes/arduino_ci.git', branch: '2020-10-16_suggestions'
```

rosogon, 2020-10-21:
Hi @ianfixes . I get this error when running `bundle exec arduino_ci_remote.rb --skip-compilation`:

```
Requested unittest platform 'uno' is defined in 'platforms' YML...             ✓
Using pre-existing library...                                               libA
Unit testing test.cpp with g++... 
/home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bundler/gems/arduino_ci-6196faa19e2e/lib/arduino_ci/cpp_library.rb:291:in `block in arduino_library_src_dirs': undefined method `new' for #<ArduinoCI::CppLibrary:0x00000002ee0988> (NoMethodError)
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bundler/gems/arduino_ci-6196faa19e2e/lib/arduino_ci/cpp_library.rb:291:in `map'
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bundler/gems/arduino_ci-6196faa19e2e/lib/arduino_ci/cpp_library.rb:291:in `arduino_library_src_dirs'
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bundler/gems/arduino_ci-6196faa19e2e/lib/arduino_ci/cpp_library.rb:298:in `include_args'
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bundler/gems/arduino_ci-6196faa19e2e/lib/arduino_ci/cpp_library.rb:345:in `test_args'
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bundler/gems/arduino_ci-6196faa19e2e/lib/arduino_ci/cpp_library.rb:375:in `build_for_test_with_configuration'
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bundler/gems/arduino_ci-6196faa19e2e/exe/arduino_ci_remote.rb:236:in `block (4 levels) in perform_unit_tests'
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bundler/gems/arduino_ci-6196faa19e2e/exe/arduino_ci_remote.rb:101:in `perform_action'
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bundler/gems/arduino_ci-6196faa19e2e/exe/arduino_ci_remote.rb:122:in `attempt_multiline'
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bundler/gems/arduino_ci-6196faa19e2e/exe/arduino_ci_remote.rb:235:in `block (3 levels) in perform_unit_tests'
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bundler/gems/arduino_ci-6196faa19e2e/exe/arduino_ci_remote.rb:234:in `each'
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bundler/gems/arduino_ci-6196faa19e2e/exe/arduino_ci_remote.rb:234:in `block (2 levels) in perform_unit_tests'
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bundler/gems/arduino_ci-6196faa19e2e/exe/arduino_ci_remote.rb:232:in `each'
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bundler/gems/arduino_ci-6196faa19e2e/exe/arduino_ci_remote.rb:232:in `block in perform_unit_tests'
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bundler/gems/arduino_ci-6196faa19e2e/exe/arduino_ci_remote.rb:231:in `each'
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bundler/gems/arduino_ci-6196faa19e2e/exe/arduino_ci_remote.rb:231:in `perform_unit_tests'
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bundler/gems/arduino_ci-6196faa19e2e/exe/arduino_ci_remote.rb:396:in `<top (required)>'
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bin/arduino_ci_remote.rb:23:in `load'
        from /home/roman/romanfiles/projects/Arduino/arduino_ci_two_libraries/.bundle/ruby/2.3.0/bin/arduino_ci_remote.rb:23:in `<main>'
```

ianfixes, 2020-10-26:
You're right.. and I meant to respond a few days ago that (1) the test code didn't exercise this the way I thought it would and (2) there's a small rabbit hole here with getting library specs to load recursively.  And possibly an inconsistency (library name vs path on disk) in the current code.

But this is in progress.

ianfixes, 2020-10-27:
OK, I think this is ready for you to give it another try.  At any rate, I was able to confirm what I _believe_ is the desired behavior via unit testing.

This fixes a host of issues you didn't raise, including being able to recursively bring in library dependencies by looking at `library.properties`.  I expect this will work fine for local testing, but I haven't bubbled up the "give me all the libraries" functionality to the CI test scripts.  That can be future work if this satisfies your local testing.  Please let me know!

rosogon, 2020-11-02:
Hi @ianfixes . Just tested and it works for my cases (both with 1.0 libraries). Thanks!

PR #183 (2020-11-13): Suggested fixes
### Trying it out
You can test this beta of the gem directly from github by putting this in your `Gemfile`:

```ruby
gem 'arduino_ci', git: 'https://github.com/ianfixes/arduino_ci.git', branch: '2020-10-16_suggestions'
```

## Highlights from `CHANGELOG.md`
### Added
- `arduino_ci_remote.rb` CLI switch `--skip-examples-compilation`
- `CppLibrary.header_files` to find header files
- `LibraryProperties` to read metadata from Arduino libraries
- `CppLibrary.library_properties_path`, `CppLibrary.library_properties?`, `CppLibrary.library_properties` to expose library properties of a [... cut]
closed 2020-11-13 by ianfixes