Arduino-CI/arduino_ci #176: ARDUINO=100 always added as define to g++

jgfoster, 2020-10-11:
## Attempt to override ARDUINO macro gives duplicate

- OS: macOS
- `ruby 2.6.3p62 (2019-04-16 revision 67580) [universal.x86_64-darwin19]`
- `Bundler version 1.17.2`
- `arduino_ci (0.3.0)`
- `Apple clang version 12.0.0 (clang-1200.0.32.2)`
- Arduino IDE version: 1.8.13
- `java version "11.0.7" 2020-04-14 LTS`

## `.arduino-ci.yaml`:
```
platforms:
  mega2560:
    board: arduino:avr:mega:cpu=atmega2560
    package: arduino:avr
    gcc:
      features:
      defines:
        - __AVR_ATmega2560__
        # added this line
        - ARDUINO_CI
        - ARDUINO=10813
      warnings:
      flags:

unittest:
  platforms:
      - mega2560

compile:
  platforms:
    - mega2560
```

## Warning
```
In file included from <built-in>:399:
<command line>:4:9: warning: 'ARDUINO' macro redefined [-Wmacro-redefined]
#define ARDUINO 10813
        ^
<command line>:1:9: note: previous definition is here
#define ARDUINO 100
        ^
1 warning generated.
```

## Arduino Architecture(s) Affected: Mega2560

ianfixes, 2020-10-16:
I'm not clear on why `ARDUINO` would need to be defined in `mega2650`, which is a statement of genuine ignorance; I set it to `100` based on [this forum.arduion.cc post](https://forum.arduino.cc/index.php?topic=263843.0), which says:

> > Why is `(ARDUINO) && ARDUINO >= 100` needed?
> 
> It's needed because early versions (pre 1.0) had a Serial method "print" which allowed binary when used with the qualifier BYTE, but this has now been superseded by the "write" method.

In other words, I was under the impression that this was used by the IDE and not by a particular board.  Can someone enlighten me on the proper use of this `#define`?

jgfoster, 2020-10-16:
I'll look for the specifics, but there was some change to Arduino core library at 1.0.6 and an external library had `#if ARDUINO < 106` to take different code paths. When it runs in the Arduino IDE it uses 10813 and uses the correct code. When it runs in the mocks library is uses the wrong code and fails.

jgfoster, 2020-10-16:
[This](https://github.com/arduino-libraries/Ethernet/blob/75a3c37b5e513305b82e926ca6a4f8190f536c9d/src/Ethernet.cpp) is where I ran into the problem (line 90):

```
#elif ARDUINO > 106 || TEENSYDUINO > 121
```

ianfixes, 2020-10-19:
> [`ARDUINO=100` is] not used by the IDE, it's defined by the IDE

Sorry for my poor wording there, I meant to say that the definition is _associated_ with the IDE as a whole and not individual boards.  i.e. the IDE "uses" the `ARDUINO` variable (setting it to 100) as a signal to the code it's compiling, and the boards do not do this.

Based on what I'm hearing here, it sounds like it's perfectly appropriate for me to be defining `ARDUINO=100` in `CppLIbrary`, since I am indeed attempting to "Indicate that the code is being compiled by the Arduino build system".

Worrying about the `ARDUINO > 106` case is going to be its own can of worms, and we should probably open a separate issue to track what that feature might involve.  

(note to future self: on one hand, it could involve iterating the entire CI process over a set of possible values like `100`, `106`, etc.  On the other hand, the companion IDE performing the CI tests will be only one version, and we could talk about whether proper Arduino library development means keeping current with one's IDE version.)

I've been hesitant to merge #179 because it breaks my fundamental assumption that there is only one Arduino IDE version for me to install and imitate (thus, whether the right value is `100` or `106` I can hard-code it because I'm hard-coding the IDE.)  Am I thinking about this the wrong way?

jgfoster, 2020-10-19:
Actually, I think you are right to be concerned. I brought in the current version of Ethernet and found that it didn't work. My work-around might be wrong. It would probably have been better if I brought in an old version of Ethernet to match ARDUINO=100. But, my preference would be for Arduino CI to be more current (1.6 if not 1.8.13). If you were going to move forward, then I'd be helped; if you don't, then I should go look for an old version of Ethernet.

My preference for a current version of ARDUINO is that I'm writing production code and will be deploying on 1.8.13 (or later).

ianfixes, 2020-10-19:
I think the right course of action for me is to resolve #185 before anything else

jgfoster, 2020-10-27:
I agree that #185 is better.

closed 2020-10-27 by jgfoster