Arduino-CI/arduino_ci #181: Implement Wire

jgfoster, 2020-10-14:
`cpp/arduino/Wire.h` has a number of `// TODO` comments. I'd like to complete the implementation and would appreciate guidance (or at least feedback on some ideas).

One extreme would be to pull in an existing implementation with the assumption that a "real" implementation would work by reading and writing pins (pins 20 & 21 on the Duo and Mega2560). With that, it is just like any other code that reads and writes pins.

A middle ground would be to write a new implementation that reads and write pins, but that seems like duplicate work.

The other extreme would be to mock the API at a higher level (and not set pins). Tests would not look at pins but would need to call a new testing API that could get and set values at a higher level than bits on a pin. This could be something added to Godmode or to Wire itself. 

I propose a new high-level API (I would want it anyway) and that it be in Wire (I'd prefer not to further pollute Godmode). Thoughts?

ianfixes, 2020-10-16:
This refers to https://www.arduino.cc/en/Reference/Wire

The TODO items correspond to where I'd put some sort of GODMODE hook, but I wasn't sure of how much to leverage existing structures for this or write new ones -- does it override what goes on with PinHistory, or interact with #145, etc.

jgfoster, 2020-10-16:
I expect there will have a PR on this in a couple days. Then we will have something concrete to discuss and critique.

PR #189 (2020-11-19): High-level wire implementation, complete with unit tests
Fixes #181 

Tracks addresses read from/written to on a high level using maps and vectors, as well as the content at that address, of course.

Let me know if you'd like more tests, though I think I covered everything that we deemed useful at the moment.
closed 2020-11-19 by ianfixes