Arduino-CI/arduino_ci #187: Fix _SFR_IO8 macro to avoid "cannot take the address of an rvalue"

ianfixes, 2020-10-26:
## Issue Summary 

see https://github.com/Arduino-CI/SD/issues/2

```
error: cannot take the address of an rvalue of type 'int'
  {&DDRE, &PINE, &PORTE, 0},  // E0  0
   ^~~~~
```

[Working theory](https://github.com/Arduino-CI/SD/issues/2#issuecomment-716266934) is that this is caused by a badly formed macro. 

https://github.com/Arduino-CI/arduino_ci/blob/master/cpp/arduino/avr/io.h#L99

```diff
- #define _SFR_IO8(io_addr) (io_addr) // this macro is all we need from the sfr file 
+ #define _MMIO_BYTE(mem_addr) (*(volatile uint8_t *)(mem_addr))
+ #define _SFR_IO8(io_addr) _MMIO_BYTE((io_addr) + __SFR_OFFSET)
```

Will see if `volatile` is sufficient to fix

jgfoster, 2020-10-26:
Also see #120. I copied code from there and got things to compile.

ianfixes, 2020-10-27:
I was able to reproduce this with a unit test, and added the `volatile` keyword (and cast) to fix the error.  Currently in my feature branch.

ianfixes, 2020-11-04:
This took a lot longer than expected due to chasing my tail on linker errors (and segfaults).  I think I am out of the woods now though.

PR #183 (2020-11-13): Suggested fixes
### Trying it out
You can test this beta of the gem directly from github by putting this in your `Gemfile`:

```ruby
gem 'arduino_ci', git: 'https://github.com/ianfixes/arduino_ci.git', branch: '2020-10-16_suggestions'
```

## Highlights from `CHANGELOG.md`
### Added
- `arduino_ci_remote.rb` CLI switch `--skip-examples-compilation`
- `CppLibrary.header_files` to find header files
- `LibraryProperties` to read metadata from Arduino libraries
- `CppLibrary.library_properties_path`, `CppLibrary.library_properties?`, `CppLibrary.library_properties` to expose library properties of a [... cut]
closed 2020-11-13 by ianfixes