Arduino-CI/arduino_ci #188: Do not compile tests from included libraries

jgfoster, 2020-10-26:
I have added the Arduino CI enabled version of LiquidCrystal to another library and it finds it and properly references the headers. My problem is that it also tries to compile the test directories and I only want the `src/` files.

A portion of `.arduino-ci.yml` follows:

```
unittest:
    exclude_dirs:
        - "/Users/jfoster/Documents/Arduino/libraries/LiquidCrystal/test/"
    platforms:
        - mega2560
    libraries:
      - "LiquidCrystal"
```

Following is the error. From the "Last command" you can see that it is including .cpp files from `LiquidCrystal/test/`, and I don't want those. I believe it is just including everything under `libraries/LiquidCrystal`. I've tried to exclude the directory (see above). Any advice?

```
Located Arduino binary...                              /Applications/Arduino.app
The set of compilers (1) isn't empty...                                        ✓
Checking g++ version... 
    Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
    Apple clang version 12.0.0 (clang-1200.0.32.21)
    Target: x86_64-apple-darwin19.6.0
    Thread model: posix
    InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
...Checking g++ version                                                        ✓
libasan availability for g++...                                             true
Requested unittest platform 'mega2560' is defined in 'platforms' YML...        ✓
Using pre-existing library...                                      LiquidCrystal
Unit testing Blink.cpp with g++... 

Last command:  $ g++ -std=c++0x -o /Users/jfoster/Documents/Arduino/libraries/TankControllerLib/unittest_Blink.cpp.bin -DARDUINO=100 -g -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address -D__AVR_ATmega2560__ -DARDUINO_CI -I/Users/jfoster/Documents/Arduino/libraries/TankControllerLib/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/arduino -I/Users/jfoster/Documents/Arduino/libraries/TankControllerLib/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/unittest -I/Users/jfoster/Documents/Arduino/libraries/TankControllerLib/src/Devices -I/Users/jfoster/Documents/Arduino/libraries/TankControllerLib/src -I/Users/jfoster/Documents/Arduino/libraries/TankControllerLib/src/UIState -I/Users/jfoster/Documents/Arduino/libraries/LiquidCrystal -I/Users/jfoster/Documents/Arduino/libraries/LiquidCrystal/src /Users/jfoster/Documents/Arduino/libraries/TankControllerLib/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/arduino/Arduino.cpp /Users/jfoster/Documents/Arduino/libraries/TankControllerLib/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/arduino/Godmode.cpp /Users/jfoster/Documents/Arduino/libraries/TankControllerLib/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/arduino/stdlib.cpp /Users/jfoster/Documents/Arduino/libraries/TankControllerLib/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/unittest/ArduinoUnitTests.cpp /Users/jfoster/Documents/Arduino/libraries/TankControllerLib/src/Devices/LiquidCrystal_TC.cpp /Users/jfoster/Documents/Arduino/libraries/TankControllerLib/src/TankControllerLib.cpp /Users/jfoster/Documents/Arduino/libraries/TankControllerLib/src/UIState/MainMenu.cpp /Users/jfoster/Documents/Arduino/libraries/LiquidCrystal/src/LiquidCrystal.cpp /Users/jfoster/Documents/Arduino/libraries/LiquidCrystal/src/LiquidCrystal_CI.cpp /Users/jfoster/Documents/Arduino/libraries/LiquidCrystal/test/BaseClass.cpp /Users/jfoster/Documents/Arduino/libraries/LiquidCrystal/test/Common.cpp /Users/jfoster/Documents/Arduino/libraries/LiquidCrystal/test/TestClass.cpp /Users/jfoster/Documents/Arduino/libraries/LiquidCrystal/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/arduino/Arduino.cpp /Users/jfoster/Documents/Arduino/libraries/LiquidCrystal/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/arduino/Godmode.cpp /Users/jfoster/Documents/Arduino/libraries/LiquidCrystal/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/arduino/stdlib.cpp /Users/jfoster/Documents/Arduino/libraries/LiquidCrystal/vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0/cpp/unittest/ArduinoUnitTests.cpp /Users/jfoster/Documents/Arduino/libraries/TankControllerLib/test/Blink.cpp

/Users/jfoster/Documents/Arduino/libraries/LiquidCrystal/test/Common.cpp:91:3: error: unknown type name 'LiquidCrystal_Test'; did you mean 'LiquidCrystal_Base'?
  LiquidCrystal_Test lcd1(rs, enable, d4, d5, d6, d7);
  ^~~~~~~~~~~~~~~~~~
  LiquidCrystal_Base
/Users/jfoster/Documents/Arduino/libraries/LiquidCrystal/src/LiquidCrystal.h:54:7: note: 'LiquidCrystal_Base' declared here
class LiquidCrystal_Base : public Print {
      ^
/Users/jfoster/Documents/Arduino/libraries/LiquidCrystal/test/Common.cpp:92:3: error: unknown type name 'LiquidCrystal_Test'; did you mean 'LiquidCrystal_Base'?
  LiquidCrystal_Test lcd2(rs, rw, enable, d4, d5, d6, d7);
  ^~~~~~~~~~~~~~~~~~
  LiquidCrystal_Base
/Users/jfoster/Documents/Arduino/libraries/LiquidCrystal/src/LiquidCrystal.h:54:7: note: 'LiquidCrystal_Base' declared here
class LiquidCrystal_Base : public Print {
      ^
```

ianfixes, 2020-10-27:
This may be a duplicate of #168, can you see if #183 fixes it?

jgfoster, 2020-10-27:
What is the best way to test #183? There are 57 files changed, so this doesn't seem like something to test by hand. Should I add `ianfixes` as a remote, then checkout that branch and copy it into `vendor/bundle/ruby/2.6.0/gems/arduino_ci-0.3.0`? It would be one less step if it were pushed from `ianfixes` to `arduino_ci` as a branch.

jgfoster, 2020-10-27:
I think I'm using the new code and the problem still exists. Specifically, I'm including `LiquidCrystal` and the error comes when compiling [`test/Common.cpp`](https://github.com/Arduino-CI/LiquidCrystal/blob/master/test/Common.cpp). This file is listed in [`arduino_ci.yml`](https://github.com/Arduino-CI/LiquidCrystal/blob/master/.arduino-ci.yml) as something to be rejected when testing (it relies on macros that are defined in other files). So, either you need to parse `.arduino-ci.yml` in referenced libraries (seems complicated) or you should exclude test directories completely when importing external libraries (my preference).

jgfoster, 2020-10-27:
Yes, #183 fixes it. Thanks!

closed 2020-10-27 by jgfoster

ianfixes, 2020-10-27:
You can get the gem directly from github by putting this in your `Gemfile`:

```ruby
gem 'arduino_ci', git: 'https://github.com/ianfixes/arduino_ci.git', branch: '2020-10-16_suggestions'
```