Arduino-CI/arduino_ci #196: `_SFR_IO8` no longer visible after definition moved to a new file

jgfoster, 2020-11-05:
As part of `8180efbf09a8337dcb89b6a7c8dccf97f26bb627` the definition of `_SFR_IO8` was moved from `cpp/arduino/arv/io.h` to `cpp/arduino/Arduino.h`. Unfortunately, this macro is referenced from `cpp/arduino/SPI.h` and now is not found. I'm happy to submit a PR with a fix, but I'm not sure which of the preceding files should have `#include <Arduino.h>`.

```text
Run bundle exec arduino_ci_remote.rb
Located Arduino binary...                    /home/runner/arduino_ci_ide/arduino
The set of compilers (1) isn't empty...                                        ✓
Checking g++ version... 
    Using built-in specs.
    COLLECT_GCC=g++
    COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
    OFFLOAD_TARGET_NAMES=nvptx-none
    OFFLOAD_TARGET_DEFAULT=1
    Target: x86_64-linux-gnu
    Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.5.0-3ubuntu1~18.04' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
    Thread model: posix
    gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04) 
...Checking g++ version                                                        ✓
libasan availability for g++...                                             true
Requested unittest platform 'mega2560' is defined in 'platforms' YML...        ✓
Library conforms to Arduino library specification...                         1.5
Using pre-existing library...                                      LiquidCrystal
Installing aux library 'RTClib'...                                             ✓
Unit testing Blink.cpp with g++... 

Last command:  $ g++ -std=c++0x -o /home/runner/work/TankControllerLib/TankControllerLib/unittest_Blink.cpp.bin -DARDUINO=100 -D__AVR__ -g -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address -D__AVR_ATmega2560__ -DARDUINO_CI -I/home/runner/.rubies/ruby-2.6.6/lib/ruby/gems/2.6.0/bundler/gems/arduino_ci-8180efbf09a8/cpp/arduino -I/home/runner/.rubies/ruby-2.6.6/lib/ruby/gems/2.6.0/bundler/gems/arduino_ci-8180efbf09a8/cpp/unittest -I/home/runner/work/TankControllerLib/TankControllerLib/src -I/home/runner/work/TankControllerLib/TankControllerLib/src/Devices -I/home/runner/work/TankControllerLib/TankControllerLib/src/UIState -I/home/runner/Arduino/libraries/LiquidCrystal/src -I/home/runner/Arduino/libraries/RTClib /home/runner/.rubies/ruby-2.6.6/lib/ruby/gems/2.6.0/bundler/gems/arduino_ci-8180efbf09a8/cpp/arduino/Arduino.cpp /home/runner/.rubies/ruby-2.6.6/lib/ruby/gems/2.6.0/bundler/gems/arduino_ci-8180efbf09a8/cpp/arduino/Godmode.cpp /home/runner/.rubies/ruby-2.6.6/lib/ruby/gems/2.6.0/bundler/gems/arduino_ci-8180efbf09a8/cpp/arduino/stdlib.cpp /home/runner/.rubies/ruby-2.6.6/lib/ruby/gems/2.6.0/bundler/gems/arduino_ci-8180efbf09a8/cpp/unittest/ArduinoUnitTests.cpp /home/runner/work/TankControllerLib/TankControllerLib/src/TankControllerLib.cpp /home/runner/work/TankControllerLib/TankControllerLib/src/Devices/LiquidCrystal_TC.cpp /home/runner/work/TankControllerLib/TankControllerLib/src/UIState/MainMenu.cpp /home/runner/Arduino/libraries/LiquidCrystal/src/LiquidCrystal.cpp /home/runner/Arduino/libraries/LiquidCrystal/src/LiquidCrystal_CI.cpp /home/runner/Arduino/libraries/RTClib/RTClib.cpp /home/runner/work/TankControllerLib/TankControllerLib/test/Blink.cpp

In file included from /home/runner/.rubies/ruby-2.6.6/lib/ruby/gems/2.6.0/bundler/gems/arduino_ci-8180efbf09a8/cpp/arduino/avr/iom2560.h:38:0,
                 from /home/runner/.rubies/ruby-2.6.6/lib/ruby/gems/2.6.0/bundler/gems/arduino_ci-8180efbf09a8/cpp/arduino/avr/io.h:172,
                 from /home/runner/.rubies/ruby-2.6.6/lib/ruby/gems/2.6.0/bundler/gems/arduino_ci-8180efbf09a8/cpp/arduino/avr/pgmspace.h:17,
                 from /home/runner/.rubies/ruby-2.6.6/lib/ruby/gems/2.6.0/bundler/gems/arduino_ci-8180efbf09a8/cpp/arduino/ArduinoDefines.h:3,
                 from /home/runner/.rubies/ruby-2.6.6/lib/ruby/gems/2.6.0/bundler/gems/arduino_ci-8180efbf09a8/cpp/arduino/Godmode.h:2,
                 from /home/runner/.rubies/ruby-2.6.6/lib/ruby/gems/2.6.0/bundler/gems/arduino_ci-8180efbf09a8/cpp/arduino/Godmode.cpp:1:
/home/runner/.rubies/ruby-2.6.6/lib/ruby/gems/2.6.0/bundler/gems/arduino_ci-8180efbf09a8/cpp/arduino/SPI.h: In member function ‘uint16_t SPIClass::transfer16(uint16_t)’:
/home/runner/.rubies/ruby-2.6.6/lib/ruby/gems/2.6.0/bundler/gems/arduino_ci-8180efbf09a8/cpp/arduino/avr/iomxx0_1.h:387:17: error: ‘_SFR_IO8’ was not declared in this scope
 #define SPCR    _SFR_IO8(0x2C)
                 ^
/home/runner/.rubies/ruby-2.6.6/lib/ruby/gems/2.6.0/bundler/gems/arduino_ci-8180efbf09a8/cpp/arduino/SPI.h:97:11: note: in expansion of macro ‘SPCR’
     if (!(SPCR & (1 << DORD))) {
           ^~~~
...Unit testing Blink.cpp with g++                                             ✗
```

matthijskooijman, 2020-11-06:
Sounds like this macro should just be defined in `io.h`, since that's (effectively) where it's defined in the original avr libc as well, and then SPI.h should just include that? But maybe there's another reason that Arduino.h needs it (haven't checked), but it could also just include `io.h`?

ianfixes, 2020-11-07:
It sounds like I don't have enough unit tests defined to capture the compilation needs here.  So this is sort of a PR comment on #183?

jgfoster, 2020-11-07:
Yes, this probably should have been a PR comment on #183. If you revert the recent edit then things will work again for us.

PR #183 (2020-11-13): Suggested fixes
### Trying it out
You can test this beta of the gem directly from github by putting this in your `Gemfile`:

```ruby
gem 'arduino_ci', git: 'https://github.com/ianfixes/arduino_ci.git', branch: '2020-10-16_suggestions'
```

## Highlights from `CHANGELOG.md`
### Added
- `arduino_ci_remote.rb` CLI switch `--skip-examples-compilation`
- `CppLibrary.header_files` to find header files
- `LibraryProperties` to read metadata from Arduino libraries
- `CppLibrary.library_properties_path`, `CppLibrary.library_properties?`, `CppLibrary.library_properties` to expose library properties of a [... cut]
closed 2020-11-13 by ianfixes