Arduino-CI/arduino_ci #199: Add GitHub Actions

jgfoster, 2020-11-09:
`arduino_ci` has two automated testing systems set up. We should add GitHub Actions. (I can work on this but just wanted to note it as an issue.)

ianfixes, 2020-11-10:
I'm open to this, but I'm not clear on the problem being solved here.  We currently test 3 platforms on every PR or code push: windows, linux, and OSX.  What do GitHub actions add to this?

jgfoster, 2020-11-10:
This isn't adding a new operating system, this is adding a new testing infrastructure. There are two values: 

First, since `arduino_ci` is a testing framework, it should broadly demonstrate and use the various testing platforms out there. (Actually, I didn't realize that the existing two-part actions were testing three operating systems; I thought that they were demonstrating the two primary testing hosts.)

Second, the two existing testing actions require external setup. While you have gone to the trouble to set up the external systems for this GitHub repository, the rest of us (so far) haven't bothered to do so. The virtue of GitHub Actions is that they are built in to GitHub. By forking your repository I will get the GitHub Action configuration (if/when this PR is accepted). Then my fork will automatically run tests on every push _to my repository._ With that, even if I fail to run the test suite before doing a local commit (hand slap!), one will be run when I do a push. This is pushing best practices to the edge.

ianfixes, 2020-11-10:
Understood, I think I made the more relevant comment on #200

ianfixes, 2020-11-10:
Thought better of it, created #203

ianfixes, 2020-12-09:
I think this is complete, but we can reopen if I'm misreading it

closed 2020-12-09 by ianfixes