Arduino-CI/arduino_ci #203: Add demonstration of using GitHub Actions to execute Arduino CI

ianfixes, 2020-11-10:
Set up GitHub Actions for this repository to demonstrate how a user of `arduino_ci` would set it up for themselves

Related to #199 but this concerns how an end user (not a contributor to `arduino_ci` itself) would leverage this tool

jgfoster, 2020-11-10:
Should we choose a project like [Blink](https://github.com/Arduino-CI/Blink) and make it the "poster child" for how the author of an external library would leverage `arduino_ci`? If so, then this issue would be placed there and the local `README.md` would point to that project? (And, perhaps, now that I know a bit more I should re-record the video describing it!?)

ianfixes, 2020-11-10:
Yes... this might be a sub-task of #164 ... ditto for #77

prestoncarman, 2020-11-11:
Following the example in #77 and the README example configs, I set up automated testing for these vendors in the [Blink](https://github.com/Arduino-CI/Blink) repository. The change is in [PR](https://github.com/Arduino-CI/Blink/pull/1). The update includes a table on the README showing all the Vendors and Operating systems that are used in the sample project.

[image]

Notes: macOS test is dependent on #199.

jgfoster, 2020-11-12:
You can get macOS to pass now by explicitly downloading a more recent version of `Arduino.app` in your Action script. See what was [added to Blink](https://github.com/Arduino-CI/Blink/commit/273c26741343f6563030e68f0aee053647841507).

ianfixes, 2020-11-13:
I really like this, I might not get to it today but I will soon

ianfixes, 2020-12-09:
I think this is completed, _and_ we have our own action now https://github.com/marketplace/actions/arduino_ci

closed 2020-12-09 by ianfixes