Arduino-CI/arduino_ci #219: Allow specification of `std=c++17` for tests

jgfoster, 2020-11-17:
I see that `cpp_library.rb:401` specifies `-std=c++0x` as the version of C++. I'd like to use some features that are in `std=c++17` (I want the [filesystem](https://en.cppreference.com/w/cpp/filesystem) features). Would it be reasonable to bump up to version 17 or make it optional?

ianfixes, 2020-11-19:
I'm actually falling somewhere in between, see #146 

Summary: I think this project should be using the same hard-coded c++ standard as the official Arduino compiler(s).  It's not doing that yet.

So to be blunt, in my opinion this is either a duplicate of #146 or a won't-fix.

jgfoster, 2020-11-19:
We found it pretty difficult to mock [SD](https://github.com/Arduino-CI/SD/) in a cross-platform way without having [filesystem](https://en.cppreference.com/w/cpp/filesystem). So, I'll make a gentle appeal for at least an option to use a later library at your own risk. But I'm sure we can find a work-around if needed...

jgfoster, 2020-11-29:
I've had further thoughts on this and will present them in a new issue.

closed 2020-11-29 by jgfoster