Arduino-CI/arduino_ci #227: Compile `arduino_ci` once as a linkable library and use for each test

jgfoster, 2020-11-29:
Each test is compiled with the mocks as a single compile/link operation. When there are many tests, this can take a bit of time. One possible optimization would be to compile `arduino_ci` once as a linkable library at the beginning and then link it with each subsequent test (which would compile more quickly.

In addition to the performance improvement, this would allow us to separate the process of building the mocks from the process of building the tests. In particular, this would allow us to use some C++ libraries and features when building the mocks that are not available when building the tests. That is, we want the tests to be built using an environment as close as possible to the Arduino CLI/IDE, but there isn't any reason (other than convenience) why we need to apply the same limits to the mocks. (This is a follow-up to #219.) (Linking different libraries from the same compiler [should be safe](https://stackoverflow.com/questions/46746878/is-it-safe-to-link-c17-c14-and-c11-objects).)

ianfixes, 2020-11-29:
I like this idea and I'm absolutely open to this approach.  There is a lot of c++ work that I'd like to accomplish, hopefully with the support of the `arduino-cli` backend (to provide compiler flags for platforms, and possibly even #147), and better exploitation of already-compiled assets would be a great addition to that.

jgfoster, 2020-11-29:
I've been investigating this further, especially as it relates to the second motivation (to allow the mocks to be compiled with C++17). Unfortunately, this isn't quite as simple as I first thought. When making mocks for other libraries, the distinction between base files and mock files is not clear except by naming convention or reading the code. For this to work we do need some way to distinguish mock files from non-mock files. Ideas include:

- Naming convention (the name includes `ci`)
- File contents (a special comment or `#define`)
- Special handling in `arduino_ci.yml

I'm wondering if the second motivation should be split into another issue that would follow the initial work on core `arduino_ci` mocks.

ianfixes, 2021-01-12:
Should I try to make a two-stage build of some kind where I compile the unit test libraries to object files as a first pass, and then compile from the objects all subsequent times?

jgfoster, 2021-01-12:
Yes, it seems like we could (1) compile `arduino_ci` as a linkable library, and then (2) link it with each subsequent test (which would compile more quickly). I don't think this would need to be done as a separate build stage; it should be enough just to have it as a separate `g++` line in the existing script.

closed 2021-10-17 by jgfoster