Arduino-CI/arduino_ci #234: Syntax error flagged in 'arduino_backend.rb'

ethomas997, 2020-12-03:
Seeing this error in GitHub CI:
```
$ bundle exec arduino_ci.rb
Traceback (most recent call last):
	9: from /home/travis/.rvm/gems/ruby-2.5.3/bin/ruby_executable_hooks:24:in `<main>'
	8: from /home/travis/.rvm/gems/ruby-2.5.3/bin/ruby_executable_hooks:24:in `eval'
	7: from /home/travis/.rvm/gems/ruby-2.5.3/bin/arduino_ci.rb:23:in `<main>'
	6: from /home/travis/.rvm/gems/ruby-2.5.3/bin/arduino_ci.rb:23:in `load'
	5: from /home/travis/.rvm/gems/ruby-2.5.3/gems/arduino_ci-1.1.0/exe/arduino_ci.rb:2:in `<top (required)>'
	4: from /home/travis/.rvm/gems/ruby-2.5.3/gems/arduino_ci-1.1.0/exe/arduino_ci.rb:2:in `require'
	3: from /home/travis/.rvm/gems/ruby-2.5.3/gems/arduino_ci-1.1.0/lib/arduino_ci.rb:2:in `<top (required)>'
	2: from /home/travis/.rvm/gems/ruby-2.5.3/gems/arduino_ci-1.1.0/lib/arduino_ci.rb:2:in `require'
	1: from /home/travis/.rvm/gems/ruby-2.5.3/gems/arduino_ci-1.1.0/lib/arduino_ci/arduino_installation.rb:3:in `<top (required)>'
/home/travis/.rvm/gems/ruby-2.5.3/gems/arduino_ci-1.1.0/lib/arduino_ci/arduino_installation.rb:3:in `require': /home/travis/.rvm/gems/ruby-2.5.3/gems/arduino_ci-1.1.0/lib/arduino_ci/arduino_backend.rb:55: syntax error, unexpected ']' (SyntaxError)
...tual_args = has_env ? args[1..] : args  # need to shift over...
```
Example build here:
https://travis-ci.com/github/RotorHazard/RotorHazard/jobs/453265718

ianfixes, 2020-12-03:
Hi, thanks for letting me know!  I think I have only my documentation to blame.

The issue here is a combination of 2 things.  The most pressing one is in the `Gemfile`: 
https://github.com/RotorHazard/RotorHazard/blob/2dec371a39f510b9d4a8192d9b0510dae3541ff4/src/node/Gemfile#L2

```ruby
gem 'arduino_ci'
```

This doesn't specify a specific version, so it's picking up the recently-released latest version -- a major release with breaking changes (it runs on ruby 2.6).  

Travis is using a python image, and from your output above it looks like that image includes ruby 2.5.
https://github.com/RotorHazard/RotorHazard/blob/2dec371a39f510b9d4a8192d9b0510dae3541ff4/.travis.yml#L3-L5

It looks like the fix here is for me to just stick to 2.5 syntax for now, I can roll out a new release and I will update the docs.

ianfixes, 2020-12-03:
I recently set up a GitHub action to make Arduino CI easier to use:
https://github.com/Arduino-CI/action

...but it won't work for you yet, as I see that your Arduino library is in a subdirectory of your repo. Would you be up for experimenting with a branch of that if I add the ability to run the tests in a subdir of a repository?

ianfixes, 2020-12-03:
Sorry... the fastest way to fix the problem you have right now is to update the line in the `Gemfile` to pick the last [release](https://github.com/Arduino-CI/arduino_ci/releases) that worked for you:

```ruby
gem 'arduino_ci', '~> 0.4'
```

ethomas997, 2020-12-03:
Thanks for the workaround.  Regarding experimenting with the Arduino CI action, I don't have a huge amount of time I can put into it, but if there's something I can do to help out let me know.

ianfixes, 2020-12-09:
I think I have something on my local branch that will fix this, but I looked at your repository and it seems to contain an Arduino _sketch_ but not an Arduino _library_.  Converting a  sketch to [a library](https://arduino.github.io/arduino-cli/library-specification/) is fairly easy, the sketch just becomes one of the examples and (in addition to appearing in the Arduino menu tree) it will compile exactly as before.

If you have access to an OSX or Linux machine, I have a no-install version of Arduino CI in beta that uses a Docker container for all the testing stuff.  You'd be running something like this in your terminal (changing `/pathTo/RotorHazard` to whatever it is on your system):

```bash
docker run --rm \
  -v "/pathTo/RotorHazard:/library_under_test" \
  --workdir /library_under_test \
  --env USE_SUBDIR=src/node \
  docker.pkg.github.com/arduino-ci/action/ubuntu
```

ethomas997, 2020-12-16:
Command fails with:
`docker: Error response from daemon: Get https://docker.pkg.github.com/v2/arduino-ci/action/ubuntu/manifests/latest: no basic auth credentials.`

ianfixes, 2020-12-16:
Hmm, looks like this is [a known & confusing issue on the GitHub side](https://github.community/t/docker-pull-from-public-github-package-registry-fail-with-no-basic-auth-credentials-error/16358/730).  Don't worry about it for now, I can revisit this if/when they make it easier to switch the Docker hosting to a more publicly accessible (i.e. authentication-free) system.

ethomas997, 2020-12-16:
OK -- let me know if at some point there something you'd like me test on our GitHub repo.

ianfixes, 2020-12-17:
I don't want to impose on your time.  If you get curious and want to get around the docker credential thing, for now you'd have to [create a personal access token on GitHub](https://github.com/settings/tokens) and store it in a file or environment variable.  Then you'd run one of these as appropriate:

```bash
# as an environment variable
echo $GITHUB_TOKEN | docker login --username ethomas997 --password-stdin

# as a file
cat github_token.txt | docker login --username ethomas997 --password-stdin
```

I'm going to see about doing more this weekend.

PR #238 (2020-12-28): Create a more complete CI environment in a Docker image
## Highlights from `CHANGELOG.md`

Working towards a solution where configuration of environment variables is all that controls a Docker image that can be used for testing

### Added
- Environment variable to run a custom initiialization script during CI testing: `CUSTOM_INIT_SCRIPT`
- Environment variable to run from a subdirectory during CI testing: `USE_SUBDIR`

### Changed
- Rubocop expected syntax downgraded from ruby 2.6 to 2.5

### Fixed
- Warnings about directory name mismatches are now based on proper comparison of strings

* See [... cut]
closed 2020-12-28 by ianfixes