Arduino-CI/arduino_ci #237: Error in Arduino-ci run because of dependency on library.properties

RobTillaart, 2020-12-07:
## Issue Summary

Problem occurred during  this run - https://github.com/RobTillaart/RunningAverage/runs/1506309632?check_suite_focus=true

The root cause was that in the library.properties file I had a typo in the line ```includes=RunningAverager.h``` instead of ```RunningAverage.h```  This caused an error in the unit test code stating that the #include file could not be found. It took some time to find the cause as the dependency was not obvious for me. Fixing the typo solved the problem.

The error message in the log file did not give any clue (see below).

## Unit Test log , Illustrating the Problem

<snippet from the log file>
Compiling ra_test.ino for arduino:avr:uno... 
Last command:  $  /usr/local/bin/arduino-cli --format json compile --fqbn arduino:avr:uno --warnings all --dry-run /github/home/Arduino/libraries/RunningAverage/examples/ra_test/ra_test.ino
/github/home/Arduino/libraries/RunningAverage/examples/ra_test/ra_test.ino:10:10: fatal error: RunningAverage.h: No such file or directory
 #include "RunningAverage.h"
          ^~~~~~~~~~~~~~~~~~

## Request

It would be good if 
- the dependency is documented
- the test environment checks the dependencies somehow, 
- e.g. it checks the consistency of library.properties file and the library.json file

Thanks,
Rob

RobTillaart, 2020-12-21:
#250 is indeed another example  + some thoughts about solution in user terms.

ianfixes, 2020-12-23:
via #250 -- check case sensitivity as well

> Today I had a typo (case mismatch) in the field: **includes=DHT.h**  which should be **includes=dht.h** 
> It took a while to find the root cause as the error message just said "**#include <dht.h> not found.** 
> The very same sketch compiled perfect in the IDE for years so that made me wonder.
> 
> **Solutions**
> 1. a better error message why the **#include**  is not found, e.g. what is expected.
> 2. a script that verifies the fields of the **library.properties** file. 
>     All mandatory fields are available and correct?
> 3. document this problem in an FAQ for known error messages 
> 4. ...

ianfixes, 2021-01-02:
I've added scans for both the library dependencies (`depends=`) existence and includes (`includes=`) existence.  Please try to outsmart my checks and let me know if I missed anything.

PR #256 (2021-01-05): Finish 2020 milestone
## Highlights from `CHANGELOG.md`

### Fixed
- `esp8266:huzzah` options updated as per upstream

* See CHANGELOG.md for more

## Issues Fixed

* Fixes #254 - Huzzah / invalid option FlashSize
* Fixes #252 - `shiftIn` support
* Fixes #249 - `pinMode` and `_NOP`
* Fixes #246 - respect `architecture=` when iterating
* Fixes #258 - remove overzealous `pinMode` macro
* Fixes #237  - scan `library.properties` for typos
* Fixes #261 - add information about platform names being case sensitive
* Fixes #262 - scan `library.properties` for typos
closed 2021-01-05 by ianfixes