Arduino-CI/arduino_ci #240: _BV macro missing in unit test

RobTillaart, 2020-12-09:
## Issue Summary

problem occurred in a unit test where I used the **_BV()** macro which is normally supported in Arduino sketches.
Replacing the test code with a normal shift (1UL << (x))  solved the issue. 
However people might expect to be able to use the same code constructs as in the sketches / libraries,
just after ```#include "Arduino.h" ```

ianfixes, 2020-12-23:
Is this the use of a documented or undocumented Arduino API?  My search suggests that it's not universal:

[via avrfreaks.net](https://www.avrfreaks.net/comment/347329#comment-347329)
> as far as I'm aware, `_BV` is only defined in the avr-libc headers. Extracted from `avr/sfr_defs.h` in avr-libc:
> ```c++
> #define _BV(bit) (1 << (bit))
> ```

matthijskooijman, 2020-12-23:
I'm pretty sure that `_BV` is not documented Arduino API. As you mentioned, it is made available by avr-libc, and also used by the AVR Arduino core, so it is de facto exposed by Arduino.h on AVR, but not guaranteed on other platforms (some might define it, but you can't really rely on this). I would even say that code should *not* rely on Arduino.h exposing `_BV`, even on AVR, and they should just include `avr/sfr_defs.h` or some other header that includes that directly, but I guess there's too much code that already assumes that the macro is available on AVR.

I would suggest to add an `avr/sfr_defs.h` in Arduino-ci with this macro, and include that from Arduino.h on AVR only, for compatibility with AVR-specific sketches.

ianfixes, 2020-12-24:
My sense is to only prioritize this if there is some "official" (Arduino-hosted) documentation explaining it, since [there are still a bunch of official functions that need implementation](https://github.com/Arduino-CI/arduino_ci/milestone/1).

RobTillaart, 2020-12-24:
low priority indeed, as workaround is simple.

Thoughts
- maybe get usage out of core Arduino code?
- check how often it is used?

ianfixes, 2022-12-12:
Is the other option here to put it behind some kind of `#ifdef`?  Maybe some more use cases here would sharpen the boundaries of this issue

RobTillaart, 2022-12-12:
As this issue has no reactions for 2 yrs the priority is low.  For me I try not to use the macro (as macros can have side effects).

That said, I searched the Arduino Libraries section (1.8.19) and the BV macro is used in multiple libs incl  SPI and SWSerial.
Maybe that is still a point of attention?