Arduino-CI/arduino_ci #244: Support for comparing floats for equality in unit test ==> AssertEqualFloat(expect, actual, abs_delta) ?

RobTillaart, 2020-12-12:
Some of my libraries use float math and errors add up when doing a lot of float math.
e.g https://github.com/RobTillaart/AverageAngle/runs/1543777648?check_suite_focus=true

To test for equality for a float output I typically use the following working construct
```cpp
  fprintf(stderr, "getTotalLength()\n");
  float diff = abs(10 - aa.getTotalLength());
  assertMoreOrEqual(0.2, diff);
```

however I would prefer a oneliner
```cpp
  AssertEqualFloat(10,  aa.getTotalLength(), 0.2);
```

output should be something like one of the three
```
AssertEqualFloat  10 == aa.getTotalLength()  +- 0.2
AssertEqualFloat  10 +- 0.2 == aa.getTotalLength()
AssertEqualFloat  10- 0.2 <  aa.getTotalLength() < 10 + 0.2
```

Maybe it should be called ```AssertAlmostEqual(...)```

Thanks,

ianfixes, 2020-12-14:
Hi Rob- thanks for contributing this.  

At the moment, [the assertions](https://github.com/Arduino-CI/arduino_ci/blob/master/cpp/unittest/Assertion.h#L33-L42) are really just some thin wrappers around [comparisions](https://github.com/Arduino-CI/arduino_ci/blob/master/cpp/unittest/Compare.h#L106-L112)

I think this will involve some extra macro work; what I have there now was borrowed from [work done by @wmacevoy](https://github.com/mmurdoch/arduinounit/blame/c90665639ec4a3d6a1102d26181c8cc40c48e4a7/src/ArduinoUnitUtility/Compare.h) and I'm hoping they can provide a recommendation as to how to proceed here.

RobTillaart, 2020-12-14:
Something like this could work?
```
#define AssertEqualFloat(arg1, arg2, arg3)  assertOp("assertEqual","expected", fabs(arg1 - arg2),compareEqual,"==","actual",arg3)
```

ianfixes, 2020-12-14:
I think you mean `compareLessEqual`, but yes.  The question is whether I should further generalize it by abstracting away the `fabs(arg1 - arg2)` as a delta function and call the macro `assertAlmostEqual` or some such.  I'm not sure what other types would be relevant for a "equal within delta" calculation, but that's the future-proofing I want to think about here.

RobTillaart, 2020-12-14:
From requirements point of view:
- I want to compare floats
- as these can lose bits an AlmostEqual variant is needed
- Almost equal should have an absolute error as parameter
- by dividing the absolute error by the expected value one can get a relative error.
- at the limits of float size there will be problems, 

AlmostEqual can of course also be used by 
- integers (any size), 
- Complex numbers ?
- vectors ?
- angles - which typically are float or int

Additional  - I just realized - I might need
- **equalsINF(float)** returns OK if float is +INF or -INF
- **equalsNAN(float)** returns OK if float is NAN

RobTillaart, 2020-12-16:
Had a good case to test float (almost) equality with my complex class.

I put the following Macro at top of .....\Complex\test\unit_test_001.cpp
```cpp
#define assertEqualFloat(arg1, arg2, arg3)  assertOp("assertEqualFloat", "expected", fabs(arg1 - arg2), compareLessOrEqual, "<=", "actual", arg3)
```
Screenshot:
[image]

so that works pretty well.

RobTillaart, 2020-12-16:
Got the **assertEqualINF** and **assertEqualNAN** working, not needed yet but quite handy.

```cpp
#define assertEqualFloat(arg1, arg2, arg3)  assertOp("assertEqualFloat", "expected", fabs(arg1 - arg2), compareLessOrEqual, "<=", "actual", arg3)
#define assertEqualINF(arg)  assertOp("assertEqualINF", "expected", INFINITY, compareEqual, "==", "actual", arg)
#define assertEqualNAN(arg)  assertOp("assertEqualNAN", "expected", true, compareEqual, "==", "actual", isnan(arg))

unittest(test_new_operator)
{
  assertEqualINF(exp(800));
  assertEqualINF(0.0/0.0);
  assertEqualINF(42);
  
  assertEqualNAN(INFINITY - INFINITY);
  assertEqualNAN(0.0/0.0);
  assertEqualNAN(42);
}
```

**OUTPUT**:
```
# Subtest: test_new_operator
    ok 1 - assertEqualINF (__builtin_inff ()) == exp(800)
    ok 2 - assertEqualINF (__builtin_inff ()) == 0.0/0.0
    not ok 3 - assertEqualINF (__builtin_inff ()) == 42
      ---
      operator: ==
      expected: inf
      actual: 42
      at:
        file: /github/home/Arduino/libraries/Complex/test/unit_test_001.cpp
        line: 46
      ...
    ok 4 - assertEqualNAN true == isnan((__builtin_inff ()) - (__builtin_inff ()))
    ok 5 - assertEqualNAN true == isnan(0.0/0.0)
    not ok 6 - assertEqualNAN true == isnan(42)
      ---
      operator: ==
      expected: 1
      actual: 0
      at:
        file: /github/home/Arduino/libraries/Complex/test/unit_test_001.cpp
        line: 50
      ...
    1..6
not ok 1 - test_new_operator
```

RobTillaart, 2020-12-16:
Maybe the inverted**assertNot_INF** and **assertNot_NAN** are also (even more) interesting. 
To be investigated....

ianfixes, 2020-12-17:
`assertEqualINF` and `assertEqualNAN` make the case for not bothering to lump in other "almost equal" type of operations.  I think the right approach here is to adopt what you are doing.  

I should also add docs that say how to use `assertOp` to define your own behaviors (e.g. maybe comparing a binary tree of some kind where you use a class method instead of the `==` operator to judge equality).  

Have you tried a case where the assertion fails?  I'm curious what the error message should like since it involves both the expected value and a delta

closed 2020-12-28 by ianfixes

RobTillaart, 2020-12-28:
@ianfixes 
Is the solution added to the documentation?

ianfixes, 2020-12-29:
I put them here https://github.com/Arduino-CI/arduino_ci/blob/master/REFERENCE.md#assertions

PR #238 (2020-12-28): Create a more complete CI environment in a Docker image
## Highlights from `CHANGELOG.md`

Working towards a solution where configuration of environment variables is all that controls a Docker image that can be used for testing

### Added
- Environment variable to run a custom initiialization script during CI testing: `CUSTOM_INIT_SCRIPT`
- Environment variable to run from a subdirectory during CI testing: `USE_SUBDIR`

### Changed
- Rubocop expected syntax downgraded from ruby 2.6 to 2.5

### Fixed
- Warnings about directory name mismatches are now based on proper comparison of strings

* See [... cut]