Arduino-CI/arduino_ci #246: Default iteration of platforms in CI should respect library.properties "architecture" setting

ianfixes, 2020-12-14:
## Feature request

via https://github.com/Arduino-CI/action/issues/11

A `library.properties` file [can include architecture information](https://arduino.github.io/arduino-cli/library-specification/), e.g. 

```
architectures=avr
```

The CI runner should respect this by not attempting to test or compile such a library against `sam` or other architectures.

That suggests the following logic for implementation:

1. if an override `.arduino-ci.yml` file exists, use that list
2. if the `library.properties` file does not exist, use the default list
3. if the `architectures=` line does not exist, use the default list
4. if architectures are specified, generate a candidate list of platforms to test against

## Knowns and unknowns

The library specification says 2 relevant things:

> Arduino development software supports multiple microcontroller architectures (e.g. AVR, SAM, etc), meaning that libraries may need to work on multiple architectures. The new 1.5 library format doesn’t contain special support for cross-architecture libraries, but it does provide a preprocessor based mechanism for libraries to target sections of code to specific architectures.

and

> **architectures** - (defaults to `*`) a comma separated list of architectures supported by the library. If the library doesn’t contain architecture specific code use `*` to match all architectures. This field is used as one factor in determining priority when multiple libraries match an `#include` directive and to provide a warning message when the library is compiled for a board of an architecture that doesn't match any on the list.

@per1234 in practice does this mean simply comparing the middle field of a platform name (e.g. `arduino:sam:arduino_due_x` would be `sam`) to the architectures in the properties file?  Also, is there any reason not to test _all_ platforms that match a given architecture?  On one hand it seems like you'd want to guard against things like pins or serial ports not existing on certain boards, but on the other hand the extra tests are probably diminishing returns and the user can always override the config with specific boards that _need_ testing.

per1234, 2020-12-14:
>does this mean simply comparing the middle field of a platform name

Exactly. That's the board's architecture.

>it seems like you'd want to guard against things like pins or serial ports not existing on certain boards

Yep. For example, `arduino:avr:mega` has `Serial1`, but `arduino:avr:uno` does not. `arduino:avr:leonardo` has HID, but `arduino:avr:mega` does not, and so on. Then you throw different memory capacities in and it gets even more messy. So just because a library specifies `avr` compatibility in its metadata, doesn't mean it's compatible with every AVR board.

ianfixes, 2020-12-14:
OK.  For now I'll brute-force the architecture testing and we can iterate from there.

PR #256 (2021-01-05): Finish 2020 milestone
## Highlights from `CHANGELOG.md`

### Fixed
- `esp8266:huzzah` options updated as per upstream

* See CHANGELOG.md for more

## Issues Fixed

* Fixes #254 - Huzzah / invalid option FlashSize
* Fixes #252 - `shiftIn` support
* Fixes #249 - `pinMode` and `_NOP`
* Fixes #246 - respect `architecture=` when iterating
* Fixes #258 - remove overzealous `pinMode` macro
* Fixes #237  - scan `library.properties` for typos
* Fixes #261 - add information about platform names being case sensitive
* Fixes #262 - scan `library.properties` for typos
closed 2021-01-05 by ianfixes