Arduino-CI/arduino_ci #247: CI environment does not install dependency library SoftwareSerial.h

RobTillaart, 2020-12-17:
My https://github.com/RobTillaart/Cozir lib has a dependency on SoftwareSerial.h
It is mentioned in the properties file but somehow it fails to install.

Partial log. If more info is needed let me know.

```
Installing dependency of <library.properties>: 'SoftwareSerial.h'...           ✗
This may indicate a problem with your configuration; halting here
Failures: 1
Last message:  $  /usr/local/bin/arduino-cli --format json lib install --no-deps SoftwareSerial.h
========== Stdout:

========== Stderr:
Error installing name:"SoftwareSerial.h": looking for library: library SoftwareSerial.h not found
```

per1234, 2020-12-17:
This command installs libraries from the Library Manager index. Because SoftwareSerial is a [platform bundled library](https://arduino.github.io/arduino-cli/latest/platform-specification/#platform-bundled-libraries), it's not available from the Library Manager.

So this library should be installed automatically along with the platform. Of course, quite a few platforms don't provide a SoftwareSerial library. I think it's mostly only in the AVR platforms.

RobTillaart, 2020-12-17:
thanx,

For the CI test I expect those **platform bundled libraries" to be available, similar to when I install the IDE.

I might try to work around the unit test by using Serial from the GODMODE.

ianfixes, 2020-12-17:
I'm seeing `depends=` in your `library.properties`
https://github.com/RobTillaart/Cozir/blob/arduino-ci/library.properties#L11

Am I looking in the wrong place?

ianfixes, 2020-12-17:
Assuming you may have updated your branch since posting this, I'm wondering if you're attempting to install the library (which, as noted, is really bundled with the platform) instead of just restricting which platforms are tested.  

This would be a good candidate for a change I'm making where the default platform list is pulled from `library.properties` (if specified) instead of my own arbitrary list.

RobTillaart, 2020-12-17:
> I'm seeing `depends=` in your `library.properties`
> https://github.com/RobTillaart/Cozir/blob/arduino-ci/library.properties#L11
> 
> Am I looking in the wrong place?

No there I had a depends, but I am investigating a new path using the Godmode and Serial .
Have partial success so far.

ianfixes, 2020-12-18:
I'm concerned that you might be building on a bug in my library.  If Serial isn't available on the platform, it shouldn't be showing up in GODMODE either.

RobTillaart, 2020-12-18:
SoftwareSerial is not available apparently.

But I am happy to tell that I got it all working in - https://github.com/RobTillaart/Cozir - by using the normal Serial.
You might peek there to see how I used the **Godmode.Serial**

ianfixes, 2020-12-23:
That all looks correct.  But I'm wondering why not just use `SoftwareSerial` instead of `Serial`?

https://github.com/Arduino-CI/arduino_ci/blob/master/SampleProjects/TestSomething/test/softwareserial.cpp

Did it not include properly?

ianfixes, 2020-12-23:
I think I understand the workaround, you are just using the builtin `Serial` because it has the same interface.

RobTillaart, 2020-12-23:
> I think I understand the workaround, you are just using the builtin `Serial` because it has the same interface.

Yes I did, and with the GODMODE I could test the behavior.

RobTillaart, 2020-12-23:
> That all looks correct. But I'm wondering why not just use `SoftwareSerial` instead of `Serial`?
> 
> https://github.com/Arduino-CI/arduino_ci/blob/master/SampleProjects/TestSomething/test/softwareserial.cpp
> 
> Did it not include properly?

Nope
```
Installing dependency of <unittest/libraries>: 'SoftwareSerial'...             ✗
This may indicate a problem with your configuration; halting here
Failures: 1
Last message:  $  /usr/local/bin/arduino-cli --format json lib install --no-deps SoftwareSerial
========== Stdout:

========== Stderr:
Error installing name:"SoftwareSerial": looking for library: library SoftwareSerial not found
```

ianfixes, 2020-12-26:
I've submitted what I think is the correct way to implement this test as https://github.com/RobTillaart/Cozir/pull/7

However, since the library itself is written to not care about hardware vs software serial, I think the "workaround" implementation is actually the correct thing to do -- the sending/receiving is much nicer to work with in bytes than in bits.  Just verify that SoftwareSerial works for a small thing but then use Serial for the rest.

Beyond the incorrect specification of `SoftwareSerial` as a library dependency (a line I deleted in my PR), is there anything else that should be scrutinized here?

RobTillaart, 2020-12-26:
Looks good - not verified in detail - and merged., 
And yes testing is lot easier with the Serial class.

closed 2020-12-26 by RobTillaart