Arduino-CI/arduino_ci #248: `match': invalid byte sequence in  UTF-8 (ArgumentError) on Windows

Nykkolas, 2020-12-18:
## System

- OS:Windows
- `ruby -v`: ruby 2.7.2p137 (2020-10-01 revision 5445e04352) [x64-mingw32]
- `bundle -v`: Bundler version 2.1.4
- `bundle info arduino_ci`:
```
 * arduino_ci (1.1.0)
        Summary: Tools for building and unit testing Arduino libraries
        Homepage: http://github.com/Arduino-CI/arduino_ci
        Path: D:/Progs/Ruby27-x64/lib/ruby/gems/2.7.0/gems/arduino_ci-1.1.0
```

## Issue Summary
Hello

When I run `bundle exec arduino_ci.rb --skip-examples-compilation` and the "libraries/symlink" exists I have an error :

```
Installing library under test... Traceback (most recent call last):
        11: from D:/Progs/Ruby27-x64/bin/arduino_ci.rb:23:in `<main>'
        10: from D:/Progs/Ruby27-x64/bin/arduino_ci.rb:23:in `load'
         9: from D:/Progs/Ruby27-x64/lib/ruby/gems/2.7.0/gems/arduino_ci-1.1.0/exe/arduino_ci.rb:399:in `<top (required)>'    
         8: from D:/Progs/Ruby27-x64/lib/ruby/gems/2.7.0/gems/arduino_ci-1.1.0/exe/arduino_ci.rb:129:in `assure'
         7: from D:/Progs/Ruby27-x64/lib/ruby/gems/2.7.0/gems/arduino_ci-1.1.0/exe/arduino_ci.rb:102:in `perform_action'      
         6: from D:/Progs/Ruby27-x64/lib/ruby/gems/2.7.0/gems/arduino_ci-1.1.0/exe/arduino_ci.rb:400:in `block in <top (required)>'
ll_local_library'
         4: from D:/Progs/Ruby27-x64/lib/ruby/gems/2.7.0/gems/arduino_ci-1.1.0/lib/arduino_ci/host.rb:93:in `symlink?'        
         3: from D:/Progs/Ruby27-x64/lib/ruby/gems/2.7.0/gems/arduino_ci-1.1.0/lib/arduino_ci/host.rb:106:in `readlink'       
         2: from D:/Progs/Ruby27-x64/lib/ruby/gems/2.7.0/gems/arduino_ci-1.1.0/lib/arduino_ci/host.rb:106:in `map'
         1: from D:/Progs/Ruby27-x64/lib/ruby/gems/2.7.0/gems/arduino_ci-1.1.0/lib/arduino_ci/host.rb:106:in `block in readlink'
D:/Progs/Ruby27-x64/lib/ruby/gems/2.7.0/gems/arduino_ci-1.1.0/lib/arduino_ci/host.rb:106:in `match': invalid byte sequence in 
UTF-8 (ArgumentError)
```

Note that everything is ok when I run it when the symlink in the Arduino libraries folder doesn't exists.

After some internet research I solved it but replacing the line (host.rb:106) : 
`symlinks = stdout.lines.map { |l| DIR_SYMLINK_REGEX.match(l) }.compact`
with
`symlinks = stdout.lines.map { |l| DIR_SYMLINK_REGEX.match(l.scrub) }.compact`

As I don't know Ruby or your code I have no clue on the consequences of this change but I guess it may be a good start for experts to solve this :)

I wish you a good day !

ianfixes, 2020-12-18:
Hy Nykkolas, thanks for reporting this.  I don't have access to a windows machine so I'm entirely reliant on reports like yours to improve that aspect of the library!

I will probably use the fix you suggested, it looks like the correct one (and if not, unit tests should catch it).

What locale is your system running?  Also, I'm eager to hear your experience in general of using `arduino_ci` on Windows in a symlinking situation.  Basically, my library has to cover the possibility that the project directory might not be located in the system's `Arduino/libraries` directory.  Symlinks are the easiest way to accomplish that, but in Windows they don't work the same way as in OSX/Linux.  If you notice any incorrect behavior there (or can think of a better way to accomplish it) please don't hesitate to open more issues

Nykkolas, 2020-12-18:
Hi Ian

Thanks for your answer !

My locale is  `fr;Français (France)`

I use windows like 10% of the time but I'll be pleased to let you know what happens on it. Until now, the only strange thing with symlinks under windows is that you can use them only as an administrator, in a command prompt started as administrator or if you change the security parameters to "development mode".

PR #238 (2020-12-28): Create a more complete CI environment in a Docker image
## Highlights from `CHANGELOG.md`

Working towards a solution where configuration of environment variables is all that controls a Docker image that can be used for testing

### Added
- Environment variable to run a custom initiialization script during CI testing: `CUSTOM_INIT_SCRIPT`
- Environment variable to run from a subdirectory during CI testing: `USE_SUBDIR`

### Changed
- Rubocop expected syntax downgraded from ruby 2.6 to 2.5

### Fixed
- Warnings about directory name mismatches are now based on proper comparison of strings

* See [... cut]
closed 2020-12-28 by ianfixes