Arduino-CI/arduino_ci #257: Wire support needs improvement to do minimal checks.

RobTillaart, 2020-12-29:
I have several I2C based libraries that therefor use the Wire library.
These are hard to test as there is minimal support in GODMODE.

One of the functions I would like to see supported is connection test from which I expect a time out in fact.
It does a sort of "Ping" over I2C and is code wise core in every I2C scanner.
Unfortunately it doesn't give the expected result.

Example snippet from  - https://github.com/RobTillaart/MCP4725
```cpp
bool MCP4725::isConnected()
{
  Wire.beginTransmission(_deviceAddress);
  return (Wire.endTransmission() == 0);
}
```

Currently this function returns **true** currently in the unittest.
It should return **false** as the end transmission should get a NACK on transmission of the address. 
  // https://www.arduino.cc/en/Reference/WireEndTransmission

Furthermore it would be useful to be able to add a device ID in GODMODE I2C, something like 
```GodMode->Wire->add(deviceAddress);```
So that at least such basic ping could work. 

A more elaborated Wire implementation could set and get register values so I could do something like
```
GodMode->Wire->add(deviceAddress);
GodMode->Wire->setRegister(1, 0xF5); // loads value F5 in a virtual register

n = WireRequestFrom(addr, 1);  // pseudo 
 assertEqual(0xF5, n);

// and to check a written value
value = GodMode->Wire->getRegister(1); 
```

No priority as I am not blocked by it.

A small test program to see status when (not) connected.

```cpp
//    FILE: WireConnectionTest.ino
//  AUTHOR: Rob Tillaart
// VERSION: 0.1.0

#include "Wire.h"

void setup()
{
  Serial.begin(115200);
  Serial.println("Wire connection test");

  Wire.begin();

  uint8_t addr = 0x62;
  Wire.beginTransmission(addr);
  int rv = Wire.endTransmission();

  // https://www.arduino.cc/en/Reference/WireEndTransmission

  Serial.print("ADDRESS:\t");
  Serial.println(addr, HEX);
  Serial.print("RETURNS:\t");
  Serial.print(rv);
  switch (rv)
  {
    case 0:
      Serial.println("\tsuccess");
      break;
    case 1:
      Serial.println("\tdata too long to fit in transmit buffer");
      break;
    case 2:
      Serial.println("\treceived NACK on transmit of address");  // NOT CONNECTED, ADDRESS NOT FOUND
      break;
    case 3:
      Serial.println("\treceived NACK on transmit of data");
      break;
    case 4:
    default:
      Serial.println("\tother error");
      break;
  }
}

void loop()
{
}

// -- END OF FILE --
```

RobTillaart, 2020-12-29:
One of the errors I got (which did not help :)

unittest_unit_test_001.cpp.bin: /action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d86200aa8b3b/cpp/arduino/Wire.h:96: void TwoWire::beginTransmission(uint8_t): Assertion `address > 0 && address < SLAVE_COUNT' failed.

RobTillaart, 2020-12-30:
Forget the registers, 
just an input and output array as Serial has would be enough to do a lot of testing.

ianfixes, 2020-12-31:
I'll see how big the gap is here.  It seems like this should be testable already but I'll add this test case and see where things go.

ianfixes, 2021-01-02:
It looks like that functionality may already exist, although I only recently surfaced it in the docs.  (And even then... let me know what you think).

https://github.com/Arduino-CI/arduino_ci/blob/master/REFERENCE.md#wire

Does this provide what you're looking for?

RobTillaart, 2021-01-02:
Looks good, I hope to be able to test it today / tomorrow

ianfixes, 2021-01-02:
That's my understanding (this code was contributed).  You have to load up the `deque` with the data you expect to read.

If you can suggest any improvements to the mock interface as you work with it (or want to submit a patch there) I'll be glad for that feedback.

ianfixes, 2021-01-05:
Assumed fixed (or at least workable).  Please open a new issue if not -- that one can concentrate on how to make the existing capabilities more streamlined to use

closed 2021-01-05 by ianfixes