Arduino-CI/arduino_ci #258: Failing unit test on NOP

RobTillaart, 2021-01-01:
Has something changed in the NOP definition?

https://github.com/RobTillaart/MCP23017_RT/actions/runs/456403915

In file included from /github/home/Arduino/libraries/MCP23017_RT/MCP23017.h:10,
                 from /github/home/Arduino/libraries/MCP23017_RT/MCP23017.cpp:20:
/action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d86200aa8b3b/cpp/arduino/Arduino.h:40:16: error: expected unqualified-id before ‘do’
 #define _NOP() do { 0; } while (0)
                ^~

ianfixes, 2021-01-01:
My branch / PR for #256 has some changes to the location of the `_NOP` definition (addressing #249), but that shouldn't affect the `Arduino-CI/action@master` that your workflow is pointing to.

I'm wondering what happens if you try `Arduino-CI/action@latest` instead, which should pick up the candidate code for that PR.

RobTillaart, 2021-01-01:
Same problem - https://github.com/RobTillaart/MCP23017_RT/runs/1634781952
(to be continued)

ianfixes, 2021-01-02:
The issue is that I'm trying to use a macro to replace a builtin function... and that has consequences for things that happen to be named `pinMode`...

```
/root/Arduino/libraries/MCP23017_RT/MCP23017.cpp:67:16: note: in expansion of macro ‘pinMode’
 void MCP23017::pinMode(uint8_t pin, uint8_t value)
```

ianfixes, 2021-01-02:
I was able to reproduce the error locally, and I think this is now fixed in the PR branch

matthijskooijman, 2021-01-02:
>     /root/Arduino/libraries/MCP23017_RT/MCP23017.cpp:67:16: note: in expansion of macro ‘pinMode’
 void MCP23017::pinMode(uint8_t pin, uint8_t value)

This is exactly the reason why I would recommend using inline functions over macros wherever possible.

ianfixes, 2021-01-02:
> I would recommend using inline functions over macros wherever possible

No argument here.  All those macros are holdovers from the early days of this project where I wasn't even sure that the project would even compile!  I had much bigger problems than hastily implementing no-op compilation mocks as macros.  

Honestly, if there are any improvements you can suggest to the process of mocking a microprocessor compilation system for the purpose of unit testing... I'll gladly follow that advice. There really weren't any guideposts for a project like this when I started, and I consider it a small miracle that it's accomplished so much.

RobTillaart, 2021-01-02:
@ianfixes 
Everybody understand that building such a system from scratch will bring you often into the dark alleys and corners of the CI universe. Even with good guidance it can be difficult to stay on the main road. You - and the others - have done a great job! 
Respect!

ianfixes, 2021-01-02:
The biggest unknown at this point is how to make the most efficient progress toward mocking avr-libc.  If anyone is aware of some sort of pre-existing compatibility layer (providing all of avr-libc to non-avr architecture compilation) please let me know.

PR #256 (2021-01-05): Finish 2020 milestone
## Highlights from `CHANGELOG.md`

### Fixed
- `esp8266:huzzah` options updated as per upstream

* See CHANGELOG.md for more

## Issues Fixed

* Fixes #254 - Huzzah / invalid option FlashSize
* Fixes #252 - `shiftIn` support
* Fixes #249 - `pinMode` and `_NOP`
* Fixes #246 - respect `architecture=` when iterating
* Fixes #258 - remove overzealous `pinMode` macro
* Fixes #237  - scan `library.properties` for typos
* Fixes #261 - add information about platform names being case sensitive
* Fixes #262 - scan `library.properties` for typos
closed 2021-01-05 by ianfixes