Arduino-CI/arduino_ci #259: Failing unit test on micros() ?

RobTillaart, 2021-01-01:
https://github.com/RobTillaart/ML8511/issues/1

This line causes an eternal hang. I suspect that micros() is not incremented but yield() is my second suspect.
```cpp 
while (micros() - start < 1000) yield(); 
```

ianfixes, 2021-01-02:
I think this is a duplicate of #241

RobTillaart, 2021-01-02:
Yes seems so. (So can be closed)
Small delays are often loops waiting for a clock to proceed. As I run in it again the priority increased a bit (or more)

Think a timeout on unit test level has highest prio as it will be valuable for any test.
Default max 60 seconds per individual unit test seems reasonable start. Godmode could set / change it,

Implementing micros might be harder, but would allow more and better testing. E.g. communication handshakes with sensors which is a great part of my libraries.

ianfixes, 2021-01-02:
Understood.  This is one facet of a much larger issue, which is "how to make the dummy hardware more like actual hardware" -- with time passing, some outputs affecting other inputs, etc.  

The most important thing for me to get right in all of that is having something that (1) works for the majority of cases with minimal configuration and (2) works for special cases with whatever configuration it takes.  So I'm definitely taking my time there.  See #145

closed 2021-01-02 by ianfixes