Arduino-CI/arduino_ci #260: AssureEqual gives an error

RobTillaart, 2021-01-02:
Note is about **assureEqual** not assertEqual

https://github.com/RobTillaart/MS5611/runs/1636462574?check_suite_focus=true

Seems to be a macro expansion side effect too.

```
Last command:  $ g++ -std=c++0x -o /github/workspace/unittest_unit_test_001.cpp.bin -DARDUINO=100 -g -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address -D__AVR__ -D__AVR_ATmega328P__ -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -I/action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d86200aa8b3b/cpp/arduino -I/action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d86200aa8b3b/cpp/unittest -I/github/home/Arduino/libraries/MS5611 /action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d86200aa8b3b/cpp/arduino/Arduino.cpp /action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d86200aa8b3b/cpp/arduino/Godmode.cpp /action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d86200aa8b3b/cpp/arduino/stdlib.cpp /action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d86200aa8b3b/cpp/unittest/ArduinoUnitTests.cpp /github/home/Arduino/libraries/MS5611/MS5611.cpp /github/home/Arduino/libraries/MS5611/test/unit_test_001.cpp

In file included from /action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d86200aa8b3b/cpp/unittest/ArduinoUnitTests.h:4,
                 from /github/home/Arduino/libraries/MS5611/test/unit_test_001.cpp:32:
/github/home/Arduino/libraries/MS5611/test/unit_test_001.cpp: In member function ‘virtual void test_test_read_sensor::task()’:
/action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d86200aa8b3b/cpp/unittest/Assertion.h:10:3: error: expected ‘;’ before ‘do’
   do                                                                   \
   ^~
/action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d86200aa8b3b/cpp/unittest/Assertion.h:29:3: note: in expansion of macro ‘testBehaviorOp’
   testBehaviorOp(true, desc, rel1, arg1, op, op_name, rel2, arg2)
   ^~~~~~~~~~~~~~
/action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d86200aa8b3b/cpp/unittest/Assertion.h:45:38: note: in expansion of macro ‘assureOp’
 #define assureEqual(arg1,arg2)       assureOp("assureEqual","expected",arg1,compareEqual,"==","actual",arg2)
                                      ^~~~~~~~
/github/home/Arduino/libraries/MS5611/test/unit_test_001.cpp:86:3: note: in expansion of macro ‘assureEqual’
   assureEqual(MS5611_READ_OK, sensor.read());
   ^~~~~~~~~~~
...Unit testing unit_test_001.cpp with g++ for uno                             ✗
```

ianfixes, 2021-01-02:
Unfortunately the error is cryptic -- `error: expected ‘;’ before ‘do’` -- but completely accurate. The source is a missing semicolon on your end

```diff
  unittest(test_read_sensor)
  {
    MS5611 sensor(0x77);
  
-   assertTrue(sensor.begin())
+   assertTrue(sensor.begin());
  
    assureEqual(MS5611_READ_OK, sensor.read());
```

closed 2021-01-02 by ianfixes

RobTillaart, 2021-01-02:
mea maxima culpa :)

ianfixes, 2021-01-02:
Maybe, but if you hadn't posted this I wouldn't have found & fixed 10 typos in the assertion macro library: https://github.com/Arduino-CI/arduino_ci/pull/256/commits/32d384e840996565d3868167eefb54715707b10b