Arduino-CI/arduino_ci #261: .arduino-ci.yml is case sensitive for platforms

RobTillaart, 2021-01-02:
Changed **.arduino-ci.yml**   to uppercase for UNO see below, as this is a common way to write I propose to make the strings under **platforms** case independent

```
compile:
  # Choosing to run compilation tests on 2 different Arduino platforms
  platforms:
    - UNO          <<<<<<<<< UPPER CASE = common writing 
    - leonardo
    - due
    - zero
 ```

https://github.com/RobTillaart/MCP23017_RT/runs/1636467517   failed

```
Wildcard architecture in library.properties, using configured platforms... 
    UNO
    leonardo
    due
    zero
...Wildcard architecture in library.properties, using configured platforms      
Board package arduino:sam has a defined URL...                                 ✓
Installing board package arduino:sam...                                        ✓
Board package arduino:samd has a defined URL...                                ✓
Installing board package arduino:samd...                                       ✓
Board package arduino:avr has a defined URL...                                 ✓
Installing board package arduino:avr...                                        ✓
bundler: failed to load command: /action/bundle/ruby/2.6.0/bin/arduino_ci.rb (/action/bundle/ruby/2.6.0/bin/arduino_ci.rb)
NoMethodError: undefined method `[]' for nil:NilClass
  /action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d8a0a0bbd430/exe/arduino_ci.rb:462:in `block (2 levels) in perform_example_compilation_tests'
  /action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d8a0a0bbd430/exe/arduino_ci.rb:459:in `each'
  /action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d8a0a0bbd430/exe/arduino_ci.rb:459:in `block in perform_example_compilation_tests'
  /action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d8a0a0bbd430/exe/arduino_ci.rb:443:in `each'
  /action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d8a0a0bbd430/exe/arduino_ci.rb:443:in `perform_example_compilation_tests'
  /action/bundle/ruby/2.6.0/bundler/gems/arduino_ci-d8a0a0bbd430/exe/arduino_ci.rb:516:in `<top (required)>'
  /action/bundle/ruby/2.6.0/bin/arduino_ci.rb:23:in `load'
  /action/bundle/ruby/2.6.0/bin/arduino_ci.rb:23:in `<top (required)>'
```

ianfixes, 2021-01-02:
The main problem here is that YAML itself is not case sensitive.  The platform names here refer to specific keys defined in the [config `default.yml`](https://github.com/Arduino-CI/arduino_ci/blob/master/misc/default.yml#L20-L35).  

It's an open question what the default should contain and how the platforms should be named -- one day I hope to be able to delegate all of this stuff to a backend.

ianfixes, 2021-01-02:
Just to explain this a bit better: to absolve myself of having to write definitions for every possible platform that someone might want to test, the config structure builds up package definitions, then adds platform definitions that refer to the packages.  The unit testing and compilation testing config sections refer to those platform definitions.

Since the `default.yml` lives in the gem, there's (unfortunately) some distance between the platform definitions and the place where users will refer to them.  So I had to pick a convention, and lowercase was it.  Changing it now will have implications for all existing configurations currently in the wild.

I'm going to close this for now, if we reopen it then I'd prefer that it refer to a feature request in the `arduino-cli` backend or some other source that will contain (and maintain) an authoritative platform list.

closed 2021-01-02 by ianfixes

RobTillaart, 2021-01-02:
Thanks,
Please add one remark in the documentation about mandatory lower case 
e.g. here - https://github.com/Arduino-CI/arduino_ci/blob/master/REFERENCE.md#control-how-examples-are-compiled

ianfixes, 2021-01-02:
Absolutely

PR #256 (2021-01-05): Finish 2020 milestone
## Highlights from `CHANGELOG.md`

### Fixed
- `esp8266:huzzah` options updated as per upstream

* See CHANGELOG.md for more

## Issues Fixed

* Fixes #254 - Huzzah / invalid option FlashSize
* Fixes #252 - `shiftIn` support
* Fixes #249 - `pinMode` and `_NOP`
* Fixes #246 - respect `architecture=` when iterating
* Fixes #258 - remove overzealous `pinMode` macro
* Fixes #237  - scan `library.properties` for typos
* Fixes #261 - add information about platform names being case sensitive
* Fixes #262 - scan `library.properties` for typos