Arduino-CI/arduino_ci #268: Test environment does support uint64_t  - documentation should mention

RobTillaart, 2021-01-08:
Building the statHelpers library 
- https://github.com/RobTillaart/statHelpers/runs/1669359173?check_suite_focus=true

I found that printing uint64_t  is natively supported (UNO's Serial print does not )  by using **%ld**

Maybe the reference can mention 2 lines about this. 
(low prio - documentation)

ianfixes, 2021-01-08:
I'm getting a 404 on that link. 

Just so I understand the issue, is it that the test environment is allowing uint64_t even though the actual hardware won't support it?

RobTillaart, 2021-01-08:
Oops might have removed that specific run as I had many test runs.
But yes , the UNO does support uint64_t in math but cannot print them (without my printHelpers class)
The test environment can print them, so that is good for testing!

This one should work
https://github.com/RobTillaart/statHelpers/actions/runs/472101006